raspberrypi/linux #1237: autofs - ntfs - bug

manusch-hz, 2015-12-23:
Hi guys,
I spent ours on finding out whats going on here.
I wanted to mount my USB stick in Text-mode with autoMount (autofs). so i disabled GUI and got back to commandline mode. Here we dont have automount functionality.
Every time it should work, my USB stick didnt mount in the system.
At the moment there is a workaround I used.

Even an
apt-get update
and
apt-get upgrade
couldnt help.

For more information, read the description here:
(Found finaly the solution here:)

https://bugs.launchpad.net/ubuntu/+source/autofs/+bug/1448678

Hope I could help.

linuxuser44, 2016-01-06:
Same problem here. Did some more digging...
When running: sudo automount -f -v
and then trying to access the mount I get the following:
mounted indirect on /media with timeout 2, freq 1 seconds
attempting to mount entry /media/usb1
>> /sbin/mount.exfat: invalid option -- 's'
>> Usage: /sbin/mount.exfat [-d] [-o options] [-V] <device> <dir>
>> FUSE exfat 1.1.0
mount(generic): failed to mount /dev/sda1 (type exfat) on /media/usb1
failed to mount /media/usb1

manusch-hz, 2016-01-06:
Hi,
i'm very happy that i could help you.
I already made a bug report at raspbian.
But don't know how far they fixed the bug.

Sorry for my excellent English xD

popcornmix, 2016-01-06:
I'll close this as it's not a linux kernel bug. Reporting to raspbian is the right place.

closed 2016-01-06 by popcornmix