raspberrypi/linux #1242: Poor OOB experience with R-Pi2 and HDMI to VGA adapter

HughBurns, 2015-12-28:
The reported use of autodetect for HDMI during boot and setup does properly support the realistic option of inter-operating with cheap VGA converters. With entry-level computing it should not be assumed every monitor existing is HDMI, and that there remains a large base of existing VGA screens (including flat screens) still to be supported. It should be noted these commonly available VGA adapters may not report back presence and resolution to the HDMI standard. It is thus dangerous to assume all connected HDMI devices will report back, without providing users with fall-safe display fallback options.

1) Specifially the NOOBS_v1_5_0.zip image, when manually HDMI is selected/forced (1 key), asks "do you wish to make the selection permanent? But (bug 1) this "HDMI fixed output" setting is not carried over to the operating system image created by the set up program.

2) Adding a config.txt to the root SD directory of the Raspbian Win-visible partition to force HDMI works for the initial Raspbian OS boot
ie hdmi_force_hotplug=1 
framebuffer_width=1280
framebuffer_height=720

3) However (bug 2) after the localisation is set and the OS requests to reboot, the reboot again produces no HDMI output via the VGA adapter, even-though the config.txt file should be forcing this. Only a blank raster is present yet again.

I note in comparison OpenELEC-RPi2.arm-5.0.8.img operates faultlessly with no intervention (image displays on VGA monitor via adapter) with the exactly same hardware setup. So does every other HDMI output device that I have, including Android4 boxes....

Comment: Frankly the Raspbian startup/boot needs to be made more display hardware robust and OpenElec proves it can be done. As a very experieced PC user, this apparent failure to anticipate likely HDMI connection issues is a fairly disappointing introduction to your new computing concept

HughBurns, 2016-01-04:
Further testing with a Sony G520 CRT monitor (!) (because it reports what is going on) suggests the cause of "bug 2" is that the config.sys settings are being ignored by the OS after the initial start screens and Raspian is going to 1920x1080@60 desktop regardless of config.txt.

With the correct (manual) settings for an Acer AL1716 (1280x1224@60) in the config.txt 
hdmi_drive=2
hdmi_group=2
hdmi_mode=35
You get the colour flash screen and then logo with "shift for recovery" screen then a blank raster...
The reason it is blank is VGA output (from HDMI converter) is overscaning beyond what monitor supports.

Firmware is not the issue, as OpenElec auto correct config/startup works fine with exactly the same hardware and setup (only change SD card and restart). 

Under Raspbian there appear no GUi options to change/override the screen resolution manually.
As a patch are there command line options to do this? (I have extensive windows but no Linux experience)
A full fix for what appears a major OS bug would give Raspbian something like the OpenElec auto configuration startup.
I would also note a "fix" by manually adjusting config.txt is something out of the DOS era. 

Given my Pi is virtually useless until this is fixed/patched, I would appreciate your prompt advice.

I am happy to assist you in providing files to help resolution of this issue.

H

lurch, 2016-01-18:
"bug 2" seems to be because you're confusing Raspbian-standalone with Raspbian-in-NOOBS.
For a NOOBS SD-card, the "Win-visible partition" is the NOOBS boot partition, which isn't the same as the Rasbian boot partition (which is in fact separate). When you use NOOBS to boot Raspbian, the Pi effectively reboots, and reads the `config.txt` from the Raspbian boot partition.
So the it sounds like the `config.txt` that you're editing is the one for NOOBS, not the one for Raspbian. However the NOOBS GUI (accessed by pressing shift at power-on) *does* have a [built-in editor](https://github.com/raspberrypi/noobs#easy-config-file-editor) for editing the `config.txt` on the Raspbian boot partition.

But there's also a number of [displaymode-related NOOBS bugs](https://github.com/raspberrypi/noobs/labels/displaymode), which are on my TODO list, I just haven't had time to get round to them yet.

> I would also note a "fix" by manually adjusting config.txt is something out of the DOS era.

`config.txt` on the Pi's SD card, is roughly equivalent to the BIOS settings screen on a PC.

HughBurns, 2016-01-19:
Thanks for the reply.

Focusing on key issues - yes I found settings are required in both config.txt files. The first is used for the flash screen and recovery screen etc. and the second is used to actually boot the Raspbian OS. 
Some obvious documentation would be great too, and then you have the Catch 22 that if you can't see any display pre or post OS boot (as it ends up after setup), how can you edit the configuration file(s)?

My overview complaint is the logic failure in not making this whole process fully idiot-proof ("fail-safe"). I only escaped by having vintage CRT hardware around that coped with whatever the Pi (with VGA adapter attached) actually produced - this enabled me to then delve into it from first principals, with some advice from Dr. Google (at the cost of time).

One has to remember that norms in the PC industry were reset by the auto-self configuration starting in Windows 95.
The config.txt file is not quite "bios" at all, its effectively config.sys and autoexec.bat - ie like DOS with VESA display. Yes, I'm old enough to know, having been there, done that....

Going back to the issues here, changing the monitor on blank raster (with the second then reporting the blank raster resolution) suggested the error may not be determining the monitor mode, which may well be occurring automatically OK in Raspbian (like OpenELEC manages OK), but could be a failure in setting the appropriate (matching) console resolution.

I eventually made mine work by brute force settings to both config.txts, but there are other issues with this, such as the device self ignoring/varying the display settings when instructed to use one set mode only...

I return to my two bugs - 1) Like OpenELEC, NOOBS should always produce a start display automatically. It is certainly attempting to setup the second config.txt (eg sets force hdmi OK) but fails to produce/set the correct parameter settings for the os to boot with the display visible and entirely misses the first config.txt (in Win32 area) - which is not initially even present after "setup".
2) There is a bug with the autodetect producing a visible OS display, possibly due to the console resolution being incorrectly set (not being read - goes to max?). Needs more investigation.

Strategic Suggestions

This whole area of the PI Raspbian startup display setting needs a full review to understand the logic of what is going wrong and then a complete logical rethink as to how to ensure a working display is always produced for new users, not mattering what display hardware they use.

An "up and running" manual for Raspbian should also be supplied with the board by means of paper or weblink. All hardware start up options (eg display options) should be worked through to cover every reasonable possibility, and all of these should be included in this document. 

There needs to be a lot more work done on adding automatic Pi hardware detection and OS driver/setup within Raspbian... We fixed this issue 20 years ago on another major platform... Probably needs a project team to be formed to focus on adding this functionality.

You have two types of user-  tinkerers and those that want a cheap and small new type of PC (and expect most things to just work like windows XP). On-line software installation is already impressive, lets build the hardware automation to match...

closed 2016-01-19 by HughBurns