raspberrypi/linux #1243: Using usbboot to boot, format microSD, and replicate one PiZero to another

hh, 2015-12-28:
@lurch I'm thinking it would be interesting to combine https://github.com/raspberrypi/tools/tree/master/usbboot running on an rpi zero in host mode (or any rpi really) to relplicate itself to a second rpi that starts with an unformatted or emtpy microSD card, which could in turn replicate to others with very little technical know how.

In trying to start from source, does anyone know where [these binaries](https://github.com/raspberrypi/tools/blob/master/usbboot/Makefile#L7) come from:

```
	cp usbbootcode.bin /usr/share/rpiboot
	cp msd.elf /usr/share/rpiboot
	cp buildroot.elf /usr/share/rpiboot
```

https://github.com/raspberrypi/tools/blob/master/usbboot/usbbootcode.bin
https://github.com/raspberrypi/tools/blob/master/usbboot/msd.elf
https://github.com/raspberrypi/tools/blob/master/usbboot/buildroot.elf

I can't quite tell if they all come from [buildroot.patch](https://github.com/raspberrypi/tools/blob/master/usbboot/buildroot.patch) or somewhere else.

hh, 2015-12-28:
In trying to start from source, does anyone know where [these binaries](https://github.com/raspberrypi/tools/blob/master/usbboot/Makefile#L7) come from:

```
	cp usbbootcode.bin /usr/share/rpiboot
	cp msd.elf /usr/share/rpiboot
	cp buildroot.elf /usr/share/rpiboot
```

https://github.com/raspberrypi/tools/blob/master/usbboot/usbbootcode.bin
https://github.com/raspberrypi/tools/blob/master/usbboot/msd.elf
https://github.com/raspberrypi/tools/blob/master/usbboot/buildroot.elf

I can't quite tell if they all come from [buildroot.patch](https://github.com/raspberrypi/tools/blob/master/usbboot/buildroot.patch) or somewhere else.

hh, 2015-12-28:
I think this may be of interest to some of the folks who worked on getting usb gadget support up in the first place: /CC #1212

P33M, 2015-12-28:
Is this a bug report?

lurch, 2015-12-28:
It's quite a while since I last played with this - the person to ask would obviously be @ghollingworth 

I *think* that `usbbootcode.bin` is analogous to `bootcode.bin` from  https://github.com/raspberrypi/firmware/tree/master/boot , and `msd.elf` and `buildroot.elf` are then analogous to `start.elf` from that same firmware directory. I.e. these are all closed-source low-level binaries that run on the GPU itself, and so can only be made by Raspberry Pi engineers.

Given that it's not recommended to try DD-ing from a mounted filesystem, I guess for one RPi-Zero to be able to write a complete OS to another RPi-Zero containing a blank SD card, the booted RPi-Zero would need to have a copy of the raspbian.img on it's filesystem. So for that img file to then be self-replicating, is an interesting chicken-and-egg problem... ;-)   (I'm sure it's probably possible, I'm just too tired to think about it right now)

hh, 2015-12-28:
@lurch I wouldn't dd directly, I'd suggest formatting the disk / partitons, mounting them,, then ```cp -a``` or equivalent from the source to the target. :)

@P33M This is more for discussion not a bug report. Can you use a different github issue label for this so it's not looked at as a bug report or should I try and move this elsewhere?

https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=53007 has the final discussion on the binary bootloader matters. I didn't realize the internal hardware / software dependencies. No biggy.

P33M, 2015-12-28:
I'd suggest requesting the feature on the forums. That way you can 
a) get a wider audience of people that may be interested in the feature
b) get feedback that modifies the original idea

which then feeds into specific feature requests. We get notified of open issues *anyway* so pinging us doesn't add much to the discussion. if there's a specific requirement for feature Z then with requirements identified you can put the requests in the right place, e.g. firmware or NOOBS. As it stands, this request doesn't require any Linux-specific changes.

hh, 2015-12-28:
Closing for now, will update with a link to a forum post for further discussion.

closed 2015-12-28 by hh

t3chguy, 2016-01-13:
@hh, have you got a forum thread up on this? Interests me.