raspberrypi/linux #1247: multiple definition of `rpisense' and other

fordfrog, 2016-01-03:
i try to compile latest kernel without loadable module support but i get error about multiple definition of rpisense and other symbols. here are the steps to reproduce it (cross compilation):

make ARCH=arm bcm2709_defconfig
make ARCH=arm oldconfig
make ARCH=arm menuconfig (and disable loadable module support)
make ARCH=arm CROSS_COMPILE=/usr/bin/armv7a-hardfloat-linux-gnueabi- -j9

after some time the compilation fails with this error:
...
drivers/mfd/built-in.o: In function `.LANCHOR1':
rpisense-core.c:(.bss+0x0): multiple definition of `rpisense'
drivers/video/built-in.o:rpisense-fb.c:(.bss+0x1e2c): first defined here
drivers/input/built-in.o: In function `.LANCHOR1':
yealink.c:(.bss+0x3c): multiple definition of `rpisense'
drivers/video/built-in.o:rpisense-fb.c:(.bss+0x1e2c): first defined here
drivers/staging/built-in.o: In function `rtw_hal_antdiv_before_linked':
fbtft_device.c:(.text+0x5a838): multiple definition of `rtw_hal_antdiv_before_linked'
drivers/net/built-in.o:cdc_mbim.c:(.text+0xcbfec): first defined here
drivers/staging/built-in.o: In function `bss_cap_update_on_sta_join':
fbtft_device.c:(.text+0x2f714): multiple definition of `bss_cap_update_on_sta_join'
drivers/net/built-in.o:cdc_mbim.c:(.text+0xbfaa8): first defined here
drivers/staging/built-in.o: In function `rtw_txframes_pending':
fbtft_device.c:(.text+0x53f80): multiple definition of `rtw_txframes_pending'
drivers/net/built-in.o:cdc_mbim.c:(.text+0xc122c): first defined here
drivers/staging/built-in.o: In function `rtw_drvextra_cmd_hdl':
fbtft_device.c:(.text+0x32990): multiple definition of `rtw_drvextra_cmd_hdl'
drivers/net/built-in.o:cdc_mbim.c:(.text+0xa12d0): first defined here
drivers/staging/built-in.o: In function `mgt_dispatcher':
fbtft_device.c:(.text+0x3fb08): multiple definition of `mgt_dispatcher'
drivers/net/built-in.o:cdc_mbim.c:(.text+0xac59c): first defined here
drivers/staging/built-in.o: In function `rtw_get_cur_max_rate':
... (more errors)

popcornmix, 2016-01-04:
Looks like some variables are not declared as static when they should.
Can you try applying this patch: http://paste.ubuntu.com/14399820/

fordfrog, 2016-01-04:
compiles without any issues. thank's a lot :-)

popcornmix, 2016-01-04:
Thanks for testing. I've committed that patch.

closed 2016-01-04 by popcornmix