raspberrypi/linux #1248: LAN Performance on Model B regression

GM-Script-Writer-62850, 2016-01-04:
I just rebooted my pi for the 1st time is 181 days, i was on the 3.12.28+ kernel
now i am on the 4.1.13+ kernel
```
pi@raspberrypi:~$ iperf -c 10.0.0.50 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 46.1 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 51872 connected with 10.0.0.50 port 5001
[  5] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 41894
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  35.8 MBytes  29.9 Mbits/sec
[  5]  0.0-10.2 sec   112 MBytes  92.2 Mbits/sec
```

```
pi@raspberrypi:~$ iperf -c 10.0.0.50 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 36108 connected with 10.0.0.50 port 5001
[  5] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 46058
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.1 sec  15.1 MBytes  12.6 Mbits/sec
[  5]  0.0-10.3 sec  74.2 MBytes  60.6 Mbits/sec
```

lurch, 2016-01-18:
There have obviously been **many** kernel versions between 3.12.28+ and 4.1.13+.
Using different firmware versions (see https://github.com/Hexxeh/rpi-update#options ) I suspect you'd need to identify the 'last fast' and 'first slow' version, before the kernel developers are able to look into this in any more depth?

GM-Script-Writer-62850, 2016-01-18:
if you can give me some commands for say to install the 3.18, 3.16, and 4.0 i can check and see where it shows up

kukabu, 2016-01-19:
try smsc95xx.turbo_mode=Y now it's disabled by default.

GM-Script-Writer-62850, 2016-01-19:
that is a kernel parameter right?
goes in `/boot/cmdline.txt`

popcornmix, 2016-01-19:
Yes.

GM-Script-Writer-62850, 2016-01-19:
that is even worse
```
iperf -c 10.0.0.50 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  5] local 10.0.0.25 port 44952 connected with 10.0.0.50 port 5001
[  4] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 56800
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.1 sec  30.4 MBytes  25.3 Mbits/sec
[  4]  0.0-10.1 sec  46.0 MBytes  38.3 Mbits/sec
```

popcornmix, 2016-01-19:
If it's a regression then identifying when it occurred would be useful.
https://github.com/Hexxeh/rpi-firmware/commits/master contains all the firmware/kernel updates.
Click on a commit and the end of the url contains the git hash (the 40 character hex number).

Run: ``sudo rpi-update <git hash>``  to revert to a specific firmware and reboot and test.
Report here the first update that has the reduced performance.

GM-Script-Writer-62850, 2016-01-19:
`sudo rpi-update 6413da9f74871b239c5bd27d7edf90a8afeab363` - Speed: 90/30 (3.12.63)
`sudo rpi-update 8521fd34c8f66b6d109acce943f6e25ec93ec005` - Speed: 30/15 (3.18.13)
`sudo rpi-update f74b92120e0d469fc5c2dc85b2b5718d877e1cbb` Speed: 30/90 (3.12.36)

i would say it 1st appeared in this commit: 
https://github.com/Hexxeh/rpi-firmware/commit/cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45
the bad thing is this change skipped over a LOT of kernel versions, so up stream reporting would mean a lot of leg work

popcornmix, 2016-01-20:
I was hoping it wouldn't be one the "big" bumps.

Pi 2 is still getting good speeds (90+ Mbits/sec) with latest kernel. I do see lower speeds on Pi1.

Pi 1 on f74b92120e0d469fc5c2dc85b2b5718d877e1cbb
```
pi@domnfs:~ $ iperf -c 192.168.0.162 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 192.168.0.162, TCP port 5001
TCP window size: 46.1 KByte (default)
------------------------------------------------------------
[  5] local 192.168.0.122 port 50843 connected with 192.168.0.162 port 5001
[  4] local 192.168.0.122 port 5001 connected with 192.168.0.162 port 53896
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.1 sec  34.4 MBytes  28.7 Mbits/sec
[  4]  0.0-10.0 sec   110 MBytes  92.2 Mbits/sec
```
and on 8521fd34c8f66b6d109acce943f6e25ec93ec005
```
$ iperf -c 192.168.0.162 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 192.168.0.162, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  5] local 192.168.0.122 port 52950 connected with 192.168.0.162 port 5001
[  4] local 192.168.0.122 port 5001 connected with 192.168.0.162 port 56328
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.1 sec  15.1 MBytes  12.6 Mbits/sec
[  4]  0.0-10.1 sec  79.0 MBytes  65.7 Mbits/sec
```
Now there were some additional builds on BRANCH=next between 3.12.36 and 3.18.13
E.g. 656cd4dd790e8d43c6fab3cfb690256029b2cf5e is 3.17.4 and still looks good
```
pi@domnfs:~ $ iperf -c 192.168.0.162 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 192.168.0.162, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 192.168.0.122 port 33069 connected with 192.168.0.162 port 5001
[  5] local 192.168.0.122 port 5001 connected with 192.168.0.162 port 58400
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.1 sec  28.9 MBytes  24.1 Mbits/sec
[  5]  0.0-10.1 sec   109 MBytes  91.0 Mbits/sec
```
Lets see if I can narrow it any further.

popcornmix, 2016-01-20:
Bah - blind alley. All the BRANCH=next builds are good for me, as is the assumed bad ``cad071ac50ddf8a9f3c7a5d808cfa0daf5effb45``. It went wrong more recently than that...

popcornmix, 2016-01-20:
Okay the commit that lowers iperf numbers on Pi1 for me is:
https://github.com/Hexxeh/rpi-firmware/commit/9ddadee099333a3c70f22f80800afa049cc1d57b
which is not too surprising. That commit does two things, one is disabling turbo (which can be enabled again with smsc95xx.turbo_mode=Y) which doesn't seem to be the cause.

I'll check if the other change (to truesize) is definitely the commit responsible.

popcornmix, 2016-01-20:
I've added a cmdline parameter to optionally revert the truesize change.
With both the truesize and turbo options undone, the latest kernel has the same iperf numbers:
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.1.15+ #1730 Wed Jan 20 17:40:26 GMT 2016 armv6l GNU/Linux
pi@domnfs:~ $ iperf -c 192.168.0.162 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 192.168.0.162, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  5] local 192.168.0.122 port 39142 connected with 192.168.0.162 port 5001
[  3] local 192.168.0.122 port 5001 connected with 192.168.0.162 port 48250
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.1 sec  32.4 MBytes  27.0 Mbits/sec
[  3]  0.0-10.1 sec   110 MBytes  91.3 Mbits/sec
```
However both those options have been changed for a reason (and came from the smsc95xx driver author), so I wouldn't necessarily recommend changing them. While iperf may go faster, your network connection may not be as reliable.

ncguk, 2016-01-20:
@popcornmix Can you tell us what the cmdline parameter is please?

popcornmix, 2016-01-20:
It's new so won't work until an updated kernel has been fixed. You would need:
``smsc95xx.turbo_mode=Y smsc95xx.truesize_mode=Y``
to get the highest iperf speed.

GM-Script-Writer-62850, 2016-01-20:
When i 1st setup my pi about 2 years ago i had a LAN stability issue, i updated to the kernel i was on back then and had has 100% stability with my lan (NO OC)
unless my kernel updated on its own via unatteneded-grades i have used 3.12.28 for over 1.5 years with the only poweroffs being from power failure and that one time i changed the power supply

popcornmix, 2016-01-24:
Latest rpi-update firmware should support:
``smsc95xx.turbo_mode=Y smsc95xx.truesize_mode=Y``
Please test.

GM-Script-Writer-62850, 2016-01-24:
```
sudo rpi-update 299c25c84a815be9c734677b3027d34f86591e07
pi@raspberrypi:~ $ cat /boot/cmdline.txt 
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait smsc95xx.turbo_mode=Y smsc95xx.truesize_mode=Y
pi@raspberrypi:~ $ iperf -c 10.0.0.50
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 60595 connected with 10.0.0.50 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  59.6 MBytes  49.9 Mbits/sec
pi@raspberrypi:~ $ iperf -c 10.0.0.50 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  5] local 10.0.0.25 port 60596 connected with 10.0.0.50 port 5001
[  4] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 56196
[ ID] Interval       Transfer     Bandwidth
[  5]  0.0-10.1 sec  31.4 MBytes  26.1 Mbits/sec
[  4]  0.0-10.3 sec   112 MBytes  91.6 Mbits/sec
pi@raspberrypi:~ $ iperf -c 10.0.0.50
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 60597 connected with 10.0.0.50 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  58.8 MBytes  49.1 Mbits/sec
pi@raspberrypi:~ $ uname -r
4.1.16+
pi@raspberrypi:~ $
```
odd why am i only getting ~50 one way? it should be around 90
just maybe
```
pi@raspberrypi:~ $ iperf -s
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 56474
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.2 sec   114 MBytes  94.0 Mbits/sec
[  5] local 10.0.0.25 port 5001 connected with 10.0.0.50 port 56518
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  6] local 10.0.0.25 port 60598 connected with 10.0.0.50 port 5001
[  6]  0.0-10.2 sec  26.0 MBytes  21.4 Mbits/sec
[  5]  0.0-10.4 sec   113 MBytes  91.1 Mbits/sec
```
well ok then

popcornmix, 2016-01-25:
Is performance with latest kernel and ``smsc95xx.turbo_mode=Y smsc95xx.truesize_mode=Y`` any different from the older kernel?

GM-Script-Writer-62850, 2016-02-27:
now i am on branch=next

```
pi@raspberrypi:~ $ uname -r
4.4.2+
pi@raspberrypi:~ $ iperf -c 10.0.0.50
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 48540 connected with 10.0.0.50 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  42.5 MBytes  35.6 Mbits/sec
pi@raspberrypi:~ $ iperf -c 10.0.0.50
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 48542 connected with 10.0.0.50 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  55.8 MBytes  46.5 Mbits/sec
pi@raspberrypi:~ $ cat /boot/cmdline.txt 
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait smsc95xx.turbo_mode=Y smsc95xx.truesize_mode=Y
```
was the patch removed?

popcornmix, 2016-02-27:
I've added the patch now to source tree. It will appear in the BRANCH=next build.

GM-Script-Writer-62850, 2016-03-05:
i don't think it worked
```
pi@raspberrypi:~ $ sudo rpi-update fe2fd15806ab143fd73e6c606e5257afc3797de9
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    334      0 --:--:-- --:--:-- --:--:--   336
100 50.0M    0 50.0M    0     0   456k      0 --:--:--  0:01:52 --:--:--  201k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.4.4-v7+
 *** depmod 4.4.4+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to fe2fd15806ab143fd73e6c606e5257afc3797de9
 *** A reboot is needed to activate the new firmware
pi@raspberrypi:~ $ sudo reboot 

Broadcast message from pi@raspberrypi on pts/1 (Fri 2016-03-04 20:28:50 EST):

The system is going down for reboot NOW!

pi@raspberrypi:~ $ Connection to 10.0.0.25 closed by remote host.
Connection to 10.0.0.25 closed.
chad@Z97K1LLER:~$ ssh pi@10.0.0.25

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Mar  4 20:28:26 2016 from 10.0.0.50
pi@raspberrypi:~ $ iperf -c 10.0.0.50
------------------------------------------------------------
Client connecting to 10.0.0.50, TCP port 5001
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 10.0.0.25 port 40014 connected with 10.0.0.50 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  52.5 MBytes  43.9 Mbits/sec
pi@raspberrypi:~ $ 
```

Ruffio, 2016-08-17:
@GM-Script-Writer-62850 has your issue been resolved? If so, please close this issue. Thanks.

GM-Script-Writer-62850, 2016-08-17:
I had turned the option off trying to debug a stability issue i was having, i think it was a issue with the design of the plug on a usb device

closed 2016-08-17 by GM-Script-Writer-62850