raspberrypi/linux #1250: WLAN: USB Wireless device stick 0bda:818b does not work

vit1251, 2016-01-05:
Hello kernel developer,

    I pay on eBay USB Wireless device 0bda:818b it is USB WiFI WLAN card. Supporting only Wireless 802.11n standart.

    Now it does not work on current version of system. Could you provide more details about this problem? How to fix this?

Thanx.

popcornmix, 2016-01-06:
I don't believe that device is supported by upstream linux. As such we don't support it.

There may be a workaround here:
https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=103989

but really you should check for upstream linux support before buying a dongle.

pravsterrr, 2016-01-10:
Try using a different Linux kernel. Sometimes a kernel does not support one USB dongle. For example the version 4 did not support my USB whereas the version 3 did

Ruffio, 2016-08-17:
@vit1251 has your issue been resolved? If so, please close this issue. Thanks.

vit1251, 2016-08-17:
Techivally, no. But I lost this WiFi dongle and it resolve current issue ;)
Have a nice day.

closed 2016-08-17 by vit1251