raspberrypi/linux #1251: bcm2835-v4l2 has wrong values for ISO_SENSITIVITY

6by9, 2016-01-06:
From the spec (http://hverkuil.home.xs4all.nl/spec/media.html)
> V4L2_CID_ISO_SENSITIVITY 	integer menu	 
> 	Determines ISO equivalent of an image sensor indicating the sensor's sensitivity to light. The numbers are expressed in arithmetic scale, as per [ISO 12232:2006] standard, where doubling the sensor sensitivity is represented by doubling the numerical ISO value. Applications should interpret the values as standard ISO values multiplied by 1000, e.g. control value 800 stands for ISO 0.8.

The current driver advertises 0, 100, 200, 400, 800, when it should be 0, 100000, 200000, 400000, 800000. 
It seems to only be a display thing for most applications, as values seem to be selected by index into that array. 

It does appear that there is a second control for ISO as well:
> V4L2_CID_ISO_SENSITIVITY_AUTO 	enum v4l2_iso_sensitivity_type	 
> 	Enables or disables automatic ISO sensitivity adjustments.
> V4L2_CID_ISO_SENSITIVITY_MANUAL 	Manual ISO sensitivity.
> V4L2_CID_ISO_SENSITIVITY_AUTO 	Automatic ISO sensitivity adjustments.

Ought to support that too (added in 3.5 apparently) - not sure why it was missed first time around.

6by9, 2016-01-06:
I'll look into fixing this up when I get a chance.

Austriker, 2016-04-25:
I found out that the manual ISO is not working properly. It changes the value in the `v4l2-ctl --log-status`
But nothing happens on the image. 

Here is the topic where I explained how I produce the issue : 
[https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=145397](url) 

I have been going through the code but I haven't found the solution yet.

6by9, 2016-04-25:
Closing this issue as the defect originally reported is fixed.

Yes I have noted the report that it appears not to behave as expected (and had commented on the forum thread), but that is not the defect originally reported for this issue. Golden rule of issue tracking - one issue per ticket.

closed 2016-04-25 by 6by9