raspberrypi/linux #1252: Setting an ALSA audio buffer with 1024 bytes fails

vanfanel, 2016-01-06:
Hi,

While trying to code an audio program that needs to have near no delay, I found out that I can't setup ALSA to use an audio buffer with a size of 1024 bytes on the Raspberry Pi.
It seems like a small enough buffer, but for a 44100 Hz, 16 bit, stereo stream, a 2048 bytes buffer would cause a delay of:
2048 buffer size / (44100 frames per second * 4 bytes per frame) = 11.6 ms, which is pretty high for realtime audio. 1024 bytes would cause a more acceptabe ~6 ms delay.

This is the failing code: it works on my desktop PC using stock Ubuntu kernel:
`
#define SOUND_SAMPLERATE 44100

int alsa_init ()
{
	snd_pcm_t *pcm_handle;
	snd_pcm_hw_params_t *hwparams = NULL;
	snd_pcm_uframes_t buffer_size_frames;
	snd_pcm_uframes_t period_size_frames;
	size_t buffer_size_bytes;
        size_t period_size_bytes;

	const char *alsa_dev = strdup("plughw:0,0");	

	snd_pcm_open(&pcm_handle, alsa_dev, SND_PCM_STREAM_PLAYBACK, 0);
	
	snd_pcm_hw_params_malloc(&hwparams);

	int periodsize = 512;
	int periods = 2;
	if (snd_pcm_hw_params_any(pcm_handle, hwparams) < 0) {
         	fprintf(stderr, "Can not configure this PCM device.\n");
		return(-1);
	}

	if (snd_pcm_hw_params_set_access(pcm_handle, hwparams, SND_PCM_ACCESS_RW_INTERLEAVED) < 0) {
      		fprintf(stderr, "Error setting access.\n");
      		return(-1);
    	}	

   	if (snd_pcm_hw_params_set_format(pcm_handle, hwparams, SND_PCM_FORMAT_S16_LE) < 0) {
      		fprintf(stderr, "Error setting format.\n");
      		return(-1);
    	}

	int exact_rate = SOUND_SAMPLERATE;
	if (snd_pcm_hw_params_set_rate_near(pcm_handle, hwparams, &exact_rate, 0) < 0) {
		fprintf(stderr, "Error setting rate.\n");
		return(-1);
	}
	if (SOUND_SAMPLERATE != exact_rate) {
		fprintf(stderr, "The rate %d Hz is not supported by your hardware. Using %d Hz instead.\n",
			SOUND_SAMPLERATE, exact_rate);
	}

	if (snd_pcm_hw_params_set_channels(pcm_handle, hwparams, 2) < 0) {
		fprintf(stderr, "Error setting channels.\n");
		return(-1);
	}

	if (snd_pcm_hw_params_set_periods(pcm_handle, hwparams, periods , 0) < 0) {
		fprintf(stderr, "Error setting periods.\n");
		return(-1);
	}

	if (snd_pcm_hw_params_set_buffer_size(pcm_handle, hwparams, (periodsize * periods)>>2) < 0) {
		fprintf(stderr, "Error setting buffersize.\n");
		return(-1);
	}

	if (snd_pcm_hw_params(pcm_handle, hwparams) < 0) {
		fprintf(stderr, "Error setting HW params.\n");
		return(-1);
	}
}

`
It's failing on snd_pcm_hw_params_set_buffer_size() with a buffersize of 1024 bytes (512 bytes periodsize * 2 periods),while it works fine on my Ubuntu X86 workstation.
I am using the stock Raspbian kernel. I could build my own kernel if it would help. 
Any idea on how to overcome this? Or is this a hardware limit?

thanks

pelwell, 2016-01-06:
The failure is due to [this](https://github.com/raspberrypi/linux/blob/rpi-4.1.y/sound/arm/bcm2835-pcm.c#L50) line:
```
	.period_bytes_min =   1 * 1024,
```
Your 1K buffer split between two periods violates this limit.

Given the way the onboard audio is handled, 6ms delay sounds ambitious. You'd have more luck with an [I2S audio](https://github.com/raspberrypi/linux/blob/rpi-4.1.y/sound/soc/bcm/bcm2708-i2s.c#L883) device:
```
	.period_bytes_min	= 32,
```

vanfanel, 2016-01-07:
@pelwell: you are right, using only a 1024 bytes period I can configure a 1024 bytes buffer.

However, is that limit also aplicable to HDMI audio output? It would make no much sense if I understand something about how HDMI works...

pelwell, 2016-01-07:
Yes, the same restriction applies. All "internal" audio (to be output via the 3.5mm jack or HDMI) is sent over a comms channel to the VPU which performs efficient, high quality sample rate conversion (and stereo mixdown where needed) before outputting it to the appropriate connector.

vanfanel, 2016-01-07:
Then my last two questions on this:
-What would happen if I removed that limit in bcm2835-pcm.c?
-Since (rate? format?) conversion is involved, should that limit also be applied if I use hw:0,0 or hw:0,1 instead of plughw?

pelwell, 2016-01-07:
Answers:
1) If you did remove/lower the limit (be sure to change line 33 as well) it would allow you to set a smaller period, and a smaller period will certainly increase CPU load by some amount. If the period is sufficiently small you may find that the audio distorts due to underflow, but I don't know what that limit would be.
2) You've reached the limits of my ALSA knowledge, but the limit should apply to all sound output over HDMI or the audio jack, whichever route it takes within ALSA.

vanfanel, 2016-01-07:
Ok, thanks a lot, pelwell!
This is all I neded to know :+1:

vanfanel, 2016-01-07:
@pelwell : sorry, another question comes to mind, and I'm sure you can give me a good answer. 
Isn't the delay with a 1024 bytes buffersize and a 44100 Hz, 16 bit, stereo stream ~6ms?
If that's the case, what's the downside of having only 1 period instead of 2 periods of 512 bytes? I know it's a general ALSA question and not a Raspberry Pi specific one, but finding reliable information on these topics is very hard.

pelwell, 2016-01-07:
I'm guessing (this is all supposition) that the intention of the limit is to keep the granularity reasonable - we don't want to have to deal with audio a few samples at a time. Having only one period means less tolerance to jitter, and probably lower efficiency due to the lack of buffering (it must be more efficient for an mp3 player to be able to process audio in larger chunks).

closed 2016-01-16 by vanfanel