raspberrypi/linux #1253: crash on 4.1.15

PeterPablo, 2016-01-07:
I am running on a Raspberry Pi B2 with Raspbian Jessie and played a movie on kodi 15.2 an got a crash.

```
uname -a
Linux rpi2 4.1.15-v7+ #830 SMP Tue Dec 15 17:02:45 GMT 2015 armv7l GNU/Linux
```

```
grep -v '^#' /boot/config.txt 
overscan_top=-16
overscan_bottom=-16
hdmi_group=1
hdmi_mode=31
arm_freq=1000
decode_MPG2=0xbla
decode_WVC1=0xbla
dtoverlay=hifiberry-dacplus
gpu_mem=160
max_usb_current=1
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
```

```
lsusb
Bus 001 Device 008: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 007: ID 0a5c:200a Broadcom Corp. BCM2035 Bluetooth dongle
Bus 001 Device 006: ID 05e3:0605 Genesys Logic, Inc. USB 2.0 Hub
Bus 001 Device 005: ID 0480:a00d Toshiba America Info. Systems, Inc. 
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

syslog:
```
Jan  7 18:32:25 rpi2 wpa_supplicant[458]: wlan0: WPA: Group rekeying completed with e8:37:7a:2d:6f:50 [GTK=CCMP]
Jan  7 18:39:01 rpi2 CRON[26821]: (root) CMD (  [ -x /usr/lib/php5/sessionclean ] && /usr/lib/php5/sessionclean)
Jan  7 18:54:10 rpi2 kernel: [188400.707948] Unable to handle kernel paging request at virtual address fffffd48
Jan  7 18:54:10 rpi2 kernel: [188400.721828] pgd = 80004000
Jan  7 18:54:10 rpi2 kernel: [188400.730967] [fffffd48] *pgd=2e7fa821, *pte=00000000, *ppte=00000000
Jan  7 18:54:10 rpi2 kernel: [188400.741281] Internal error: Oops: 17 [#1] SMP ARM
Jan  7 18:54:10 rpi2 kernel: [188400.749824] Modules linked in: rfcomm hidp bnep cfg80211 snd_soc_pcm512x_i2c snd_soc_pcm512x regmap_i2c btusb btbcm btintel bluetooth r8188eu(C) rfkill evdev sg snd_soc_hifiberry_dacplus snd_soc_bcm2708_i2s snd_soc_hifiberry_dac regmap_mmio i2c_bcm2708 snd_soc_core bcm2835_gpiomem snd_compress snd_pcm_dmaengine clk_hifiberry_dacpro snd_pcm snd_timer snd uio_pdrv_genirq uio joydev uinput fuse ipv6
Jan  7 18:54:10 rpi2 kernel: [188400.749917] CPU: 2 PID: 26 Comm: khungtaskd Tainted: G         C      4.1.15-v7+ #830
Jan  7 18:54:10 rpi2 kernel: [188400.749923] Hardware name: BCM2709
Jan  7 18:54:10 rpi2 kernel: [188400.749929] task: add5a280 ti: ade48000 task.ti: ade48000
Jan  7 18:54:10 rpi2 kernel: [188400.749945] PC is at watchdog+0x114/0x350
Jan  7 18:54:10 rpi2 kernel: [188400.749950] LR is at watchdog+0x94/0x350
Jan  7 18:54:10 rpi2 kernel: [188400.749958] pc : [<800b2980>]    lr : [<800b2900>]    psr: 20000013
Jan  7 18:54:10 rpi2 kernel: [188400.749958] sp : ade49f10  ip : 00007f51  fp : ade49f5c
Jan  7 18:54:10 rpi2 kernel: [188400.749961] r10: 00000000  r9 : 808e6ba0  r8 : 0000034e
Jan  7 18:54:10 rpi2 kernel: [188400.749965] r7 : a9795238  r6 : fffffd48  r5 : 807ff66c  r4 : 00007f4e
Jan  7 18:54:10 rpi2 kernel: [188400.749969] r3 : a8ae2ecc  r2 : 00000000  r1 : a8ad0b80  r0 : 00000000
Jan  7 18:54:10 rpi2 kernel: [188400.749975] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Jan  7 18:54:10 rpi2 kernel: [188400.749980] Control: 10c5387d  Table: 284b406a  DAC: 00000015
Jan  7 18:54:10 rpi2 kernel: [188400.749984] Process khungtaskd (pid: 26, stack limit = 0xade48210)
Jan  7 18:54:10 rpi2 kernel: [188400.749990] Stack: (0xade49f10 to 0xade4a000)
Jan  7 18:54:10 rpi2 kernel: [188400.749996] 9f00:                                     a0000113 2d94a000 00008000 00000078
Jan  7 18:54:10 rpi2 kernel: [188400.750004] 9f20: 808e6ba0 a9795080 800b286c 00000000 00000000 ade1aa40 00000000 00000000
Jan  7 18:54:10 rpi2 kernel: [188400.750012] 9f40: 800b286c 00000000 00000000 00000000 ade49fac ade49f60 800426d0 800b2878
Jan  7 18:54:10 rpi2 kernel: [188400.750020] 9f60: ade49f94 00000000 800492e0 00000000 00000000 00000000 ade49f78 ade49f78
Jan  7 18:54:10 rpi2 kernel: [188400.750028] 9f80: 00000000 00000000 ade49f88 ade49f88 ade1aa40 800425e8 00000000 00000000
Jan  7 18:54:10 rpi2 kernel: [188400.750035] 9fa0: 00000000 ade49fb0 8000f858 800425f4 00000000 00000000 00000000 00000000
[... 128 lines cut ...]
Jan  7 19:19:30 rpi2 kernel: [189919.287683] [<8043c9dc>] (arch_timer_handler_phys) from [<8006f614>] (handle_percpu_devid_irq+0x7c/0x114)
Jan  7 19:19:30 rpi2 kernel: [189919.333345] [<8006f614>] (handle_percpu_devid_irq) from [<8006b490>] (__handle_domain_irq+0x84/0xec)
Jan  7 19:19:30 rpi2 kernel: [189919.378590] [<8006b490>] (__handle_domain_irq) from [<8000933c>] (asm_do_IRQ+0x2c/0x30)
Jan  7 19:19:30 rpi2 kernel: [189919.421628] [<8000933c>] (asm_do_IRQ) from [<80558334>] (__irq_svc+0x34/0x12c)
Jan  7 19:19:30 rpi2 kernel: [189919.465369] Exception stack(0xab475cd8 to 0xab475d20)
Jan  7 19:19:30 rpi2 kernel: [189919.501957] 5cc0:                                                       a8ae2064 00000000
Jan  7 19:19:30 rpi2 kernel: [189919.542622] 5ce0: 00000000 0000000b a8ae1e40 00000000 a8ae1f80 9c862220 ad9e7820 00000000
Jan  7 19:19:30 rpi2 kernel: [189919.584667] 5d00: 00000000 ab475d2c ab475d30 ab475d20 80515478 805579f0 20000113 ffffffff
Jan  7 19:19:30 rpi2 kernel: [189919.625302] [<80558334>] (__irq_svc) from [<805579f0>] (_raw_spin_lock+0x40/0x54)
Jan  7 19:19:30 rpi2 kernel: [189919.664182] [<805579f0>] (_raw_spin_lock) from [<80515478>] (unix_release_sock+0x5c/0x298)
Jan  7 19:19:30 rpi2 kernel: [189919.703927] [<80515478>] (unix_release_sock) from [<805156e0>] (unix_release+0x2c/0x3c)
Jan  7 19:19:30 rpi2 kernel: [189919.748286] [<805156e0>] (unix_release) from [<8045e818>] (sock_release+0x30/0xc8)
Jan  7 19:19:30 rpi2 kernel: [189919.787165] [<8045e818>] (sock_release) from [<8045e8cc>] (sock_close+0x1c/0x24)
Jan  7 19:19:30 rpi2 kernel: [189919.825623] [<8045e8cc>] (sock_close) from [<80143188>] (__fput+0x98/0x1dc)
Jan  7 19:19:30 rpi2 kernel: [189919.865936] [<80143188>] (__fput) from [<80143344>] (____fput+0x18/0x1c)
Jan  7 19:19:30 rpi2 kernel: [189919.908222] [<80143344>] (____fput) from [<80040d20>] (task_work_run+0xc0/0xf0)
Jan  7 19:19:30 rpi2 kernel: [189919.950994] [<80040d20>] (task_work_run) from [<80028d0c>] (do_exit+0x334/0xa48)
Jan  7 19:19:30 rpi2 kernel: [189919.990025] [<80028d0c>] (do_exit) from [<800294b8>] (do_group_exit+0x4c/0xe4)
Jan  7 19:19:30 rpi2 kernel: [189920.032467] [<800294b8>] (do_group_exit) from [<80033f6c>] (get_signal+0x1ac/0x6c0)
Jan  7 19:19:30 rpi2 kernel: [189920.075248] [<80033f6c>] (get_signal) from [<80012f1c>] (do_signal+0x98/0x3a0)
Jan  7 19:19:30 rpi2 kernel: [189920.113770] [<80012f1c>] (do_signal) from [<8001340c>] (do_work_pending+0xb8/0xc8)
Jan  7 19:19:30 rpi2 kernel: [189920.152756] [<8001340c>] (do_work_pending) from [<8000f7e4>] (work_pending+0xc/0x20)
```

Googling didn't quite lead me to an answer. Admittedly it might be due to insufficient power supply. Nonethelesss I wanted to create a report. Feel free to close if this looks suspiciously related to a weak power supply.

popcornmix, 2016-01-08:
Disable overclock and try again. Try with a different power supply. They are the most likely explanations.

popcornmix, 2016-01-08:
Note: sdram_freq=500 is the most likely cause of crashing.

PeterPablo, 2016-01-09:
Thank you for having a look at this! I have disabled most of the overclocking settings (only left `arm_freq=1000` and `overvoltage=2` set) and haven't experienced another crash yet. Will close this issue report.

closed 2016-01-09 by PeterPablo

PeterPablo, 2016-01-20:
Please let me give you a brief update: My raspberry pi 2 b runs stable using:
```
overscan_top=-16
overscan_bottom=-16
hdmi_group=1
hdmi_mode=31
decode_MPG2=---
decode_WVC1=---
dtoverlay=hifiberry-dacplus
max_usb_current=1
gpu_mem=160
arm_freq=1000
over_voltage=2
overscan_scale=1
```
I want to emphasize that `sdram_freq=500` was added by setting the overclocking using the **Pi2** preset of the `raspi-config` tool. May be this preset should be adapted to a more conservative default?!

popcornmix, 2016-01-21:
Overclock by definition is running the chip above the recommend frequencies. There is no guarantee that any of the frequencies will be stable.
However the sdram overclock is the one that fails most often so it may be worth lowering it.