raspberrypi/linux #1254: Filename "aux" illegal in Windows. Possibly prevents cloning of repository.

JimHawking, 2016-01-10:
In linux/drivers/gpu/drm/nouveau/nvkm/subdev/i2c there is a file named as aux.c
When cloning a repository on a windows computer (to be cross-compiled), the file cannot be created as in windows, aux is reserved (for an i/o port?). The failure to create the file causes an error message as a minimum and may have caused or contributed to the cloning being subsequently reported as having failed. 
In reality, the local repository is mostly created but the checkout that is part of the cloning fails. Subsequently checking out an earlier branch reports the same file problem but does complete so the earlier checkout problem may not be related to the file name. The aux.c file is still not created.

pelwell, 2016-01-10:
That's not the only problem - there are files which differ only by case, such doesn't work well on a case-insensitive file system.

Anyway, this is a problem with the upstream Linux kernel, not our use of it our modifications to it. You could try raising the matter with the kernel maintainers, but I think you would be wasting your time.

I suggest you install a proper Linux distribution within VirtualBox (or similar) - as well as solving the naming issues you'll probably find the performance is better.

popcornmix, 2016-01-10:
You are unlikely to succeed in building a linux kernel on windows.
As @pelwell says, install Ubuntu (or similar) image in a VirtualBox inside windows and you'll have a lot more success.

closed 2016-01-10 by popcornmix

Ferroin, 2016-01-11:
Yes, AUX is a reserved filename on windows originally used for an IO port (it's a holdover from DOS, and part of the reason I hate windows).  As pelwell mentioned, there are also issues with case insensitivity (which is another reason I hate windows...), and as popcornmix mentioned, you can't build Linux on Windows (it might work in Cygwin, but it's not likely).

rustyx, 2018-12-14:
Another possibility is [WSL](https://docs.microsoft.com/en-us/windows/wsl/about), open `bash` then retry the checkout there:

```
# cd /mnt/c/Users/my-user/my-path-to-ubuntu-kernel-that-failed-checkout
# git checkout -f master
```

It will have no problem with those files:

[image]

huangzonghao, 2020-05-01:
Interestingly I found the name is not only reserved in Windows, but also the ntfs file system. I  was trying to move my repo from linux system drive to a ntfs storage drive in Ubuntu, but the command could not be done due to the naming issues.

> error: unable to create file drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c: Invalid argument

Wondering if it is the ntfs-3g that bounced this file name back.