raspberrypi/linux #1256: kernel oops 817 working with usb

murzz, 2016-01-12:
Custom software is working heavily with devices connected to RP via USB when kernel oops occured:
```
[ 2117.318519] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[ 2117.331580] pgd = ccb60000
[ 2117.336659] [00000034] *pgd=0cb66831, *pte=00000000, *ppte=00000000
[ 2117.345365] Internal error: Oops: 817 [#1] PREEMPT ARM

Entering kdb (current=0xcc872520, pid 2831) Oops: (null)
due to oops @ 0xc02e5714

dCPU: 0 PID: 2831 Comm: bms_router Not tainted 3.10.25+ #622
dtask: cc872520 ti: cd852000 task.ti: cd852000
PC is at assign_and_init_hc+0x50/0x55c
LR is at dwc_otg_hcd_select_transactions+0x1c8/0x35c
pc : [<c02e5714>] lr : [<c02e8078>] psr: 00000193
sp : cd853c70 ip : 00000006 fp : cd9b4628
r10: cd95a814 r9 : cd95a800 r8 : ccbad3c0
r7 : cd95a800 r6 : cd9b4600 r5 : ccbad080 r4 : cd996120
r3 : cd95a860 r2 : 00000000 r1 : 00000000 r0 : cd95a800
Flags: nzcv IRQs off FIQs on Mode SVC_32 ISA ARM Segment user
Control: 00c5387d Table: 0cb60008 DAC: 00000015
dCPU: 0 PID: 2831 Comm: bms_router Not tainted 3.10.25+ #622
[<c0013a18>] (unwind_backtrace+0x0/0xf0) from [<c0010d7c>] (show_stack+0x10/0x14)
[<c0010d7c>] (show_stack+0x10/0x14) from [<c0076980>] (kdb_dumpregs+0x28/0x50)
[<c0076980>] (kdb_dumpregs+0x28/0x50) from [<c0078bf0>] (kdb_main_loop+0x3c0/0x6c0)
[<c0078bf0>] (kdb_main_loop+0x3c0/0x6c0) from [<c007b2f0>] (kdb_stub+0x154/0x380)
```

Complete log is attached:
[putty.log.txt](https://github.com/raspberrypi/linux/files/88024/putty.log.txt)

jkuek, 2016-02-05:
```
[  398.671141] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[  398.696526] pgd = c8ec4000
[  398.716195] [00000034] *pgd=08c52831, *pte=00000000, *ppte=00000000
[  398.739833] Internal error: Oops: 817 [#1] PREEMPT ARM
[  398.762504] Modules linked in: cfg80211 rfkill joydev snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device sierra_net sierra usbserial bcm2835_gpiomem evdev snd_bcm2835 snd_pcm snd_timer snd uio_pdrv_genirq uio ipv6
[  398.820799] CPU: 0 PID: 1722 Comm: testapp Not tainted 4.1.13+ #826
[  398.847146] Hardware name: BCM2708
[  398.870037] task: c8f16040 ti: ca5a8000 task.ti: ca5a8000
[  398.895453] PC is at assign_and_init_hc+0x60/0x670
[  398.920090] LR is at dwc_otg_hcd_select_transactions+0x274/0x3a8
[  398.946452] pc : [<c0417f34>]    lr : [<c041a254>]    psr: 00000193
[  398.946452] sp : ca5a9ac0  ip : ca5a9af8  fp : ca5a9af4
[  398.998944] r10: cb306800  r9 : c8e80ec8  r8 : ca597c80
[  399.024617] r7 : cb306800  r6 : c8e80ea0  r5 : ca597940  r4 : cb3020c0
[  399.051600] r3 : cb306860  r2 : 00000000  r1 : 00000000  r0 : cb306800
[  399.078541] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  399.105644] Control: 00c5387d  Table: 08ec4008  DAC: 00000015
[  399.131470] Process testapp (pid: 1722, stack limit = 0xca5a8188)
[  399.158443] Stack: (0xca5a9ac0 to 0xca5aa000)
[  399.183113] 9ac0: 20000193 ca5a9b0c c087f842 c8e80ea0 00000010 ca5a9b0c c087f842 c8e80ea0
[  399.212159] 9ae0: cb306814 c8e80ec8 ca5a9b3c ca5a9af8 c041a254 c0417ee0 00000008 cb306838
[  399.240923] 9b00: cb30681c cb306860 00000007 20000193 cb302060 cb306800 cb306828 cb306830
[  399.269614] 9b20: 02000008 c087f670 f301080e c0892d04 ca5a9b5c ca5a9b40 c041ddd8 c0419fec
[  399.298546] 9b40: 1a9b0158 cb306800 cb306400 00000001 ca5a9b84 ca5a9b60 c04201b0 c041dd14
[  399.327929] 9b60: cb237d40 cb237b40 00000020 00000000 00000000 00000000 ca5a9b94 ca5a9b88
[  399.357034] 9b80: c041c644 c041fedc ca5a9ba4 ca5a9b98 c03eed04 c041c634 ca5a9be4 ca5a9ba8
[  399.386121] 9ba0: c006186c c03eecdc c041fe8c c041f6fc cb306800 c082dc64 00000001 c082dc64
[  399.415306] 9bc0: cb237b40 00000000 00000000 00000000 00000000 c8ee7148 ca5a9bfc ca5a9be8
[  399.445266] 9be0: c0061b08 c0061834 c082dc64 c0871eb0 ca5a9c14 ca5a9c00 c0064c24 c0061aa4
[  399.475117] 9c00: c0064b7c 00000020 ca5a9c3c ca5a9c18 c0061138 c0064b88 ca5a9c50 c001adc0
[  399.505019] 9c20: 60000013 f200b200 ca5a9c84 fffffffe ca5a9c4c ca5a9c40 c00093a4 c00610c8
[  399.535041] 9c40: ca5a9cbc ca5a9c50 c058d538 c0009384 00000000 00000fff 00000002 00064bcf
[  399.565422] 9c60: 00001000 00000000 cb7d5f3c fffffffe fffffffe 00000000 c8ee7148 ca5a9cbc
[  399.595923] 9c80: bfff4ce9 ca5a9c98 c001ac74 c001adc0 60000013 ffffffff c001eb18 fffffffe
[  399.626048] 9ca0: ca5a9cbc ca667f80 c001bce0 00000000 ca5a9ccc ca5a9cc0 c001bd3c c001ad90
[  399.656113] 9cc0: ca5a9cf4 ca5a9cd0 c03ef6a8 c001bcec 00000000 c8e80ea0 ca667f80 cb2c6600
[  399.686382] 9ce0: 00000000 fffffffe ca5a9d04 ca5a9cf8 c03ef7b0 c03ef63c ca5a9d24 ca5a9d08
[  399.716690] 9d00: c03ef7fc c03ef78c ca667f80 cb2c6600 00000000 cb306800 ca5a9d44 ca5a9d28
[  399.747231] 9d20: c03ef950 c03ef7c0 ca5a8000 ca667f80 cb2c6600 00000000 ca5a9d74 ca5a9d48
[  399.778006] 9d40: c041c264 c03ef914 c0061140 a0000013 ca5a9d88 ca667f80 cb2c6600 60000013
[  399.808724] 9d60: fffffffe fffffffe ca5a9da4 ca5a9d78 c03f04ac c041c1d8 c038b364 c0309b04
[  399.839529] 9d80: ca5a9da4 00000000 ca667f80 60000013 c8f07000 fffffffe ca5a9dcc ca5a9da8
[  399.870425] 9da0: c03f17fc c03f0478 00000001 ca667f80 ca667f8c c8c95280 c8f5e800 c8f5e800
[  399.901382] 9dc0: ca5a9e04 ca5a9dd0 c03f2c1c c03f1740 c03f327c c012e6c4 c8c95280 00000000
[  399.932547] 9de0: 60000013 00000000 ca5a8000 00000005 c8c95294 c8c95280 ca5a9e24 ca5a9e08
[  399.964232] 9e00: bf119b00 c03f2bc8 ca5a8000 00000000 c8c95280 c8c9529c ca5a9e4c ca5a9e28
[  399.996090] 9e20: bf119c3c bf119ad4 bf119b20 c8f5e804 c8f5e884 ca532200 c8ee7140 00000008
[  400.028138] 9e40: ca5a9e5c ca5a9e50 bf0e1034 bf119b2c ca5a9e7c ca5a9e60 c035ea14 bf0e100c
[  400.060297] 9e60: ffffffff ca532200 c8f5e804 00000000 ca5a9e94 ca5a9e80 c035f910 c035e990
[  400.092566] 9e80: ca532200 c7c107f8 ca5a9ea4 ca5a9e98 bf0e1a48 c035f8e8 ca5a9f1c ca5a9ea8
[  400.124503] 9ea0: c0355220 bf0e1a28 c0354f40 00000003 c8ee7140 b36feceb c8ee7140 ca5a9f78
[  400.156137] 9ec0: 00000001 b36feceb ca5a9f44 c013cff8 c013c6b8 c8ee7148 00000001 00000009
[  400.187913] 9ee0: c8ca4e38 00000008 00000000 00000000 ca5a9f7c c8ee7140 c7c107f8 cb2c9a90
[  400.219915] 9f00: c8ca4e38 00000008 00000000 c8ee7148 ca5a9f54 ca5a9f20 c013e364 c035511c
[  400.251751] 9f20: 00000000 00000000 ca5a9f7c c8f164fc c8f16040 c0899434 00000000 c000f908
[  400.283761] 9f40: ca5a8000 00000000 ca5a9f64 ca5a9f58 c013e548 c013e2e0 ca5a9f84 ca5a9f68
[  400.315976] 9f60: c003fb14 c013e53c ca5a8000 c000f908 ca5a9fb0 000000a8 ca5a9fac ca5a9f88
[  400.348267] 9f80: c0013414 c003fa84 c000f70c 00000004 00000001 00000000 00000000 b48feb94
[  400.380568] 9fa0: 00000000 ca5a9fb0 c000f764 c0013360 00000001 00000001 000003e8 000e850c
[  400.413017] 9fc0: 00000000 00000000 b48feb94 000000a8 0141faa4 b48ff8f0 00000000 b36fee14
[  400.445425] 9fe0: 00000000 b36fecf0 b6a2f934 b6a18734 80000010 000e850c 00000000 00000000
[  400.477736] [<c0417f34>] (assign_and_init_hc) from [<c041a254>] (dwc_otg_hcd_select_transactions+0x274/0x3a8)
[... cut ...]
[  401.593407] [<c013e548>] (____fput) from [<c003fb14>] (task_work_run+0x9c/0xcc)
[  401.626598] [<c003fb14>] (task_work_run) from [<c0013414>] (do_work_pending+0xc0/0xe0)
[  401.660454] [<c0013414>] (do_work_pending) from [<c000f764>] (work_pending+0xc/0x20)
[  401.694186] Code: e1520003 05942034 15941034 05872064 (15821034)
[  401.726353] ---[ end trace d0bda5d63c09711d ]---
[  401.756843] Kernel panic - not syncing: Fatal exception in interrupt
[  401.788985] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
```

luftronix, 2016-02-17:
The same thing with ODROID XU4, kernel 3.10 - so it's not RPi-specific

Ruffio, 2016-08-17:
@murzz has your issue been resolved? If so, please close this issue. Thanks.

yarmooh, 2016-09-05:
I've got similar issue at 4.4.19 kernel: 

`Sep  3 04:35:13 smseagle kernel: [44660.188699] Unable to handle kernel NULL pointer dereference at virtual address 00000034
Sep  3 04:35:13 smseagle kernel: [44660.202389] pgd = d91dc000
Sep  3 04:35:13 smseagle kernel: [44660.207690] [00000034] *pgd=192f2831, *pte=00000000, *ppte=00000000
Sep  3 04:35:13 smseagle kernel: [44660.216667] Internal error: Oops: 817 [#1] ARM
Sep  3 04:35:13 smseagle kernel: [44660.223687] Modules linked in: i2c_dev snd_bcm2835 snd_pcm snd_timer snd rtc_pcf8563 i2c_bcm2708 bcm2835_wdt dht11 industrialio uio_pdrv_genirq uio cdc_acm
Sep  3 04:35:13 smseagle kernel: [44660.243246] CPU: 0 PID: 11335 Comm: gammu-smsd Not tainted 4.4.19+ #906
Sep  3 04:35:13 smseagle kernel: [44660.252553] Hardware name: BCM2708
Sep  3 04:35:13 smseagle kernel: [44660.258551] task: d9283680 ti: dadcc000 task.ti: dadcc000
Sep  3 04:35:13 smseagle kernel: [44660.266721] PC is at assign_and_init_hc+0x60/0x648
Sep  3 04:35:13 smseagle kernel: [44660.274380] LR is at dwc_otg_hcd_select_transactions+0x26c/0x3a0
Sep  3 04:35:13 smseagle kernel: [44660.283388] pc : [<c03fb768>]    lr : [<c03fdb28>]    psr: 00000193
Sep  3 04:35:13 smseagle kernel: [44660.283388] sp : dadcda38  ip : dadcda70  fp : dadcda6c
Sep  3 04:35:13 smseagle kernel: [44660.301176] r10: dafae400  r9 : da7d03e8  r8 : ce2c2e80
Sep  3 04:35:13 smseagle kernel: [44660.309558] r7 : dafae400  r6 : da7d03c0  r5 : ce2c2bc0  r4 : daf2fba0
Sep  3 04:35:13 smseagle kernel: [44660.319244] r3 : dafae460  r2 : 00000000  r1 : 00000000  r0 : dafae400
Sep  3 04:35:13 smseagle kernel: [44660.328909] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Sep  3 04:35:13 smseagle kernel: [44660.342416] Control: 00c5387d  Table: 191dc008  DAC: 00000055
Sep  3 04:35:13 smseagle kernel: [44660.351452] Process gammu-smsd (pid: 11335, stack limit = 0xdadcc188)
Sep  3 04:35:13 smseagle kernel: [44660.361258] Stack: (0xdadcda38 to 0xdadce000)
Sep  3 04:35:13 smseagle kernel: [44660.368964] da20:                                                       dadcda84 c08708de
Sep  3 04:35:13 smseagle kernel: [44660.383822] da40: da7d03c0 dafae414 00000001 dadcda84 c08708de da7d03c0 dafae414 da7d03e8
Sep  3 04:35:13 smseagle kernel: [44660.398664] da60: dadcdab4 dadcda70 c03fdb28 c03fb714 00000008 dafae438 dafae41c dafae460
Sep  3 04:35:13 smseagle kernel: [44660.413766] da80: 00000001 20000193 daf2f9c0 dafae400 dafae428 dafae430 02000008 c0870708
Sep  3 04:35:13 smseagle kernel: [44660.428994] daa0: f301080e 00000008 dadcdad4 dadcdab8 c0401874 c03fd8c8 00000000 dafae400
Sep  3 04:35:13 smseagle kernel: [44660.444251] dac0: dafae000 00000008 dadcdb04 dadcdad8 c0403ba8 c04017b0 c03fff80 dac1ec00
Sep  3 04:35:13 smseagle kernel: [44660.459642] dae0: daebcb40 00000038 00000000 00000000 00000000 dac1ec00 dadcdb14 dadcdb08
Sep  3 04:35:13 smseagle kernel: [44660.475284] db00: c03fff9c c04038c0 dadcdb24 dadcdb18 c03d1018 c03fff8c dadcdb5c dadcdb28
Sep  3 04:35:13 smseagle kernel: [44660.491101] db20: c005aae4 c03d0ff0 dadcdb4c dafae41c da603aa8 dac1ec00 dac1ec10 00000000
Sep  3 04:35:13 smseagle kernel: [44660.507244] db40: 00000000 00000000 024000c2 da559000 dadcdb74 dadcdb60 c005ac78 c005aa5c
Sep  3 04:35:13 smseagle kernel: [44660.523706] db60: 00020000 dac1ec00 dadcdb8c dadcdb78 c005da04 c005ac4c 00000038 c0862448
Sep  3 04:35:13 smseagle kernel: [44660.540229] db80: dadcdb9c dadcdb90 c005a158 c005d974 dadcdbc4 dadcdba0 c005a3f0 c005a134
Sep  3 04:35:13 smseagle kernel: [44660.556895] dba0: dadcdbf0 c0821538 dadcdbf0 ffffffff dadcdc24 00020902 dadcdbd4 dadcdbc8
Sep  3 04:35:13 smseagle kernel: [44660.573853] dbc0: c0010748 c005a3a4 dadcdbec dadcdbd8 c000940c c0010728 c011dfa0 80000013
Sep  3 04:35:13 smseagle kernel: [44660.590930] dbe0: dadcdc6c dadcdbf0 c0574164 c00093dc 0000000c 024080c0 ddd51000 00000022
Sep  3 04:35:13 smseagle kernel: [44660.608014] dc00: 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2 da559000 dadcdc6c
Sep  3 04:35:13 smseagle kernel: [44660.625108] dc20: dadcdc70 dadcdc40 c0110a08 c011dfa0 80000013 ffffffff c010fa50 c0110a08
Sep  3 04:35:13 smseagle kernel: [44660.642186] dc40: ffffffff 024000c2 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2
Sep  3 04:35:13 smseagle kernel: [44660.659273] dc60: dadcdcac dadcdc70 c0110a08 c011df94 ff800000 ffffffff 024000c2 c0335080
Sep  3 04:35:13 smseagle kernel: [44660.676363] dc80: dadcdcb0 c0335080 ffffffff d91e2600 00000000 00020902 00000000 da559000
Sep  3 04:35:13 smseagle kernel: [44660.693437] dca0: dadcdcdc dadcdcb0 c0110bd0 c011096c 024000c2 0000024f 00000000 ffffffff
Sep  3 04:35:13 smseagle kernel: [44660.710519] dcc0: c0335080 dadcdcd0 d91e2600 ce2a1560 dadcdcf4 dadcdce0 c0110c1c c0110b88
Sep  3 04:35:13 smseagle kernel: [44660.727597] dce0: ffffffff c0335080 dadcdd14 dadcdcf8 c0335080 c0110bec c0335060 d91e2600
Sep  3 04:35:13 smseagle kernel: [44660.744681] dd00: ce2a1560 d91e2600 dadcdd2c dadcdd18 c03393b0 c033506c 00000000 ce2a1560
Sep  3 04:35:13 smseagle kernel: [44660.761767] dd20: dadcdd4c dadcdd30 c0339c98 c0339358 da7bc00c d91e2600 da559000 0000000b
Sep  3 04:35:13 smseagle kernel: [44660.778847] dd40: dadcdd74 dadcdd50 c0333488 c0339c7c dadcdd74 dadcdd60 ce2623c0 00000100
Sep  3 04:35:13 smseagle kernel: [44660.795925] dd60: 0a60000b da7ae988 dadcddb4 dadcdd78 c0333a4c c03333fc dadcddec c01306b4
[... cut ...]
Sep  3 04:35:23 smseagle kernel: [44672.656539] [<c012c4d0>] (__vfs_write) from [<c012cd30>] (vfs_write+0xa0/0x1a8)
Sep  3 04:35:23 smseagle kernel: [44672.675936] [<c012cd30>] (vfs_write) from [<c012d6a8>] (SyS_write+0x54/0xb0)
Sep  3 04:35:23 smseagle kernel: [44672.689115] [<c012d6a8>] (SyS_write) from [<c000f820>] (ret_fast_syscall+0x0/0x1c)
Sep  3 04:35:23 smseagle kernel: [44672.708234] Code: e1520003 05942034 15941034 05872064 (15821034) 
Sep  3 04:35:23 smseagle kernel: [44672.720067] ---[ end trace cfc66c2617be70ff ]---`

yarmooh, 2016-09-05:
I've got similar issue at 4.4.19 kernel: 
```Sep  3 04:35:13 smseagle kernel: [44660.188699] Unable to handle kernel NULL pointer dereference at virtual address 00000034
Sep  3 04:35:13 smseagle kernel: [44660.202389] pgd = d91dc000
Sep  3 04:35:13 smseagle kernel: [44660.207690] [00000034] *pgd=192f2831, *pte=00000000, *ppte=00000000
Sep  3 04:35:13 smseagle kernel: [44660.216667] Internal error: Oops: 817 [#1] ARM
Sep  3 04:35:13 smseagle kernel: [44660.223687] Modules linked in: i2c_dev snd_bcm2835 snd_pcm snd_timer snd rtc_pcf8563 i2c_bcm2708 bcm2835_wdt dht11 industrialio uio_pdrv_genirq uio cdc_acm
Sep  3 04:35:13 smseagle kernel: [44660.243246] CPU: 0 PID: 11335 Comm: gammu-smsd Not tainted 4.4.19+ #906
Sep  3 04:35:13 smseagle kernel: [44660.252553] Hardware name: BCM2708
Sep  3 04:35:13 smseagle kernel: [44660.258551] task: d9283680 ti: dadcc000 task.ti: dadcc000
Sep  3 04:35:13 smseagle kernel: [44660.266721] PC is at assign_and_init_hc+0x60/0x648
Sep  3 04:35:13 smseagle kernel: [44660.274380] LR is at dwc_otg_hcd_select_transactions+0x26c/0x3a0
Sep  3 04:35:13 smseagle kernel: [44660.283388] pc : [<c03fb768>]    lr : [<c03fdb28>]    psr: 00000193
Sep  3 04:35:13 smseagle kernel: [44660.283388] sp : dadcda38  ip : dadcda70  fp : dadcda6c
Sep  3 04:35:13 smseagle kernel: [44660.301176] r10: dafae400  r9 : da7d03e8  r8 : ce2c2e80
Sep  3 04:35:13 smseagle kernel: [44660.309558] r7 : dafae400  r6 : da7d03c0  r5 : ce2c2bc0  r4 : daf2fba0
Sep  3 04:35:13 smseagle kernel: [44660.319244] r3 : dafae460  r2 : 00000000  r1 : 00000000  r0 : dafae400
Sep  3 04:35:13 smseagle kernel: [44660.328909] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Sep  3 04:35:13 smseagle kernel: [44660.342416] Control: 00c5387d  Table: 191dc008  DAC: 00000055
Sep  3 04:35:13 smseagle kernel: [44660.351452] Process gammu-smsd (pid: 11335, stack limit = 0xdadcc188)
Sep  3 04:35:13 smseagle kernel: [44660.361258] Stack: (0xdadcda38 to 0xdadce000)
Sep  3 04:35:13 smseagle kernel: [44660.368964] da20:                                                       dadcda84 c08708de
Sep  3 04:35:13 smseagle kernel: [44660.383822] da40: da7d03c0 dafae414 00000001 dadcda84 c08708de da7d03c0 dafae414 da7d03e8
Sep  3 04:35:13 smseagle kernel: [44660.398664] da60: dadcdab4 dadcda70 c03fdb28 c03fb714 00000008 dafae438 dafae41c dafae460
Sep  3 04:35:13 smseagle kernel: [44660.413766] da80: 00000001 20000193 daf2f9c0 dafae400 dafae428 dafae430 02000008 c0870708
Sep  3 04:35:13 smseagle kernel: [44660.428994] daa0: f301080e 00000008 dadcdad4 dadcdab8 c0401874 c03fd8c8 00000000 dafae400
Sep  3 04:35:13 smseagle kernel: [44660.444251] dac0: dafae000 00000008 dadcdb04 dadcdad8 c0403ba8 c04017b0 c03fff80 dac1ec00
Sep  3 04:35:13 smseagle kernel: [44660.459642] dae0: daebcb40 00000038 00000000 00000000 00000000 dac1ec00 dadcdb14 dadcdb08
Sep  3 04:35:13 smseagle kernel: [44660.475284] db00: c03fff9c c04038c0 dadcdb24 dadcdb18 c03d1018 c03fff8c dadcdb5c dadcdb28
Sep  3 04:35:13 smseagle kernel: [44660.491101] db20: c005aae4 c03d0ff0 dadcdb4c dafae41c da603aa8 dac1ec00 dac1ec10 00000000
Sep  3 04:35:13 smseagle kernel: [44660.507244] db40: 00000000 00000000 024000c2 da559000 dadcdb74 dadcdb60 c005ac78 c005aa5c
Sep  3 04:35:13 smseagle kernel: [44660.523706] db60: 00020000 dac1ec00 dadcdb8c dadcdb78 c005da04 c005ac4c 00000038 c0862448
Sep  3 04:35:13 smseagle kernel: [44660.540229] db80: dadcdb9c dadcdb90 c005a158 c005d974 dadcdbc4 dadcdba0 c005a3f0 c005a134
Sep  3 04:35:13 smseagle kernel: [44660.556895] dba0: dadcdbf0 c0821538 dadcdbf0 ffffffff dadcdc24 00020902 dadcdbd4 dadcdbc8
Sep  3 04:35:13 smseagle kernel: [44660.573853] dbc0: c0010748 c005a3a4 dadcdbec dadcdbd8 c000940c c0010728 c011dfa0 80000013
Sep  3 04:35:13 smseagle kernel: [44660.590930] dbe0: dadcdc6c dadcdbf0 c0574164 c00093dc 0000000c 024080c0 ddd51000 00000022
Sep  3 04:35:13 smseagle kernel: [44660.608014] dc00: 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2 da559000 dadcdc6c
Sep  3 04:35:13 smseagle kernel: [44660.625108] dc20: dadcdc70 dadcdc40 c0110a08 c011dfa0 80000013 ffffffff c010fa50 c0110a08
Sep  3 04:35:13 smseagle kernel: [44660.642186] dc40: ffffffff 024000c2 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2
Sep  3 04:35:13 smseagle kernel: [44660.659273] dc60: dadcdcac dadcdc70 c0110a08 c011df94 ff800000 ffffffff 024000c2 c0335080
Sep  3 04:35:13 smseagle kernel: [44660.676363] dc80: dadcdcb0 c0335080 ffffffff d91e2600 00000000 00020902 00000000 da559000
Sep  3 04:35:13 smseagle kernel: [44660.693437] dca0: dadcdcdc dadcdcb0 c0110bd0 c011096c 024000c2 0000024f 00000000 ffffffff
Sep  3 04:35:13 smseagle kernel: [44660.710519] dcc0: c0335080 dadcdcd0 d91e2600 ce2a1560 dadcdcf4 dadcdce0 c0110c1c c0110b88
Sep  3 04:35:13 smseagle kernel: [44660.727597] dce0: ffffffff c0335080 dadcdd14 dadcdcf8 c0335080 c0110bec c0335060 d91e2600
Sep  3 04:35:13 smseagle kernel: [44660.744681] dd00: ce2a1560 d91e2600 dadcdd2c dadcdd18 c03393b0 c033506c 00000000 ce2a1560
Sep  3 04:35:13 smseagle kernel: [44660.761767] dd20: dadcdd4c dadcdd30 c0339c98 c0339358 da7bc00c d91e2600 da559000 0000000b
Sep  3 04:35:13 smseagle kernel: [44660.778847] dd40: dadcdd74 dadcdd50 c0333488 c0339c7c dadcdd74 dadcdd60 ce2623c0 00000100
Sep  3 04:35:13 smseagle kernel: [44660.795925] dd60: 0a60000b da7ae988 dadcddb4 dadcdd78 c0333a4c c03333fc dadcddec c01306b4
[... cut ...]
Sep  3 04:35:23 smseagle kernel: [44672.656539] [<c012c4d0>] (__vfs_write) from [<c012cd30>] (vfs_write+0xa0/0x1a8)
Sep  3 04:35:23 smseagle kernel: [44672.675936] [<c012cd30>] (vfs_write) from [<c012d6a8>] (SyS_write+0x54/0xb0)
Sep  3 04:35:23 smseagle kernel: [44672.689115] [<c012d6a8>] (SyS_write) from [<c000f820>] (ret_fast_syscall+0x0/0x1c)
Sep  3 04:35:23 smseagle kernel: [44672.708234] Code: e1520003 05942034 15941034 05872064 (15821034) 
Sep  3 04:35:23 smseagle kernel: [44672.720067] ---[ end trace cfc66c2617be70ff ]---```

yarmooh, 2016-09-05:
Hello,

I've got similar issue at 4.4.19 kernel:

```
Sep  3 04:35:13 smseagle kernel: [44660.188699] Unable to handle kernel NULL pointer dereference at virtual address 00000034
Sep  3 04:35:13 smseagle kernel: [44660.202389] pgd = d91dc000
Sep  3 04:35:13 smseagle kernel: [44660.207690] [00000034] *pgd=192f2831, *pte=00000000, *ppte=00000000
Sep  3 04:35:13 smseagle kernel: [44660.216667] Internal error: Oops: 817 [#1] ARM
Sep  3 04:35:13 smseagle kernel: [44660.223687] Modules linked in: i2c_dev snd_bcm2835 snd_pcm snd_timer snd rtc_pcf8563 i2c_bcm2708 bcm2835_wdt dht11 industrialio uio_pdrv_genirq uio cdc_acm
Sep  3 04:35:13 smseagle kernel: [44660.243246] CPU: 0 PID: 11335 Comm: gammu-smsd Not tainted 4.4.19+ #906
Sep  3 04:35:13 smseagle kernel: [44660.252553] Hardware name: BCM2708
Sep  3 04:35:13 smseagle kernel: [44660.258551] task: d9283680 ti: dadcc000 task.ti: dadcc000
Sep  3 04:35:13 smseagle kernel: [44660.266721] PC is at assign_and_init_hc+0x60/0x648
Sep  3 04:35:13 smseagle kernel: [44660.274380] LR is at dwc_otg_hcd_select_transactions+0x26c/0x3a0
Sep  3 04:35:13 smseagle kernel: [44660.283388] pc : [<c03fb768>]    lr : [<c03fdb28>]    psr: 00000193
Sep  3 04:35:13 smseagle kernel: [44660.283388] sp : dadcda38  ip : dadcda70  fp : dadcda6c
Sep  3 04:35:13 smseagle kernel: [44660.301176] r10: dafae400  r9 : da7d03e8  r8 : ce2c2e80
Sep  3 04:35:13 smseagle kernel: [44660.309558] r7 : dafae400  r6 : da7d03c0  r5 : ce2c2bc0  r4 : daf2fba0
Sep  3 04:35:13 smseagle kernel: [44660.319244] r3 : dafae460  r2 : 00000000  r1 : 00000000  r0 : dafae400
Sep  3 04:35:13 smseagle kernel: [44660.328909] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Sep  3 04:35:13 smseagle kernel: [44660.342416] Control: 00c5387d  Table: 191dc008  DAC: 00000055
Sep  3 04:35:13 smseagle kernel: [44660.351452] Process gammu-smsd (pid: 11335, stack limit = 0xdadcc188)
Sep  3 04:35:13 smseagle kernel: [44660.361258] Stack: (0xdadcda38 to 0xdadce000)
Sep  3 04:35:13 smseagle kernel: [44660.368964] da20:                                                       dadcda84 c08708de
Sep  3 04:35:13 smseagle kernel: [44660.383822] da40: da7d03c0 dafae414 00000001 dadcda84 c08708de da7d03c0 dafae414 da7d03e8
Sep  3 04:35:13 smseagle kernel: [44660.398664] da60: dadcdab4 dadcda70 c03fdb28 c03fb714 00000008 dafae438 dafae41c dafae460
Sep  3 04:35:13 smseagle kernel: [44660.413766] da80: 00000001 20000193 daf2f9c0 dafae400 dafae428 dafae430 02000008 c0870708
Sep  3 04:35:13 smseagle kernel: [44660.428994] daa0: f301080e 00000008 dadcdad4 dadcdab8 c0401874 c03fd8c8 00000000 dafae400
Sep  3 04:35:13 smseagle kernel: [44660.444251] dac0: dafae000 00000008 dadcdb04 dadcdad8 c0403ba8 c04017b0 c03fff80 dac1ec00
Sep  3 04:35:13 smseagle kernel: [44660.459642] dae0: daebcb40 00000038 00000000 00000000 00000000 dac1ec00 dadcdb14 dadcdb08
Sep  3 04:35:13 smseagle kernel: [44660.475284] db00: c03fff9c c04038c0 dadcdb24 dadcdb18 c03d1018 c03fff8c dadcdb5c dadcdb28
Sep  3 04:35:13 smseagle kernel: [44660.491101] db20: c005aae4 c03d0ff0 dadcdb4c dafae41c da603aa8 dac1ec00 dac1ec10 00000000
Sep  3 04:35:13 smseagle kernel: [44660.507244] db40: 00000000 00000000 024000c2 da559000 dadcdb74 dadcdb60 c005ac78 c005aa5c
Sep  3 04:35:13 smseagle kernel: [44660.523706] db60: 00020000 dac1ec00 dadcdb8c dadcdb78 c005da04 c005ac4c 00000038 c0862448
Sep  3 04:35:13 smseagle kernel: [44660.540229] db80: dadcdb9c dadcdb90 c005a158 c005d974 dadcdbc4 dadcdba0 c005a3f0 c005a134
Sep  3 04:35:13 smseagle kernel: [44660.556895] dba0: dadcdbf0 c0821538 dadcdbf0 ffffffff dadcdc24 00020902 dadcdbd4 dadcdbc8
Sep  3 04:35:13 smseagle kernel: [44660.573853] dbc0: c0010748 c005a3a4 dadcdbec dadcdbd8 c000940c c0010728 c011dfa0 80000013
Sep  3 04:35:13 smseagle kernel: [44660.590930] dbe0: dadcdc6c dadcdbf0 c0574164 c00093dc 0000000c 024080c0 ddd51000 00000022
Sep  3 04:35:13 smseagle kernel: [44660.608014] dc00: 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2 da559000 dadcdc6c
Sep  3 04:35:13 smseagle kernel: [44660.625108] dc20: dadcdc70 dadcdc40 c0110a08 c011dfa0 80000013 ffffffff c010fa50 c0110a08
Sep  3 04:35:13 smseagle kernel: [44660.642186] dc40: ffffffff 024000c2 00002274 ce2a17a0 024002c2 ffffffff 00020902 024000c2
Sep  3 04:35:13 smseagle kernel: [44660.659273] dc60: dadcdcac dadcdc70 c0110a08 c011df94 ff800000 ffffffff 024000c2 c0335080
Sep  3 04:35:13 smseagle kernel: [44660.676363] dc80: dadcdcb0 c0335080 ffffffff d91e2600 00000000 00020902 00000000 da559000
Sep  3 04:35:13 smseagle kernel: [44660.693437] dca0: dadcdcdc dadcdcb0 c0110bd0 c011096c 024000c2 0000024f 00000000 ffffffff
Sep  3 04:35:13 smseagle kernel: [44660.710519] dcc0: c0335080 dadcdcd0 d91e2600 ce2a1560 dadcdcf4 dadcdce0 c0110c1c c0110b88
Sep  3 04:35:13 smseagle kernel: [44660.727597] dce0: ffffffff c0335080 dadcdd14 dadcdcf8 c0335080 c0110bec c0335060 d91e2600
Sep  3 04:35:13 smseagle kernel: [44660.744681] dd00: ce2a1560 d91e2600 dadcdd2c dadcdd18 c03393b0 c033506c 00000000 ce2a1560
Sep  3 04:35:13 smseagle kernel: [44660.761767] dd20: dadcdd4c dadcdd30 c0339c98 c0339358 da7bc00c d91e2600 da559000 0000000b
Sep  3 04:35:13 smseagle kernel: [44660.778847] dd40: dadcdd74 dadcdd50 c0333488 c0339c7c dadcdd74 dadcdd60 ce2623c0 00000100
Sep  3 04:35:13 smseagle kernel: [44660.795925] dd60: 0a60000b da7ae988 dadcddb4 dadcdd78 c0333a4c c03333fc dadcddec c01306b4
[... cut ...]
Sep  3 04:35:23 smseagle kernel: [44672.656539] [<c012c4d0>] (__vfs_write) from [<c012cd30>] (vfs_write+0xa0/0x1a8)
Sep  3 04:35:23 smseagle kernel: [44672.675936] [<c012cd30>] (vfs_write) from [<c012d6a8>] (SyS_write+0x54/0xb0)
Sep  3 04:35:23 smseagle kernel: [44672.689115] [<c012d6a8>] (SyS_write) from [<c000f820>] (ret_fast_syscall+0x0/0x1c)
Sep  3 04:35:23 smseagle kernel: [44672.708234] Code: e1520003 05942034 15941034 05872064 (15821034) 
Sep  3 04:35:23 smseagle kernel: [44672.720067] ---[ end trace cfc66c2617be70ff ]---
```

popcornmix, 2016-09-05:
@P33M any ideas?

P33M, 2017-05-04:
Dupe of https://github.com/raspberrypi/linux/issues/830

closed 2017-05-04 by P33M