raspberrypi/linux #1258: Jessie: raspi-config unable to change keyboard layout

astroleo, 2016-01-16:
After installing the Jessie image from raspbian.org, I wanted to change to a German keyboard layout using `sudo raspi-config` > `5 Internationalization Options` > `13 Change Keyboard Layout`. I had a German Macintosh USB keyboard connected to the Raspberry Pi 2.0 Model B. But when choosing this option, nothing happens and I am returned to the main menu of `raspi-config`. After exiting the config, I see these error messages on the console:
```
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
Reloading keymap. This may take a short while
```
I guess this is related to the change to `systemd`in Debian Jessie.

lurch, 2016-01-18:
I suspect https://github.com/RPi-Distro/raspi-config/issues would be a much better place to report this bug.

astroleo, 2016-01-18:
OK, thanks. I have filed the bug report there. You may close the issue here.

closed 2016-01-18 by popcornmix