raspberrypi/linux #1259: rpi-4.4.y: ethernet hw csum failures when receiving ipv6 multicast traffic

Noltari, 2016-01-16:
When adding support for 4.4 on OpenWrt I get the following warnings whenever the RPi receives IPv6 multicast traffic: https://github.com/openwrt-es/openwrt/commit/964b80d4c0f259ef392aa984d69d41559f19efbd
The linux-4.4 patches where extracted from 4.4.y branch on this repo. However, I must say it could still be something OpenWrt specific.

Manually removing lines 1181-1186 seems to fix it (-> disabling hardware checksums):
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/net/usb/smsc95xx.c#L1181

Any ideas?

BTW, here's my repo: https://github.com/openwrt-es/openwrt/commits/brcm2708-next

popcornmix, 2016-01-17:
Do you get the same issue with raspbian and untouched rpi-4.4.y kernel?

Noltari, 2016-01-18:
I will test that as soon as I can and report my results.

Noltari, 2016-01-22:
Hi again,

I didn't have time to do any tests, but doing some researchs it appears that this problem started on linux-4.2 and has been there since that version.
However, it may have been unnoticed since a bridge between different interfaces is required to trigger the error. It doesn't happen on single network interfaces.

Some other reports of this issue:
https://bugzilla.redhat.com/show_bug.cgi?id=1266601
https://discourse.osmc.tv/t/hw-csum-failure/9990/21

popcornmix, 2016-01-22:
Let us know you find any patches to fix it...

Noltari, 2016-01-22:
It appears it was already reported on https://github.com/raspberrypi/linux/issues/1083 but I didn't see it :/

popcornmix, 2016-01-23:
Okay, closing as duplicate of #1083
Please add any useful discoveries to that issue.

closed 2016-01-23 by popcornmix