raspberrypi/linux #1262: how to capture bmp use v4l2

dodoma, 2016-01-19:
I'm using raspberrypi 2 mode b, 

```c
    CLEAR(format);
    format.type = V4L2_BUF_TYPE_VIDEO_CAPTURE;
    if (ioctl(fd, VIDIOC_G_FMT, &format) == -1) {
        err = merr_raise(MERR_ASSERT, "ioctl G_FMT %s", strerror(errno));
        goto error;
    }

    format.fmt.pix.pixelformat = V4L2_PIX_FMT_RGB32;
    if (ioctl(fd, VIDIOC_S_FMT, &format) == -1) {
        err = merr_raise(MERR_ASSERT, "ioctl S_FMT %s", strerror(errno));
        goto error;
    }
```

second ioctl() return ok, but i can't get the bitmap image.
default JPEG format is ok.
Am i wrong somewhere?

```
Driver Info (not using libv4l2):
	Driver name   : bm2835 mmal
	Card type     : mmal service 16.1
	Bus info      : platform:bcm2835-v4l2
	Driver version: 4.1.15
	Capabilities  : 0x85200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x05200005
		Video Capture
		Video Overlay
		Read/Write
		Streaming
		Extended Pix Format
Priority: 2
```

6by9, 2016-01-19:
BMP is an image container format with header that describes the pixel data that follows. JPEG is the same.
RGB32 is a raw pixel format. No headers at all.

V4L2 does not support BMP as a container format. JPEG is the only one. The full list of formats is in https://github.com/raspberrypi/linux/blob/rpi-4.4.y/include/uapi/linux/videodev2.h#L468

If you want a decent viewer for raw RGB or YUV images, then Vooya (http://www.offminor.de/) is my go-to app, and it's free on x86 Linux.

NB That VIDIOC_S_FMT will also do a VIDIOC_TRY_FMT to munge the request into an acceptable format. You really ought to check the resulting format before arbitrarily using it. ```v4l2-ctl -V``` is the quickest and easiest way to do it manually, or you can do it programmatically.
As a simple test, you can use v4l2-ctl to do all the steps in capturing RGB32:
```
v4l2-ctl -v width=1280,height=720,pixelformat=RGB4
v4l2-ctl --stream-mmap=3 --stream-count=10 --stream-to=~/thing.rgb
```
If that doesn't work then report back.

dodoma, 2016-01-20:
Thanks for your reply, I can capture raw pixel format(RGB24) and view it use imagemagick now.

BTW, There is some concepts cfused me:
What is the role for v4l2 and mmal? Is mmal build on top of v4l2?
I need to modprobe bcm2835-v4l2 to use /dev/video0 through v4l2, but mmal's API seems not to be(don't try yet).
seems  mmal and v4l2 seprated, and doese [bcm2835](https://github.com/raspberrypi/linux/tree/rpi-4.1.y/drivers/media/platform/bcm2835) the source code of the bcm2835-v4l2 driver, if it is, where is the mmal's source code?

It takes about 100ms for JPEG capture, and 200ms for raw pixel RGB capture(I need to QBUF, seelct, DQBUF 2 times to get the latest frame, this also confused me).
If I use mmal, should it be faster than v4l2? Which one you sugguest, I want high speed camera still image capture and process(mainly on machine vision)? 
Personaly I prefer v4l2, but seems mmal is the rasperberry pi recommended one.

6by9, 2016-01-20:
MMAL is the API talking to the GPU multimedia services. All the image processing is done on the GPU.
The V4L2 driver is built on top of MMAL to match the Linux kernel standard API.

MMAL code is in https://github.com/raspberrypi/userland/tree/master/interface/mmal, along with the example camera apps in https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/raspicam 
RaspiVidYuv gives you a MMAL example of how to get the raw image data back for doing your own processing on, though the standard version just saves it to disc.

There should be minimal performance difference between MMAL and V4L2. Use whichever you're comfortable with.

JPEG is hardware accelerated from the YUV image out of the ISP (Image Sensor Pipeline). RGB has to be converted from the YUV on the GPU processor - you don't mention resolution, but I wouldn't expect converting even 5MPix to take 100ms. You are moving a lot more data around with RGB32 (20MB vs 1-2MB for JPEG), and that obviously takes some time.

If you want high speed, then consider if you need RGB, or is monochrome sufficient. The Y/luma channel of YUV is the mono version of the image, with the chroma channels giving subsampled colour information. Select format V4L2_PIX_FMT_YUV420 (YU12) and treat it as a width x height 8bpp mono image. Ignore all extra data in each buffer. It'll save a fair amount of processing and memory bandwidth.

Also consider what resolution you need. By default, raw pixel data >720P is treated as stills captures which adds in some extra processing on the GPU. Those modes are limited to about 5fps. You can move that boundary using the max_video_[width|height] parameters of the kernel module (add them to the modprobe line). Set them to 2592x1944 and you can get 5MPix YUV at 15fps, which is the max the sensor can go at. 
Just found the forum thread where we benchmarked raw captures - https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=107571&p=740712#p740800 
> BGR3 - 9 fps
> RGB3 - 11-12 fps
> l420 - 15 fps

You're using RGB32 whilst those are RGB24. The conversion from RGB24 to BGR24 is done as an additional stage, hence the drop in performance. Do you need RGB32? There is no alpha channel data, so you'll save a load of memory traffic by switching to RGB24.

Drop the resolution to VGA and you can go up to 90fps. I'm pretty sure I checked that a while back and did get 89-90fps for YU12. Not sure on results with RGB/BGR

PS You may want to browse the camera forum (https://www.raspberrypi.org/forums/viewforum.php?f=43) rather than ask questions here. There are more people who read those than the check out these issues pages.

closed 2016-01-20 by dodoma