raspberrypi/linux #1263: Module bcm2708_wdog may be duplicate and should be removed

Terminus-IMRC, 2016-01-19:
There are two(!) watchdog drivers which uses the same hardware registers: ```bcm2708_wdog``` and ```bcm2835_wdt```.

```bcm2708_wdog``` looks just a temporary implementation because it processes ioctls and other stuffs only in the code. In contrast, ```bcm2835_wdog``` uses Linux's generic watchdog code to do the stuffs. In addition, ```bcm2835_wdog``` was merged to Linux mainline.

So I think ```bcm2708_wdog``` is duplicate and should be removed from the tree.

pelwell, 2016-01-19:
If you look at the rpi-4.3.y and rpi-4.4.y branches you will see that it has already been removed, and we will be switching to rpi-4.4.y very soon.

closed 2016-01-19 by pelwell

Terminus-IMRC, 2016-01-19:
Thank you for your reply.
I'm really sorry for my lack of research...

pelwell, 2016-01-19:
It's no problem - your suggestion was a good one, but great minds think alike.