raspberrypi/linux #1266: Stuck on rainbow screen with >=4.3.y kernel

SandPox, 2016-01-26:
Hello, I try to run my RPI2 with >=4.3.y kernel and it just stuck on rainbow screen, in UART just have "Uncompressing Linux… OK, booting the kernel" and stuck there forever, I've done the same compiling config with 4.2.y and 4.1.y and it's working just fine, is it a know problem for under development branch or just me? is compiling >=4.3.y kernel require extra configuration or cross compiling version? I use defconfig bcm2709_defconfig with gcc-linaro-arm-linux-gnueabihf-raspbian-x64 cross compiler.

Thanks.

popcornmix, 2016-01-26:
Have you followed https://www.raspberrypi.org/documentation/linux/kernel/building.md
(including the mkknlimg step)?

pelwell, 2016-01-26:
And copying the .dtbs?

SandPox, 2016-01-26:
@popcornmix before I did not have done this step, I just recompile the whole thing and do mkknlimg but still stuck on rainbow screen

@pelwell yes

as i remember, a long time ago i compile 4.3.y and successful boot on my rpi2 but recently it's doesn't boot anymore on branch 4.3.y and newer

pelwell, 2016-01-26:
1) How recent is your firmware?

2) If you have a custom dt-blob.dts, try removing it.

3) If you have a spare card and some time, try installing a clean recent raspbian, using rpi-update to get a new firmware and kernel, and then just adding your new kernel and modules.

SandPox, 2016-01-26:
Issue solved, the problem is from my custom Debian armhf, i don't know why but it just not want to boot debian armhf while raspbian image from rpi homepage boot just fine, do you have any idea how i can diagnose this problem because there is nothing usable in UART at all

and yes, it boot with 4.2.y kernel, i'm use debian sid

closed 2016-01-26 by SandPox