raspberrypi/linux #1272: Kernel panics (USB-triggered?)

jkuek, 2016-01-28:
I'm experiencing an infrequent crash, which has a number of reported errors:

- Internal error: Oops: 5 [#1] ARM
- Kernel Panic - Not syncing: Fatal exception in interrupt
- Internal error: Oops: 805 [#1] ARM

Other common messages seen are 
"Unable to handle kernel paging request at virtual address 0x......"

In normal use, this seems to occur because one or USB devices have spontaneously disconnected and reconnected. The root cause of this is still unknown.

However I am able to make the crash easily occur by running the following script to force USB devices to reconnect:

```
echo 0 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
echo 1 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
```

I'm not running any other applications at the time, and this is at the moment just a fresh install of Raspbian Jessie Lite.

I'm using a RPi B+ , with the following USB connections:
Port 2 - keyboard
Port 4 - Sierra Wireless 3g modem
Port 5 - Audio card

I've attached the output of lsusb -v, [lsusb.txt](https://github.com/raspberrypi/linux/files/107627/lsusb.txt)

I initially thought it might be related to one of the USB devices, but I tried unplugging the modem and audio card but the problem still occurred. Interestingly, I have never seen the crash if I don't have any USB devices connected.

I originally noticed the problem on Raspbian Jessie Lite, kernel 4.1.13+. I also tried running rpi-update to update the kernel to 4.1.16+, but the problem persisted.

I'm not sure how to get the output of the kernel dumps, after the problem occurs, I reset the unit and there's no kernel info in the logs.  Some photos are here: [image]
[image]

What other information can I provide to help resolve the issue?

jkuek, 2016-01-28:
I am able to crash Raspbian Wheezy 3.18.11+ with the same script, although the trace is different

jkuek, 2016-02-02:
I updated to the 4.4.0 development kernel and the problem still occurs. It's worth noting that having multiple USB devices plugged in seems to make it easier to reproduce (i.e. less cycles of USB reset needed to crash), but it can occur with just a keyboard plugged in.

jkuek, 2016-02-03:
Here's the syslog output after one run of my script:
```
echo 0 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
echo 1 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower

```
-------------------------------------------

Feb  3 02:57:17 raspberrypi kernel: [  129.093029] ERROR::dwc_otg_hcd_urb_enqueue:505: Not connected
Feb  3 02:57:17 raspberrypi kernel: [  129.093029] 
Feb  3 02:57:17 raspberrypi avahi-daemon[347]: Interface eth0.IPv6 no longer relevant for mDNS.
Feb  3 02:57:17 raspberrypi avahi-daemon[347]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::f222:ec88:3118:6fa.
Feb  3 02:57:17 raspberrypi kernel: [  129.292945] usb 1-1: USB disconnect, device number 2
Feb  3 02:57:17 raspberrypi kernel: [  129.292986] usb 1-1.1: USB disconnect, device number 3
Feb  3 02:57:17 raspberrypi kernel: [  129.293262] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-20980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Feb  3 02:57:17 raspberrypi kernel: [  129.293356] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Feb  3 02:57:18 raspberrypi kernel: [  129.313049] ------------[ cut here ]------------
Feb  3 02:57:18 raspberrypi kernel: [  129.313131] WARNING: CPU: 0 PID: 15 at include/asm-generic/dma-mapping-common.h:274 __DWC_DMA_FREE+0xac/0xc8()
Feb  3 02:57:18 raspberrypi kernel: [  129.313148] Modules linked in: cfg80211 rfkill evdev snd_usb_audio joydev snd_usbmidi_lib snd_hwdep sierra_net snd_rawmidi snd_seq_device sierra usbserial bcm2835_gpiomem snd_bcm2835 bcm2835_wdt snd_pcm snd_timer snd uio_pdrv_genirq uio
Feb  3 02:57:18 raspberrypi kernel: [  129.313255] CPU: 0 PID: 15 Comm: kworker/0:1 Not tainted 4.4.0+ #834
Feb  3 02:57:18 raspberrypi kernel: [  129.313269] Hardware name: BCM2708
Feb  3 02:57:18 raspberrypi kernel: [  129.313307] Workqueue: usb_hub_wq hub_event
Feb  3 02:57:18 raspberrypi kernel: [  129.313377] [<c00169b8>] (unwind_backtrace) from [<c0013944>] (show_stack+0x20/0x24)
Feb  3 02:57:18 raspberrypi kernel: [  129.313412] [<c0013944>] (show_stack) from [<c02df2f4>] (dump_stack+0x20/0x28)
Feb  3 02:57:18 raspberrypi kernel: [  129.313447] [<c02df2f4>] (dump_stack) from [<c0021a28>] (warn_slowpath_common+0x8c/0xc4)
Feb  3 02:57:18 raspberrypi kernel: [  129.313479] [<c0021a28>] (warn_slowpath_common) from [<c0021b1c>] (warn_slowpath_null+0x2c/0x34)
Feb  3 02:57:18 raspberrypi kernel: [  129.313515] [<c0021b1c>] (warn_slowpath_null) from [<c03f8368>] (__DWC_DMA_FREE+0xac/0xc8)
Feb  3 02:57:18 raspberrypi kernel: [  129.313549] [<c03f8368>] (__DWC_DMA_FREE) from [<c03f1e20>] (dwc_otg_hcd_qh_free+0xbc/0xec)
Feb  3 02:57:18 raspberrypi kernel: [  129.313579] [<c03f1e20>] (dwc_otg_hcd_qh_free) from [<c03ea6bc>] (dwc_otg_hcd_endpoint_disable+0xac/0xc0)
Feb  3 02:57:18 raspberrypi kernel: [  129.313609] [<c03ea6bc>] (dwc_otg_hcd_endpoint_disable) from [<c03edca8>] (endpoint_disable+0x28/0x34)
Feb  3 02:57:18 raspberrypi kernel: [  129.313649] [<c03edca8>] (endpoint_disable) from [<c03c4148>] (usb_hcd_disable_endpoint+0x3c/0x40)
Feb  3 02:57:18 raspberrypi kernel: [  129.313687] [<c03c4148>] (usb_hcd_disable_endpoint) from [<c03c66c8>] (usb_disable_endpoint+0x70/0x8c)
Feb  3 02:57:18 raspberrypi kernel: [  129.313720] [<c03c66c8>] (usb_disable_endpoint) from [<c03c6904>] (usb_disable_device+0x1bc/0x1d8)
Feb  3 02:57:18 raspberrypi kernel: [  129.313752] [<c03c6904>] (usb_disable_device) from [<c03bec10>] (usb_disconnect+0x78/0x208)
Feb  3 02:57:18 raspberrypi kernel: [  129.313781] [<c03bec10>] (usb_disconnect) from [<c03bed44>] (usb_disconnect+0x1ac/0x208)
Feb  3 02:57:18 raspberrypi kernel: [  129.313809] [<c03bed44>] (usb_disconnect) from [<c03bf85c>] (hub_event+0x300/0xe4c)
Feb  3 02:57:18 raspberrypi kernel: [  129.313839] [<c03bf85c>] (hub_event) from [<c0036b84>] (process_one_work+0x110/0x39c)
Feb  3 02:57:18 raspberrypi kernel: [  129.313866] [<c0036b84>] (process_one_work) from [<c0036f6c>] (worker_thread+0x15c/0x4e4)
Feb  3 02:57:18 raspberrypi kernel: [  129.313901] [<c0036f6c>] (worker_thread) from [<c003c6e8>] (kthread+0xe0/0xfc)
Feb  3 02:57:18 raspberrypi kernel: [  129.313933] [<c003c6e8>] (kthread) from [<c000f608>] (ret_from_fork+0x14/0x2c)
Feb  3 02:57:18 raspberrypi kernel: [  129.313950] ---[ end trace 38aa7ae261971bca ]---
Feb  3 02:57:18 raspberrypi kernel: [  129.313979] ------------[ cut here ]------------
Feb  3 02:57:18 raspberrypi kernel: [  129.314012] WARNING: CPU: 0 PID: 15 at include/asm-generic/dma-mapping-common.h:274 __DWC_DMA_FREE+0xac/0xc8()
Feb  3 02:57:18 raspberrypi kernel: [  129.314025] Modules linked in: cfg80211 rfkill evdev snd_usb_audio joydev snd_usbmidi_lib snd_hwdep sierra_net snd_rawmidi snd_seq_device sierra usbserial bcm2835_gpiomem snd_bcm2835 bcm2835_wdt snd_pcm snd_timer snd uio_pdrv_genirq uio
Feb  3 02:57:18 raspberrypi kernel: [  129.314112] CPU: 0 PID: 15 Comm: kworker/0:1 Tainted: G        W       4.4.0+ #834
Feb  3 02:57:18 raspberrypi kernel: [  129.314125] Hardware name: BCM2708
Feb  3 02:57:18 raspberrypi kernel: [  129.314147] Workqueue: usb_hub_wq hub_event
Feb  3 02:57:18 raspberrypi kernel: [  129.314187] [<c00169b8>] (unwind_backtrace) from [<c0013944>] (show_stack+0x20/0x24)
Feb  3 02:57:18 raspberrypi kernel: [  129.314215] [<c0013944>] (show_stack) from [<c02df2f4>] (dump_stack+0x20/0x28)
Feb  3 02:57:18 raspberrypi kernel: [  129.314243] [<c02df2f4>] (dump_stack) from [<c0021a28>] (warn_slowpath_common+0x8c/0xc4)
[... cut ...]
Feb  3 02:59:13 raspberrypi dhclient: DHCPOFFER from 192.168.166.2
Feb  3 02:59:13 raspberrypi dhclient: DHCPACK from 192.168.166.2
Feb  3 02:59:13 raspberrypi avahi-daemon[347]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.166.77.
Feb  3 02:59:13 raspberrypi avahi-daemon[347]: New relevant interface eth0.IPv4 for mDNS.
Feb  3 02:59:13 raspberrypi avahi-daemon[347]: Registering new address record for 192.168.166.77 on eth0.IPv4.
Feb  3 02:59:14 raspberrypi dhclient: bound to 192.168.166.77 -- renewal in 310951 seconds.

HiassofT, 2016-02-03:
The reason for the 4.4 backtrace seems to be the same issue that we had with the i2s driver, dma_free_coherent being called with interrupts disabled. dwc_otg_hcd_qh_free calls DWC_DMA_FREE inside a spinlock_irqsave block.

Looks like preparation for using DMA pool had been started but it looks unfinished and everything's "#if 0"ed out.

jkuek, 2016-02-03:
Interestingly, it seems like there have been stability issues for a while:
https://www.raspberrypi.org/forums/viewtopic.php?p=756815#p756815

Also note I don't believe my problem is power-related, as the USB modem is actually self-powered via a separate power adapter

jkuek, 2016-02-04:
Here's another backtrace from 4.1.13 :
[ 1306.473403] Unable to handle kernel NULL pointer dereference at virtual address                                        00000034
[ 1306.483428] pgd = c0004000
[ 1306.487913] [00000034] *pgd=00000000
[ 1306.493312] Internal error: Oops: 817 [#1] PREEMPT ARM
[ 1306.500239] Modules linked in: cfg80211 rfkill joydev evdev snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_dev                                       ice sierra sierra_net usbserial bcm2835_gpiomem snd_bcm2835 snd_pcm snd_timer snd uio_pdrv_genirq uio ipv6
[ 1306.525746] CPU: 0 PID: 19 Comm: kworker/0:1 Not tainted 4.1.13+ #826
[ 1306.534669] Hardware name: BCM2708
[ 1306.540656] Workqueue: usb_hub_wq hub_event
[ 1306.547385] task: dadb0000 ti: dada6000 task.ti: dada6000
[ 1306.555513] PC is at assign_and_init_hc+0x60/0x670
[ 1306.563086] LR is at dwc_otg_hcd_select_transactions+0xbc/0x3a8
[ 1306.571858] pc : [<c0417f34>]    lr : [<c041a09c>]    psr: 00000193
[ 1306.571858] sp : dada7860  ip : dada7898  fp : dada7894
[ 1306.589187] r10: da5b9f28  r9 : daf22860  r8 : da79c700
[ 1306.597457] r7 : daf22800  r6 : da5b9f00  r5 : da79cc40  r4 : daef70c0
[ 1306.607107] r3 : daf22860  r2 : 00000000  r1 : 00000000  r0 : daf22800
[ 1306.616813] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[ 1306.627518] Control: 00c5387d  Table: 1a598008  DAC: 00000015
[ 1306.636521] Process kworker/0:1 (pid: 19, stack limit = 0xdada6188)
[ 1306.646250] Stack: (0xdada7860 to 0xdada8000)
[ 1306.654149] 7860: 20000193 daf22828 dada78ac daf22830 00000010 daf22828 dada78ac daf22830
[ 1306.666039] 7880: daf22800 daf22860 dada78dc dada7898 c041a09c c0417ee0 dada78c4 daf22838
[ 1306.678017] 78a0: da79c5c0 00000000 d87eb050 20000193 d87eb038 daf22800 daf22828 daf22830
[ 1306.689998] 78c0: 02000008 c087f670 f301080e c0892d04 dada78fc dada78e0 c041ddd8 c0419fec
[ 1306.702174] 78e0: 1c2e1a31 daf22800 daf22400 00000001 dada7924 dada7900 c04201b0 c041dd14
[ 1306.714441] 7900: dae42d40 dae42b40 00000020 00000000 00000000 00000000 dada7934 dada7928
[ 1306.726809] 7920: c041c644 c041fedc dada7944 dada7938 c03eed04 c041c634 dada7984 dada7948
[ 1306.739150] 7940: c006186c c03eecdc c03f0e7c c041c294 00000003 c082dc64 dada7994 c082dc64
[ 1306.751707] 7960: dae42b40 00000000 00000000 00000000 d87eb000 00000000 dada799c dada7988
[ 1306.764368] 7980: c0061b08 c0061834 c082dc64 c0871eb0 dada79b4 dada79a0 c0064c24 c0061aa4
[ 1306.777154] 79a0: c0064b7c 00000020 dada79dc dada79b8 c0061138 c0064b88 dada79f0 c006e8d4
[ 1306.789902] 79c0: 20000013 f200b200 dada7a24 00001388 dada79ec dada79e0 c00093a4 c00610c8
[ 1306.802825] 79e0: dada7a44 dada79f0 c058d538 c0009384 00001391 60000013 00000000 cccccccd
[ 1306.815922] 7a00: d8449680 dada7a48 00000000 dada7a8c 00001388 d87eb000 00000000 dada7a44
[ 1306.829072] 7a20: dada7a48 dada7a38 c03f3170 c006e8d4 20000013 ffffffff dada7a7c dada7a48
[ 1306.842307] 7a40: c03f3170 c006e8c0 00000000 dada7a4c dada7a4c 00000000 dada7a7c d86667c0
[ 1306.855535] 7a60: 00000004 00000007 00000000 80002800 dada7abc dada7a80 c03f3268 c03f30d8
[ 1306.868985] 7a80: 00000000 00000000 00000000 da499c00 d8666740 d8666740 00000004 00000007
[ 1306.882516] 7aa0: 00000000 00000021 00000000 da499c00 dada7b04 dada7ac0 c03e6258 c03f31a4
[ 1306.896110] 7ac0: 00000000 00000007 d8666740 00000004 00001388 c09410a4 bf1062a0 00000007
[ 1306.909776] 7ae0: da499c00 bf10643c 00000000 d87fe400 bf106168 c087efa8 dada7b2c dada7b08
[ 1306.923462] 7b00: c03e62b0 c03e61dc 00000007 da77b214 00000004 c09410a4 dada7b74 da77b200
[ 1306.937269] 7b20: dada7b5c dada7b30 bf105844 c03e6290 00000007 da77b214 00000004 d87fe400
[ 1306.951255] 7b40: bf106168 da77b200 da499c00 bf1066cc dada7b74 dada7b60 bf1058a0 bf105808
[ 1306.965365] 7b60: 00000000 da499c00 dada7b8c dada7b78 bf105d6c bf105884 d87fe420 d87eb000
[ 1306.979575] 7b80: dada7bbc dada7b90 c03f6af8 bf105d2c c03f6a20 c09246c4 d87fe420 00000000
[ 1306.993876] 7ba0: bf1066cc 00000f4e 00000000 c087efa8 dada7be4 dada7bc0 c038f590 c03f6a2c
[ 1307.008166] 7bc0: 00000007 bf1066cc d87fe420 c038f648 d87eb068 c09246a0 dada7bfc dada7be8
[ 1307.022561] 7be0: c038f6a8 c038f3f0 00000000 d87fe420 dada7c24 dada7c00 c038d8c8 c038f654
[ 1307.037097] 7c00: dad49ddc da764c34 d87fe428 d87fe420 d87fe454 c087efc0 dada7c44 dada7c28
[ 1307.051822] 7c20: c038f3d8 c038d86c dad49d80 d87fe428 d87fe420 c087efc0 dada7c64 dada7c48
[ 1307.066615] 7c40: c038e9dc c038f360 60000013 d87fe428 d87fe420 00000000 dada7ca4 dada7c68
[ 1307.081470] 7c60: c038c9d8 c038e954 d87fe220 d87eb000 dada7ca4 dada7c80 c03f42f0 d87fe400
[ 1307.096410] 7c80: 00000003 d87fee5c 00000007 d87eb068 00000004 00000004 dada7d1c dada7ca8
[ 1307.111448] 7ca0: c03f5568 c038c638 00000001 00000000 00000000 00000000 00001388 d86ce780
[ 1307.126512] 7cc0: dada7cec dad4df50 dae7b600 00000001 d87da520 d87eb004 00000004 00000001
[ 1307.141607] 7ce0: d87fee00 d87da52c d87eb068 d87fee5c c03fe674 d87eb000 00000001 00000000
[ 1307.156782] 7d00: c087f59c 00000f48 00000000 c087ee9c dada7d34 dada7d20 c03fe874 c03f4ff0
[ 1307.172057] 7d20: c09246c4 d87eb068 dada7d44 dada7d38 c03f5a9c c03fe844 dada7d6c dada7d48
[ 1307.187374] 7d40: c038f590 c03f5a80 00000007 c087f59c d87eb068 c038f648 d869c868 dac04d40
[ 1307.202735] 7d60: dada7d84 dada7d70 c038f6a8 c038f3f0 00000000 d87eb068 dada7dac dada7d88
[ 1307.218131] 7d80: c038d8c8 c038f654 dad49ddc dad2ff94 d87eb070 d87eb068 d87eb09c c087efc0
[... cut ...]
[ 1308.175495] [<c003ba38>] (process_one_work) from [<c003bf00>] (worker_thread+0x178/0x538)
[ 1308.191001] [<c003bf00>] (worker_thread) from [<c0041714>] (kthread+0xe0/0xfc)
[ 1308.205527] [<c0041714>] (kthread) from [<c000f7d8>] (ret_from_fork+0x14/0x3c)
[ 1308.220029] Code: e1520003 05942034 15941034 05872064 (15821034)
[ 1308.233520] ---[ end trace d5434125319bc15b ]---
[ 1308.245484] Kernel panic - not syncing: Fatal exception in interrupt
[ 1308.259168] ---[ end Kernel panic - not syncing: Fatal exception in interrupt

jkuek, 2016-02-04:
Here's kern.log showing multiple oops and eventual panic on 4.4.1
[panic 4.4.1.txt](https://github.com/raspberrypi/linux/files/116770/panic.4.4.1.txt)

jkuek, 2016-02-04:
Another log, different backtrace
[dwc error 4.4.1.txt](https://github.com/raspberrypi/linux/files/116859/dwc.error.4.4.1.txt)

jkuek, 2016-02-09:
Updated the kernel using apt-get to 4.1.17+ #834, still getting a crash, but this time it says 

[crash 4.1.17.txt](https://github.com/raspberrypi/linux/files/122476/crash.4.1.17.txt)
kernel BUG at drivers/usb/host/dwc_otg/dwc_otg_hcd_intr.c:2425!
Internal error: Oops - BUG: 0 [#1] ARM

Is anyone still maintaining the dwc_otg driver? I see there is a dwc2 driver in mainline, should I switch to that instead?

P33M, 2016-02-10:
It's possible that memory's being leaked when disconnecting/reconnecting as dwc_otg will stop and start the host driver.

dwc2 is the upstream driver but lacks FIQ support. If you have no low- or full-speed devices then it will probably work OK. It is the go-to for Pi Zero in device mode, however.

tpfkanep, 2016-03-11:
I found **usb_modeswitch 2.3.0-1** (on archlinux) to be the cause of my kernel panics when I plug in/out my usb modem. When I downgrade the package to the previous version (2.2.6-1), all is fine.

jkuek, 2016-03-13:
@tpfkanep This particular problem doesn't just happen with the USB modem. I can reproduce it with a USB audio card or other devices as well. Having multiple USB devices seems to make it easier to reproduce though.

jkuek, 2016-03-21:
Seen again on 4.4.6+ #861:

```
[  767.863412] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[  767.875664] pgd = c0004000
[  767.882356] [00000034] *pgd=00000000
[  767.889882] Internal error: Oops: 817 [#1] ARM
[  767.898390] Modules linked in: cfg80211 rfkill joydev evdev snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device sierra sierra_net usbserial snd_soc_bcm2835_i2s regmap_mmio snd_soc_core snd_pcm_dmaengine spi_bcm2835 snd_bcm2835 bcm2835_gpiomem bcm2835_wdt snd_pcm snd_timer snd uio_pdrv_genirq uio
[  767.935694] CPU: 0 PID: 173 Comm: kworker/0:5 Not tainted 4.4.6+ #861
[  767.946820] Hardware name: BCM2708
[  767.955090] Workqueue: usb_hub_wq hub_event
[  767.964199] task: da75b680 ti: d8c18000 task.ti: d8c18000
[  767.974568] PC is at assign_and_init_hc+0x60/0x648
[  767.984376] LR is at dwc_otg_hcd_select_transactions+0xb8/0x3a0
[  767.995263] pc : [<c03ed8c0>]    lr : [<c03efacc>]    psr: 00000193
[  767.995263] sp : d8c19698  ip : d8c196d0  fp : d8c196cc
[  768.016908] r10: d8c00a48  r9 : daf94460  r8 : da5ca1c0
[  768.027283] r7 : daf94400  r6 : d8c00a20  r5 : da5ca080  r4 : daf169c0
[  768.039059] r3 : daf94460  r2 : 00000000  r1 : 00000000  r0 : daf94400
[  768.050823] Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[  768.063511] Control: 00c5387d  Table: 18e78008  DAC: 00000055
[  768.074696] Process kworker/0:5 (pid: 173, stack limit = 0xd8c18188)
[  768.086546] Stack: (0xd8c19698 to 0xd8c1a000)
[  768.096315] 9680:                                                       daf94428 d8c196e4
[  768.110122] 96a0: daf94430 daf94400 00000001 daf94428 d8c196e4 daf94430 daf94400 daf94460
[  768.124018] 96c0: d8c19714 d8c196d0 c03efacc c03ed86c 00000000 daf94438 c03edc30 daf169c0
[  768.137960] 96e0: da405f28 20000193 c08a88a0 daf94400 daf94428 daf94430 02000008 c08a8700
[  768.151863] 9700: f301080e 00000008 d8c19734 d8c19718 c03f39c0 c03efa20 00000000 daf94400
[  768.165837] 9720: daf94000 00000008 d8c19764 d8c19738 c03f5cf4 c03f38fc c03f20d8 dac1ec00
[  768.179936] 9740: daea1bc0 00000038 00000000 00000000 00000000 dac1ec00 d8c19774 d8c19768
[  768.194135] 9760: c03f20f4 c03f5a0c d8c19784 d8c19778 c03c4fc4 c03f20e4 d8c197bc d8c19788
[  768.208363] 9780: c005a9b4 c03c4f9c 0000000a 00000010 32343537 dac1ec00 dac1ec10 00000000
[  768.222554] 97a0: 00000000 00000000 da57fc38 daf82580 d8c197d4 d8c197c0 c005ab48 c005a92c
[  768.236665] 97c0: 00020000 dac1ec00 d8c197ec d8c197d8 c005d8d4 c005ab1c 00000038 c089a864
[  768.250759] 97e0: d8c197fc d8c197f0 c005a028 c005d844 d8c19824 d8c19800 c005a2c0 c005a004
[  768.264700] 9800: d8c19850 c0859534 d8c19850 ffffffff d8c19884 daf19c00 d8c19834 d8c19828
[  768.278615] 9820: c0010748 c005a274 d8c1984c d8c19838 c000940c c0010728 c03fcdac 60000013
[  768.292592] 9840: d8c198b4 d8c19850 c05a5024 c00093dc 00000000 00000000 da405f00 c0858f74
[  768.306637] 9860: 60000013 5a7b0d00 da5ca5c0 daf94400 daf19c00 da57fc38 daf82580 d8c198b4
[  768.320729] 9880: d8c198a0 d8c198a0 c03fcd98 c03fcdac 60000013 ffffffff c03fcd98 c00b4f28
[  768.334716] 98a0: 00000000 00000000 d8c1990c d8c198b8 c03f1f14 c03fcd80 00000004 da7b0d00
[  768.348820] 98c0: 5a7b0d00 00000001 00000000 c00198ac 00000000 da57fc50 00000400 60000013
[  768.363089] 98e0: c0862b5c daf82580 daf19c00 daf82588 00000000 02400000 00000002 00000000
[  768.377438] 9900: d8c199bc d8c19910 c03c6fcc c03f1d2c daf94400 c03f6024 d8c19954 d8c19928
[  768.391853] 9920: c03f6024 c03fbf14 00000000 00000000 d8c00a20 c03f6de4 da5ca080 da5ca1c0
[  768.406331] 9940: 00000001 d8d71840 d8c19974 c03f033c 00000008 daf94438 daf9441c daf94460
[  768.420856] 9960: d8d71840 d8c6e780 d8c19994 daf94400 00000000 d8d71858 da5ca1c0 00000001
[  768.435291] 9980: d8d71840 d8c6e780 d8c199c4 d8c19998 60000013 daf82580 02400000 00000000
[  768.449779] 99a0: 00000000 da57fc00 00000002 00000000 d8c19a04 d8c199c0 c03c88b4 c03c6f10
[  768.464401] 99c0: c03f1f14 c03fcd80 00000040 00000000 00000000 00000001 00000040 daf82580
[  768.479203] 99e0: d8c19a08 00000000 d8c19a4c 00001388 00000000 00000000 d8c19a3c d8c19a08
[  768.494090] 9a00: c03c8e38 c03c85d8 00000000 d8c19a0c d8c19a0c 00000000 d8c19a3c da7b0d00
[  768.509039] 9a20: 00000004 00000007 80002800 da57fc00 d8c19a7c d8c19a40 c03c8f60 c03c8df0
[  768.524095] 9a40: d8c4d514 00000021 00000021 00000000 da7b0a80 da7b0a80 00000004 00000007
[  768.539232] 9a60: 00000000 00000021 00000000 da42dc00 d8c19ac4 d8c19a80 c03bc54c c03c8ebc
[  768.554370] 9a80: 00000000 00000007 da7b0a80 00000004 00001388 c096e33c c82a0640 00000007
[  768.569488] 9aa0: da42dc00 bf0d947c d8f6b400 da57fc00 bf0d96e8 00000000 d8c19aec d8c19ac8
[  768.584690] 9ac0: c03bc5a8 c03bc4d0 00000007 d8c4d514 00000004 c05a15b0 d8c18000 d8c4d500
[  768.599951] 9ae0: d8c19b1c d8c19af0 bf0d884c c03bc588 00000007 d8c4d514 00000004 da57fc00
[  768.615319] 9b00: bf0d96e8 d8c4d500 da42dc00 d8f6b420 d8c19b34 d8c19b20 bf0d88a8 bf0d8810
[  768.630825] 9b20: 00000000 da42dc00 d8c19b4c d8c19b38 bf0d8d74 bf0d888c bf0d96e0 bf0d91a8
[  768.646366] 9b40: d8c19b7c d8c19b50 c03cc63c bf0d8d34 c03cc52c c095195c d8f6b420 00000000
[  768.661940] 9b60: bf0d96e8 00000911 00000000 c08a8000 d8c19ba4 d8c19b80 c03670d8 c03cc538
[  768.677499] 9b80: 00000007 d8f6b420 d8c19bf0 bf0d96e8 00000001 c0951938 d8c19bc4 d8c19ba8
[  768.693059] 9ba0: c03673dc c0366f30 d8c00a00 00000000 d8c19bf0 c0367330 d8c19bec d8c19bc8
[... cut ...]
[  770.003361] [<c03c3afc>] (hub_event) from [<c00374c4>] (process_one_work+0x11c/0x39c)
[  770.018798] [<c00374c4>] (process_one_work) from [<c0037784>] (worker_thread+0x40/0x4d8)
[  770.034538] [<c0037784>] (worker_thread) from [<c003d118>] (kthread+0xdc/0xf8)
[  770.049393] [<c003d118>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
[  770.064216] Code: e1520003 05942034 15941034 05872064 (15821034)
[  770.077988] ---[ end trace 465aa20bc07a3b82 ]---
[  770.090270] Kernel panic - not syncing: Fatal exception in interrupt
```

tpfkanep, 2016-07-03:
A user on the usb_modeswitch forum posted the following solution (working great on my Pi for a couple of days now):

> I solved it - use the following config:
> 
> TargetVendor=0x1bbb
> TargetProductList="0000,0017,00b7,011e,0191,0195"
> **StandardEject=1**
> 
> instead of:
> 
> TargetVendor=0x1bbb
> TargetProductList="0000,0017,00b7,011e,0191,0195"
> MessageContent="55534243123456788000000080000606f50402527000000000000000000000"

[http://www.draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=4&t=2498](url)

jkuek, 2016-07-03:
@tpfkanep Thanks for the info.

This might fix your particular issue, but as previously mentioned, I can reproduce the kernel panics even without usb_modeswitch and a usb modem, e.g with a USB audio card.

Ruffio, 2016-08-17:
@jkuek has your issue been resolved? If so, please close this issue. Thanks.

vpelletier, 2016-12-22:
I experience what looks like the same crash. This raspberry pi B+ is headless, uses the onboard ethernet adapter and has 3 USB devices plugged in. I connected another, otherwise unused, B+ to the failing pi's on-board serial console to capture the crash, see below.

This is with a pure Debian sid armel install plus `rpi_update`-provided kernel. Both Debian and the lernel were updated about 4 days ago, and the pi was last rebooted a bit under 4 days ago:

```python
>>> import time
>>> time.ctime(246972.664501)
'Sat Jan  3 20:36:12 1970'
```

```
[246972.664501] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[246972.672987] pgd = d5adc000
[246972.675879] [00000034] *pgd=00000000
[246972.679683] Internal error: Oops: 805 [#1] ARM

Entering kdb (current=0xd63e8da0, pid 2769) Oops: (null)
due to oops @ 0xc03fcae0
CPU: 0 PID: 2769 Comm: python Not tainted 4.4.38+ #938
Hardware name: BCM2708
task: d63e8da0 ti: d5afe000 task.ti: d5afe000
PC is at assign_and_init_hc+0x60/0x648
LR is at dwc_otg_hcd_select_transactions+0x26c/0x3a0
pc : [<c03fcae0>]    lr : [<c03feea0>]    psr: 00000193
sp : d5aff808  ip : d5aff840  fp : d5aff83c
r10: dafae400  r9 : d5869988  r8 : d5a2e680
r7 : dafae400  r6 : d5869960  r5 : d5a2e780  r4 : daf2f9c0
r3 : dafae460  r2 : 00000000  r1 : 00000000  r0 : dafae400
Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 00c5387d  Table: 15adc008  DAC: 00000055
CPU: 0 PID: 2769 Comm: python Not tainted 4.4.38+ #938
Hardware name: BCM2708
[<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
[<c0013c20>] (show_stack) from [<c02e437c>] (dump_stack+0x20/0x28)
[<c02e437c>] (dump_stack) from [<c0010a9c>] (show_regs+0x1c/0x20)
[<c0010a9c>] (show_regs) from [<c00969f4>] (kdb_main_loop+0x32c/0x74c)
[<c00969f4>] (kdb_main_loop) from [<c00994f4>] (kdb_stub+0x1d0/0x404)
[<c00994f4>] (kdb_stub) from [<c008fb10>] (kgdb_handle_exception+0x28c/0x7bc)
[<c008fb10>] (kgdb_handle_exception) from [<c0016244>] (kgdb_notify+0x3c/0x94)
[<c0016244>] (kgdb_notify) from [<c003e018>] (notifier_call_chain+0x54/0x94)
[<c003e018>] (notifier_call_chain) from [<c003e220>] (notify_die+0x48/0x54)
[<c003e220>] (notify_die) from [<c0013cec>] (die+0xc8/0x374)
[<c0013cec>] (die) from [<c001e264>] (__do_kernel_fault.part.0+0x64/0x1f4)
[<c001e264>] (__do_kernel_fault.part.0) from [<c0576598>] (do_page_fault+0x234/0x3c4)
[<c0576598>] (do_page_fault) from [<c05767d4>] (do_translation_fault+0xac/0xb4)
[<c05767d4>] (do_translation_fault) from [<c0009204>] (do_DataAbort+0x44/0xc4)
[<c0009204>] (do_DataAbort) from [<c0575cfc>] (__dabt_svc+0x3c/0x60)
Exception stack(0xd5aff7b8 to 0xd5aff800)
f7a0:                                                       dafae400 00000000
f7c0: 00000000 dafae460 daf2f9c0 d5a2e780 d5869960 dafae400 d5a2e680 d5869988
f7e0: dafae400 d5aff83c d5aff840 d5aff808 c03feea0 c03fcae0 00000193 ffffffff
[<c0575cfc>] (__dabt_svc) from [<c03fcae0>] (assign_and_init_hc+0x60/0x648)
[<c03fcae0>] (assign_and_init_hc) from [<c03feea0>] (dwc_otg_hcd_select_transactions+0x26c/0x3a0)
[<c03feea0>] (dwc_otg_hcd_select_transactions) from [<c0402060>] (release_channel+0x138/0x224)
[<c0402060>] (release_channel) from [<c040251c>] (halt_channel+0x9c/0x150)
[<c040251c>] (halt_channel) from [<c0402634>] (complete_non_periodic_xfer+0x64/0x84)
[<c0402634>] (complete_non_periodic_xfer) from [<c0403334>] (handle_hc_xfercomp_intr+0x2b0/0x46c)
[<c0403334>] (handle_hc_xfercomp_intr) from [<c04047b0>] (dwc_otg_hcd_handle_hc_n_intr+0x394/0x718)
[<c04047b0>] (dwc_otg_hcd_handle_hc_n_intr) from [<c0404bec>] (dwc_otg_hcd_handle_hc_intr+0xb8/0xf8)
[<c0404bec>] (dwc_otg_hcd_handle_hc_intr) from [<c0404ea0>] (dwc_otg_hcd_handle_intr+0x274/0x370)
[<c0404ea0>] (dwc_otg_hcd_handle_intr) from [<c0401314>] (dwc_otg_hcd_irq+0x1c/0x28)
[<c0401314>] (dwc_otg_hcd_irq) from [<c03d21e0>] (usb_hcd_irq+0x34/0x48)
[<c03d21e0>] (usb_hcd_irq) from [<c005ac60>] (handle_irq_event_percpu+0x94/0x1f0)
[<c005ac60>] (handle_irq_event_percpu) from [<c005adf4>] (handle_irq_event+0x38/0x4c)
[<c005adf4>] (handle_irq_event) from [<c005db80>] (handle_level_irq+0x9c/0x134)
[<c005db80>] (handle_level_irq) from [<c005a2d4>] (generic_handle_irq+0x30/0x44)
[<c005a2d4>] (generic_handle_irq) from [<c005a56c>] (__handle_domain_irq+0x58/0xb8)
[<c005a56c>] (__handle_domain_irq) from [<c0010748>] (handle_IRQ+0x2c/0x30)
[<c0010748>] (handle_IRQ) from [<c000940c>] (bcm2835_handle_irq+0x3c/0x58)
[<c000940c>] (bcm2835_handle_irq) from [<c0575d64>] (__irq_svc+0x44/0x5c)
Exception stack(0xd5affac0 to 0xd5affb08)
fac0: 00000000 00000001 00000001 00000000 d5a68b38 da46cb80 dafe6000 40000113
fae0: dafe6480 00000000 00000000 d5affb44 d5affb10 d5affb10 c03c8880 c03c8894
fb00: 40000113 ffffffff
[<c0575d64>] (__irq_svc) from [<c03c8894>] (usbnet_start_xmit+0x570/0x658)
[<c03c8894>] (usbnet_start_xmit) from [<c049987c>] (dev_hard_start_xmit+0x258/0x3dc)
[<c049987c>] (dev_hard_start_xmit) from [<c04bb1d0>] (sch_direct_xmit+0xf8/0x1dc)
[<c04bb1d0>] (sch_direct_xmit) from [<c0499d44>] (__dev_queue_xmit+0x278/0x554)
[<c0499d44>] (__dev_queue_xmit) from [<c049a03c>] (dev_queue_xmit+0x1c/0x20)
[<c049a03c>] (dev_queue_xmit) from [<c04d4558>] (ip_finish_output2+0x18c/0x360)
[<c04d4558>] (ip_finish_output2) from [<c04d6078>] (ip_finish_output+0x168/0x250)
[<c04d6078>] (ip_finish_output) from [<c04d6ba4>] (ip_output+0xa4/0x114)
[<c04d6ba4>] (ip_output) from [<c04d630c>] (ip_local_out+0x50/0x54)
[... cut ...]
[<c04edb28>] (tcp_transmit_skb) from [<c04ee1e0>] (tcp_write_xmit+0x1ec/0xea0)
[<c04ee1e0>] (tcp_write_xmit) from [<c04ef274>] (tcp_push_one+0x44/0x50)
[<c04ef274>] (tcp_push_one) from [<c04e16dc>] (tcp_sendmsg+0x3f8/0xadc)
[<c04e16dc>] (tcp_sendmsg) from [<c0509548>] (inet_sendmsg+0x44/0x78)
[<c0509548>] (inet_sendmsg) from [<c047cd58>] (sock_sendmsg+0x24/0x34)
[<c047cd58>] (sock_sendmsg) from [<c047ddb0>] (SyS_sendto+0xc8/0xf0)
[<c047ddb0>] (SyS_sendto) from [<c047de00>] (SyS_send+0x28/0x30)
[<c047de00>] (SyS_send) from [<c000f820>] (ret_fast_syscall+0x0/0x1c)
```

P33M, 2016-12-22:
Please provide the output of `sudo lsusb -v`. Also please provide information on

- Details of any userspace programs running and accessing devices
- The typical length of time it takes to trigger an OOPS and the usage case when the OOPS happens

vpelletier, 2016-12-23:
That pi is used to stream TV in my lan. Are plugged in:

- a micro SD card adapter, to which the main sd card is rsync'ed periodically (once a month)

- a TV tuner

- a smartcard reader (needed to decrypt broadcast)

- plus the b+ internal hub and ethernet adapter

The workload at crash time is: [`recfsusb2i`](http://ktvwiki.22web.org/?BonDriver_FSUSB2i&i=1) (userland usb driver for the TV tuner) reading the encrypted mpeg transport stream, talking to the smartcard reader via `pcscd`, decrypting the stream, piping the output to a short python program which handles a tcp socket, sending the output verbatim to any connected peer.

In the case of this crash, a single host was connected.

Crashs happen with a rather low probability compared to the number of transmitted packets (TCP bandwidth when streaming is peer*2MB/s, peer being 1 or 0 in practice). For about an hour per day of streaming, crashes happen once or twice a week. I expect the TV tuner to also send 2MB/s of data and the smartcard reader just a tiny amount (not measured, though).

So, from a USB level, I expect the relevant pieces to be:
- an interrupt endpoint (smartcard reader card inserted/removed notification), plus the interrupt endpoints for the internal hub
- 4MB/s total bulk bandwidth usage, 2 up and 2 down

I rebooted from kdb, and have built a kernel *with* vmlinux image:

```bash
$ git checkout rpi-4.4.y
$ git rev-parse HEAD
5e46914b3417fe9ff42546dcacd0f41f9a0fb172
$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcmrpi_defconfig
$ nice -n 19 make -j16 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage vmlinux modules dtbs
```

and just booted it. Next time, I should have an actually usable kgdb... @Hexxeh: Could you build and make vmlinux available whenever you make a release ? It would help debugging.

Here are the descriptors, minus serial numbers:

```
Bus 001 Device 006: ID 1e3d:4082 Chipsbank Microelectronics Co., Ltd CBM4082 SD Card Reader
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1e3d Chipsbank Microelectronics Co., Ltd
  idProduct          0x4082 CBM4082 SD Card Reader
  bcdDevice            1.00
  iManufacturer           0
  iProduct                2 Flash Disk
  iSerial                 3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
[... 398 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

jkuek, 2016-12-23:
I haven't tried recently but when I originally reported the issue, I was able to reproduce it fairly often by disconnecting/reconnecting USB devices from the RPi. The disconnects were either manual, or by changing the buspower file as documented.

It was definitely easier to reproduce with multiple USB devices plugged in, but the issue did still occur with a single USB device (I tried both USB audio and USB 3g modem)

vpelletier, 2016-12-24:
I got a similar crash and have a working gdb this time. It turns out I needed more than just vmlinux to get full debug symbols.

The crash is still in `assign_and_init_hc`, but this time it is the userland driver which crashed.

kdb output:
```
Entering kdb (current=0xd8b06d00, pid 4391) Oops: (null)
due to oops @ 0xc04039a8
CPU: 0 PID: 4391 Comm: recfsusb2i Not tainted 4.4.39+ #76
Hardware name: BCM2708
task: d8b06d00 ti: d4c46000 task.ti: d4c46000
PC is at assign_and_init_hc+0x60/0x648
LR is at dwc_otg_hcd_select_transactions+0xb8/0x3a4
pc : [<c04039a8>]    lr : [<c0405ba4>]    psr: 00000193
sp : d4c47de0  ip : d4c47e20  fp : d4c47e1c
r10: d8ba6208  r9 : dafae460  r8 : d8bb3a00
r7 : dafae400  r6 : d8ba61e0  r5 : d8bb30c0  r4 : daf2f960
r3 : dafae460  r2 : 00000000  r1 : 00000000  r0 : dafae400
Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 00c5387d  Table: 14ca0008  DAC: 00000055
CPU: 0 PID: 4391 Comm: recfsusb2i Not tainted 4.4.39+ #76
Hardware name: BCM2708
[<c0016a1c>] (unwind_backtrace) from [<c0013a14>] (show_stack+0x20/0x24)
[<c0013a14>] (show_stack) from [<c02e9ee4>] (dump_stack+0x20/0x28)
[<c02e9ee4>] (dump_stack) from [<c0010850>] (show_regs+0x1c/0x20)
[<c0010850>] (show_regs) from [<c00980b0>] (kdb_main_loop+0x32c/0x870)
[<c00980b0>] (kdb_main_loop) from [<c009ad04>] (kdb_stub+0x1c4/0x3f8)
[<c009ad04>] (kdb_stub) from [<c0091238>] (kgdb_handle_exception+0x2a8/0x7b8)
[<c0091238>] (kgdb_handle_exception) from [<c0015f74>] (kgdb_notify+0x3c/0x94)
[<c0015f74>] (kgdb_notify) from [<c003e594>] (notifier_call_chain+0x54/0x90)
[<c003e594>] (notifier_call_chain) from [<c003e798>] (notify_die+0x48/0x54)
[<c003e798>] (notify_die) from [<c0013ad8>] (die+0xc0/0x340)
[<c0013ad8>] (die) from [<c001e16c>] (__do_kernel_fault.part.0+0x64/0x1f4)
[<c001e16c>] (__do_kernel_fault.part.0) from [<c057e3f8>] (do_page_fault+0x1e8/0x414)
[<c057e3f8>] (do_page_fault) from [<c000922c>] (do_DataAbort+0x48/0xc8)
[<c000922c>] (do_DataAbort) from [<c057dbbc>] (__dabt_svc+0x3c/0x60)
Exception stack(0xd4c47d90 to 0xd4c47dd8)
7d80:                                     dafae400 00000000 00000000 dafae460
7da0: daf2f960 d8bb30c0 d8ba61e0 dafae400 d8bb3a00 dafae460 d8ba6208 d4c47e1c
7dc0: d4c47e20 d4c47de0 c0405ba4 c04039a8 00000193 ffffffff
[<c057dbbc>] (__dabt_svc) from [<c04039a8>] (assign_and_init_hc+0x60/0x648)
[<c04039a8>] (assign_and_init_hc) from [<c0405ba4>] (dwc_otg_hcd_select_transactions+0xb8/0x3a4)
[<c0405ba4>] (dwc_otg_hcd_select_transactions) from [<c0409a40>] (dwc_otg_hcd_handle_sof_intr+0xd0/0x120)
[<c0409a40>] (dwc_otg_hcd_handle_sof_intr) from [<c040bccc>] (dwc_otg_hcd_handle_intr+0x2d4/0x360)
[<c040bccc>] (dwc_otg_hcd_handle_intr) from [<c0408188>] (dwc_otg_hcd_irq+0x1c/0x28)
[<c0408188>] (dwc_otg_hcd_irq) from [<c03d907c>] (usb_hcd_irq+0x34/0x48)
[<c03d907c>] (usb_hcd_irq) from [<c005b3f8>] (handle_irq_event_percpu+0x58/0x218)
[<c005b3f8>] (handle_irq_event_percpu) from [<c005b5f0>] (handle_irq_event+0x38/0x4c)
[<c005b5f0>] (handle_irq_event) from [<c005e3f4>] (handle_level_irq+0xa4/0x13c)
[<c005e3f4>] (handle_level_irq) from [<c005aab0>] (generic_handle_irq+0x30/0x44)
[<c005aab0>] (generic_handle_irq) from [<c005ad48>] (__handle_domain_irq+0x58/0xb8)
[<c005ad48>] (__handle_domain_irq) from [<c0010504>] (handle_IRQ+0x28/0x2c)
[<c0010504>] (handle_IRQ) from [<c000943c>] (bcm2835_handle_irq+0x44/0x5c)
[<c000943c>] (bcm2835_handle_irq) from [<c057def4>] (__irq_usr+0x54/0x80)
Exception stack(0xd4c47fb0 to 0xd4c47ff8)
7fa0:                                     00000004 00000003 ee5a6c66 d28c953f
7fc0: f2fe785c 07fa333e 3afb0755 2071e31d cf722380 890575ba aa3c7534 4677563a
7fe0: c2b528aa bed679f8 3d01346b b6f7736c 20000010 ffffffff
```

reassembled gdb session:
```
(gdb) lx-dmesg
[...]
[81169.614263] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[81169.622633] pgd = d4ca0000
[81169.625424] [00000034] *pgd=18a15831, *pte=00000000, *ppte=00000000
[81169.631923] Internal error: Oops: 817 [#1] ARM
[81269.525220] Unable to handle kernel NULL pointer dereference at virtual address 00000034
[81269.542157] pgd = c0004000
[... 30 lines cut ...]
1204            DWC_CIRCLEQ_REMOVE_INIT(&hcd->free_hc_list, hc, hc_list_entry);
   0xc0403994 <+76>:    ldr     r2, [r4, #48]   ; 0x30
   0xc0403998 <+80>:    cmp     r2, r3
   0xc040399c <+84>:    ldreq   r2, [r4, #52]   ; 0x34
   0xc04039a0 <+88>:    ldrne   r1, [r4, #52]   ; 0x34
   0xc04039a4 <+92>:    streq   r2, [r7, #100]  ; 0x64
=> 0xc04039a8 <+96>:    strne   r1, [r2, #52]   ; 0x34
   0xc04039ac <+100>:   ldr     r2, [r4, #52]   ; 0x34
   0xc04039b0 <+104>:   cmp     r3, r2
   0xc04039b4 <+108>:   ldr     r3, [r4, #48]   ; 0x30
   0xc04039b8 <+112>:   strne   r3, [r2, #48]   ; 0x30
   0xc04039bc <+116>:   streq   r3, [r7, #96]   ; 0x60
   0xc04039c0 <+120>:   mov     r2, #0
   0xc04039c8 <+128>:   str     r2, [r4, #48]   ; 0x30
   0xc04039cc <+132>:   str     r2, [r4, #52]   ; 0x34

1205
1206            qh->channel = hc;
   0xc04039d0 <+136>:   str     r4, [r6, #16]

```

So the null pointer dereference happens when removing an entry from `free_hc_list.cqh_first`.

I'll keep reading the code and report if I find something - but if anyone is familiar with this code, please do not wait for me.

vpelletier, 2016-12-25:
So, after more investigations...

The `qh=0` in `#0  0xc04039a8 in assign_and_init_hc (hcd=0xdafae400, qh=0x0 <__vectors_start>) at drivers/usb/host/dwc_otg/dwc_otg_hcd.c:1204` is a red herring: just r1 being being reused early in the function.

The issue seems to revolve around `free_hc_list` getting broken, until the first entry has null pointers in its `hc_list_entry`. At a glance, the circular list operations are fine. The pattern seems to be producer/consumer, producer being an interrupt handler (and related ddma code), and consumer being dwc_otg_hcd.c, initialising the list on initialisation, filling it on hcd role start and "doing stuff" when the free list is not empty. I did not check locking yet, and am trying to see if there would be a "surprise" re-init which could confuse the data structure. I'm not sure how to add tests for locking.

vpelletier, 2017-01-23:
For the reference, I switched to dwc2 driver (loading the proper devicetree overlay), and did not have a crash so far. If I understand correctly, the dwc2 module lacks some ARM-specific IRQ-ish mechanism the out-of[-vanilla]-tree driver has. As the hc_ptr_array items are produced in the interrupt handler it makes some sense.

At a glance, I did not notice a system load increase, but this rpi is comfortably overspec for what it does so it would take a large one for an increase to be noticeable.

jkuek, 2017-01-23:
Thanks for the update. Last time I tried the dwc2 driver, I had some problems with other USB full-speed devices, eg. keyboards. I'm not sure whether that is still the case.

P33M, 2017-05-04:
There's been several bugfixes that may have a bearing on this issue. Please retest with a Pi that has latest rpi-update kernel.

jkuek, 2017-05-08:
I re-ran the tests today, and could still cause the system to throw a kernel panic.

I've attached the console output. 
[console.txt](https://github.com/raspberrypi/linux/files/982490/console.txt)

For completeness, here is my USB set-up:
```
pi@raspberrypi:~$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 6, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 4, If 1, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 4, If 2, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 4, If 0, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 4, If 7, Class=Vendor Specific Class, Driver=sierra_net, 480M
        |__ Port 4: Dev 4, If 3, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 5: Dev 5, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 5, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 5, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 5, If 3, Class=Human Interface Device, Driver=usbhid, 12M
pi@raspberrypi:~$ lsusb
Bus 001 Device 005: ID 0d8c:0016 C-Media Electronics, Inc.
Bus 001 Device 004: ID 1199:68a3 Sierra Wireless, Inc. MC8700 Modem
Bus 001 Device 006: ID 1a2c:0021 China Resource Semico Co., Ltd Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

To trigger the panic, I am running this script:
```
#!/bin/bash
tail -f /var/log/syslog &

while :
do
	./usb
	sleep 10
done
```
which calls this script:
```
echo 0 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
sleep 5
echo 1 >  /sys/bus/platform/drivers/dwc_otg/20980000.usb/buspower
```
I ran it a few times, the issue appears to happen randomly

P33M, 2017-05-08:
I'm running the same script (with a few modifications) with 3x full-speed devices and a keyboard in addition to a connected eth0 interface.

I'm not seeing a crash, which implies that whatever happening is a function of what devices you have connected (:()
What does userspace do when the devices get connected? Do programs start automatically using the devices?

P33M, 2017-05-08:
Ah. I have seen one crash after ~30 minutes of runtime. Investigating...

P33M, 2017-05-08:
I've also seen two crashes where in dwc_otg_hcd_handle_hc_fsm(), qh->qtd_list ends up being NULL. Unsure if this is related to having a corrupted free_hc_list.

I have found that hcd->available_host_channels gets steadily incremented above the hardware maximum of 8 channels as we potentially increment it twice when dequeueing a transfer currently being performed by the FIQ. This may or may not be enough to trigger double-frees/double-adds in free_hc_list...

P33M, 2017-05-08:
Smoking gun located: we modify host channel lists, queue heads and URB lists in `kill_urbs_in_qh_list()`, which is called from `cil_hcd_disconnect()`. It appears that we do this without taking either the HCD lock or the host channel lock.

jkuek, 2017-05-09:
Sounds like great progress. As I've mentioned previously, different combinations of USB devices do seem to influence how often the crash occurs, but I've seen it fail with just a USB keyboard attached.

jkuek, 2017-05-09:
@P33M There was a comment in this issue from HiassofT on 4 Feb 2016. I don't claim to understand it, but it might help get to the bottom of what's going on.

P33M, 2017-05-09:
The more I dig the more wrong-ness I keep finding...
- As far as I can tell, hcd->channel_lock is basically useless as it's only held when the hcd->lock is being held.
- There are multiple call sites for cil_hcd_disconnect(), and depending on the state of the OTG core (device, B host or A host) the hcd->lock is held in A host but dropped in the other two cases (?)
- It's possible to queue a FIQ split transaction with the state machine not in the idle state. This is easily reproducible as the second or third transfer when enumerating any low- or full-speed device typically ends up in this state. This is bad because based on my current understanding, the host channel shouldn't even be considered for a transfer if one is still running.

P33M, 2017-05-10:
As far as point 2) goes, cil_hcd_disconnect() is called while holding the _core_ lock but not the _hcd_ lock. After adding the HCD lock inside the disconnect callback, I no longer get spurious crashes in dwc_otg_hcd_handle_hc_fsm() (or at least, I've not seen any more) but the crash due to a corrupt free_hc_list in assign_and_init_hc() is still there. 
`hcd->available_host_channels` is still getting incremented above the hardware maximum which is baffling.

I think I will swap away from root port disconnect crashes while I have a more reproducible bug pending (queueing transactions without a sane initial FIQ state).

P33M, 2017-05-12:
Point 3) is a small wild goose chase - because there are only two non-periodic transaction queues (inactive and active), on each scheduling call `dwc_otg_queue_transactions()` will repeatedly process queue heads that have already been assigned to a channel. I've replicated the early-out mechanism for FIQ enabled channels.

I've fixed several potential crash-inducing conditions due to the FIQ racing against `cil_hcd_disconnect()` and as far as I can tell, the bug has gone down in severity but is still present - now I get a much longer interval between crashes in `assign_and_init_hc()`, but before then the driver "forgets" about host channels that it has available.

P33M, 2017-05-15:
I believe I have a fix for the list corruption - in `kill_all_urbs()->kill_urbs_in_qh_list()`, if a transfer is in progress, the channel is forcibly halted but HAINTMSK is not altered. Subsequently in `dwc_otg_hcd_disconnect_cb()` the channel is unconditionally cleaned up (all interrupt sources masked) and reinserted into free_hc_list.

The corruption happens because if the CPU is slow in executing the second unnecessary disable, a bit in HAINT is set as the channel already halted, which then triggers `...handle_hc_n_intr()` which results in the channel being inserted into the list a second time. 

The fix is to just let the dequeue code path run to completion and delete the second cleanup.

jkuek, 2017-05-16:
I have run the test script for a few hours - so far so good.
I notice a number of new error messages show up in syslog:

```
May 16 14:18:23 raspberrypi kernel: [ 8600.477366] ERROR::dwc_otg_hcd_urb_enqueue:487: Not connected
May 16 14:18:23 raspberrypi kernel: [ 8600.477366] 
May 16 14:18:23 raspberrypi kernel: [ 8600.477481] input: USB USB Keykoard as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.2/1-1.2:1.1/0003:1A2C:0021.0006/input/input5
May 16 14:18:23 raspberrypi kernel: [ 8600.477542] ERROR::dwc_otg_hcd_urb_enqueue:487: Not connected
May 16 14:18:23 raspberrypi kernel: [ 8600.477542] 
May 16 14:18:23 raspberrypi kernel: [ 8600.505325] ERROR::dwc_otg_hcd_urb_enqueue:487: Not connected
May 16 14:18:23 raspberrypi kernel: [ 8600.505325] 
May 16 14:18:24 raspberrypi kernel: [ 8600.545356] ERROR::dwc_otg_hcd_urb_enqueue:487: Not connected
May 16 14:18:24 raspberrypi kernel: [ 8600.545356] 
```

I've left the unit to run overnight, will report again tomorrow.

As an aside, I've seen issues on current 4.4.x kernels where sometimes disconnecting/reconnecting my 3G modem can seemingly cause all other USB devices to disconnect and reconnect. I had put this down as a probably hardware problem (perhaps power-related?), but have you seen anything in your recent travels that might suggest a software reason?

P33M, 2017-05-16:
3G modems typically play fast and loose with the USB maximum Vbus current draw (500mA). Peak loading when the modem is transmitting can exceed several watts.

I can well imagine that the power supplies draw a surge on hotplug that upsets Vbus on the Pi.

I've run the disconnect/reconnect script overnight with 3 full-speed devices plugged in, no crashes resulted. Closing as fixed.

closed 2017-05-16 by P33M

jkuek, 2017-05-16:
Ah yes, but in my case the modem is externally powered, and it's just a USB connection to the Pi.

I concur with your findings, I ran the script overnight and the unit is still running! Thanks.

Will those syslog error messages be in the official release? The additional extra empty line could be removed.

jkuek, 2017-05-17:
@P33M just noticed one issue:

I stopped the script after running it for > 24 hours, but my USB network and 3G devices were no longer working.

My USB devices were present:

```
pi@raspberrypi:~$ lsusb
Bus 001 Device 100: ID 0d8c:0016 C-Media Electronics, Inc.
Bus 001 Device 099: ID 1199:68a3 Sierra Wireless, Inc. MC8700 Modem
Bus 001 Device 101: ID 1a2c:0021 China Resource Semico Co., Ltd Keyboard
Bus 001 Device 098: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 097: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~$ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 97, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 98, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 101, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 101, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 99, If 1, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 99, If 2, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 99, If 0, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 4: Dev 99, If 7, Class=Vendor Specific Class, Driver=sierra_net, 480M
        |__ Port 4: Dev 99, If 3, Class=Vendor Specific Class, Driver=sierra, 480M
        |__ Port 5: Dev 100, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 100, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 100, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 5: Dev 100, If 3, Class=Human Interface Device, Driver=usbhid, 12M
pi@raspberrypi:~$
```

However, neither eth0 nor wwan0 (3g modem) are accessible.
```
pi@raspberrypi:~$ ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:90 errors:0 dropped:0 overruns:0 frame:0
          TX packets:90 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:12554 (12.2 KiB)  TX bytes:12554 (12.2 KiB)
pi@raspberrypi:~$ip addr show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
12436: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:8f:13:fc brd ff:ff:ff:ff:ff:ff
12437: wwan0: <BROADCAST,MULTICAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether de:65:07:14:49:07 brd ff:ff:ff:ff:ff:ff
```

I checked the syslog and the dhcpcd.service failed about 2 hours after starting the test

```
May 16 16:35:31 raspberrypi kernel: [ 8074.530821] Indeed it is in host mode hprt0 = 00021501
May 16 16:35:31 raspberrypi kernel: [ 8074.730699] usb 1-1: new high-speed USB device number 18 using dwc_otg
May 16 16:35:31 raspberrypi kernel: [ 8074.730914] Indeed it is in host mode hprt0 = 00001101
May 16 16:35:31 raspberrypi kernel: [ 8074.961137] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
May 16 16:35:31 raspberrypi kernel: [ 8074.961160] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
May 16 16:35:31 raspberrypi kernel: [ 8074.977154] hub 1-1:1.0: USB hub found
May 16 16:35:31 raspberrypi kernel: [ 8074.977326] hub 1-1:1.0: 5 ports detected
May 16 16:35:32 raspberrypi kernel: [ 8075.290717] usb 1-1.1: new high-speed USB device number 19 using dwc_otg
May 16 16:35:32 raspberrypi kernel: [ 8075.421146] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
May 16 16:35:32 raspberrypi kernel: [ 8075.421169] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
May 16 16:35:32 raspberrypi kernel: [ 8075.424638] smsc95xx v1.0.5
[... 82 lines cut ...]
May 16 16:35:41 raspberrypi kernel: [ 8084.392216] sierra ttyUSB2: Sierra USB modem converter now disconnected from ttyUSB2
May 16 16:35:41 raspberrypi kernel: [ 8084.392336] sierra 1-1.4:1.2: device disconnected
May 16 16:35:41 raspberrypi kernel: [ 8084.402119] sierra ttyUSB3: Sierra USB modem converter now disconnected from ttyUSB3
May 16 16:35:41 raspberrypi kernel: [ 8084.402246] sierra 1-1.4:1.3: device disconnected
May 16 16:35:41 raspberrypi kernel: [ 8084.402711] sierra_net 1-1.4:1.7 wwan0: unregister 'sierra_net' usb-20980000.usb-1.4, Sierra Wireless USB-to-WWAN Modem
May 16 16:35:41 raspberrypi kernel: [ 8084.412115] sierra_net 1-1.4:1.7 wwan0 (unregistered): usb_control_msg failed, status -19
May 16 16:35:41 raspberrypi kernel: [ 8084.413073] usb 1-1.5: USB disconnect, device number 21
May 16 16:35:41 raspberrypi systemd[1]: dhcpcd.service: main process exited, code=killed, status=11/SEGV
May 16 16:35:41 raspberrypi dhcpcd[13144]: sending signal TERM to pid 514
May 16 16:35:41 raspberrypi dhcpcd[13144]: dhcpcd not running
May 16 16:35:41 raspberrypi dhcpcd[13144]: dhcpcd[13144]: sending signal TERM to pid 514
May 16 16:35:41 raspberrypi dhcpcd[13144]: dhcpcd[13144]: dhcpcd not running
May 16 16:35:41 raspberrypi systemd[1]: dhcpcd.service: control process exited, code=exited status=1
[... cut ...]
May 16 16:35:42 raspberrypi ntpd[538]: Deleting interface #1528 wwan0, 100.74.94.245#123, interface stats: received=4, sent=4, dropped=0, active_time=8 secs
May 16 16:35:42 raspberrypi ntpd[538]: 122.252.184.186 interface 100.74.94.245 -> (none)
May 16 16:35:43 raspberrypi ntpd[538]: 103.242.70.5 interface 100.74.94.245 -> (none)
May 16 16:35:43 raspberrypi ntpd[538]: 130.217.226.50 interface 100.74.94.245 -> (none)
May 16 16:35:43 raspberrypi ntpd[538]: 203.97.213.52 interface 100.74.94.245 -> (none)
May 16 16:35:43 raspberrypi ntpd[538]: peers refreshed
```

After restarting dhcpcd, everything is ok again.

P33M, 2017-05-17:
So eth0 came back without a reboot, correct?

jkuek, 2017-05-17:
as reported, eth0 seemed to be ok from a driver perspective (it was present in /sys/class/net) but did not show in the list when I ran ifconfig. 
Everything was fine again after starting dhcpcd.service (without rebooting the device)

dimaqq, 2017-09-21:
@P33M do you know if your fix (or some other fix) landed upstream?

I'm having similar hangs on desktop on stock 4.4.0 and custom USB device that's externally powered.

P33M, 2017-09-21:
The fixes are specific to the dwc_otg driver, which is a downstream implementation.

Also, unless your desktop hardware is powered by a mobile phone SoC using an OTG USB core, it's highly unlikely that this issue is the one you're seeing.

dimaqq, 2017-09-21:
Ugh, thanks, I'll keep digging!

l-spitz, 2017-09-26:
@P33M I'm getting a lot of issues similar to what you describe, when running webcams over the usb, namely sporadically get dmesg messages, and eventually get following, which requires a power cycle to recover from.  

I'm running, 
Linux version 4.9.41-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1023 SMP Tue Aug 8 16:00:15 BST 2017

Is the fix you mention in this release? 

`
[25134.316545] Unable to handle kernel NULL pointer dereference at virtual address 00000104
[25134.319093] pgd = 80004000
[25134.320352] [00000104] *pgd=00000000
[25134.321583] Internal error: Oops: 817 [#1] SMP ARM
[25134.322799] Modules linked in: evdev snd_usb_audio snd_hwdep snd_usbmidi_lib uvcvideo snd_rawmidi snd_seq_device videobuf2_vmalloc videobuf2_memops snd_pcm videobuf2_v4l2 snd_timer videobuf2_core snd videodev media cmac bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 rfkill bcm2835_gpiomem i2c_bcm2835 uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
[25134.329754] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.41-v7+ #1023
[25134.331237] Hardware name: BCM2835
[25134.332698] task: 80c06cc0 task.stack: 80c00000
[25134.334163] PC is at uvc_queue_next_buffer+0x7c/0xf0 [uvcvideo]
[25134.335611] LR is at 0x100
[25134.337000] pc : [<7f4463dc>]    lr : [<00000100>]    psr: 60000193
               sp : 80c01d48  ip : 00000100  fp : 80c01d64
[25134.339793] r10: b67be400  r9 : b73a0b98  r8 : 00000002
[25134.341171] r7 : bad41800  r6 : b73a0ce8  r5 : b73a0b98  r4 : b3844c00
[25134.342544] r3 : b73a0b98  r2 : 00000200  r1 : 00000000  r0 : 60000193
[25134.343899] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[25134.345250] Control: 10c5383d  Table: 373d006a  DAC: 00000055
[25134.346573] Process swapper/0 (pid: 0, stack limit = 0x80c00210)
[25134.347926] Stack: (0x80c01d48 to 0x80c02000)
[25134.349287] 1d40:                   000ea600 b3844c00 b73a0800 b67be420 80c01d9c 80c01d68
[25134.352041] 1d60: 7f449b0c 7f44636c 7f4536c0 b3844c00 00000000 b3844c00 b67be400 b73a0800
[25134.354782] 1d80: b73a0ce8 80b79340 00000000 00000001 80c01dbc 80c01da0 7f4486b8 7f4499a8
[25134.357512] 1da0: b67be400 60000113 00000000 00000000 80c01ddc 80c01dc0 80557f5c 7f448640
[25134.360270] 1dc0: b346c4c0 b67be400 b67be400 00000000 80c01dfc 80c01de0 80558088 80557ee8
[25134.363150] 1de0: baad6f30 b346c4c0 b9538000 b67be400 80c01e24 80c01e00 805820c8 80558048
[25134.366144] 1e00: 80c01e44 20000113 804ec0a0 b9463940 b9463944 00000000 80c01e34 80c01e28
[25134.369264] 1e20: 80591178 80582068 80c01e5c 80c01e38 80123144 80591164 801230d0 80c02080
[25134.372539] 1e40: 00000000 80c00020 00000100 b9c05180 80c01ec4 80c01e60 8010169c 801230dc
[25134.375884] 1e60: 801741b4 80177910 00000001 00200100 80c02d00 0025e4d4 0000000a 80c7b980
[25134.379313] 1e80: 80c71f38 80c0312c 80b793b8 80c02080 80c01e60 00000001 80173074 80b7cec0
[25134.382807] 1ea0: 00000000 00000000 00000001 b9c05180 80c00000 babffa80 80c01ed4 80c01ec8
[25134.386318] 1ec0: 80122cd8 8010151c 80c01efc 80c01ed8 801736b8 80122bd8 80c01f18 80c04670
[25134.389831] 1ee0: 60000013 ffffffff 80c01f4c 80c15f08 80c01f14 80c01f00 8010150c 80173654
[25134.393347] 1f00: 80c040a4 80108a58 80c01f74 80c01f18 8071a47c 80101470 00000000 ba35c470
[25134.396863] 1f20: 00000000 00000000 80c00000 80c0312c 00000001 80c03198 80c15f08 80c15f08
[25134.400380] 1f40: babffa80 80c01f74 80c040a4 80c01f68 80108a54 80108a58 60000013 ffffffff
[25134.403896] 1f60: 80108a54 00000000 80c01f84 80c01f78 80719bc4 80108a30 80c01f9c 80c01f88
[25134.407409] 1f80: 80161df0 80719b9c 808023f0 ffffffff 80c01fac 80c01fa0 80714450 80161d18
[25134.410923] 1fa0: 80c01ff4 80c01fb0 80b00d28 807143d8 ffffffff ffffffff 00000000 80b006f8
[25134.414436] 1fc0: 00000000 80b56a30 00000000 80c7b294 80c0311c 80b56a2c 80c089a8 0000406a
[25134.417951] 1fe0: 410fd034 00000000 00000000 80c01ff8 0000807c 80b00980 00000000 00000000
[25134.421514] [<7f4463dc>] (uvc_queue_next_buffer [uvcvideo]) from [<7f449b0c>] (uvc_video_decode_isoc+0x170/0x198 [uvcvideo])
[25134.425101] [<7f449b0c>] (uvc_video_decode_isoc [uvcvideo]) from [<7f4486b8>] (uvc_video_complete+0x84/0xec [uvcvideo])
[25134.428676] [<7f4486b8>] (uvc_video_complete [uvcvideo]) from [<80557f5c>] (__usb_hcd_giveback_urb+0x80/0x160)
[25134.432217] [<80557f5c>] (__usb_hcd_giveback_urb) from [<80558088>] (usb_hcd_giveback_urb+0x4c/0xfc)
[25134.435753] [<80558088>] (usb_hcd_giveback_urb) from [<805820c8>] (completion_tasklet_func+0x6c/0x98)
[25134.439300] [<805820c8>] (completion_tasklet_func) from [<80591178>] (tasklet_callback+0x20/0x24)
[25134.442850] [<80591178>] (tasklet_callback) from [<80123144>] (tasklet_hi_action+0x74/0x10c)
[25134.446398] [<80123144>] (tasklet_hi_action) from [<8010169c>] (__do_softirq+0x18c/0x3cc)
[25134.449948] [<8010169c>] (__do_softirq) from [<80122cd8>] (irq_exit+0x10c/0x168)
[25134.453494] [<80122cd8>] (irq_exit) from [<801736b8>] (__handle_domain_irq+0x70/0xc4)
[25134.457052] [<801736b8>] (__handle_domain_irq) from [<8010150c>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[25134.460637] [<8010150c>] (bcm2836_arm_irqchip_handle_irq) from [<8071a47c>] (__irq_svc+0x5c/0x7c)
[25134.464210] Exception stack(0x80c01f18 to 0x80c01f60)
[25134.465988] 1f00:                                                       00000000 ba35c470
[25134.469402] 1f20: 00000000 00000000 80c00000 80c0312c 00000001 80c03198 80c15f08 80c15f08
[... cut ...]
[25134.476105] [<8071a47c>] (__irq_svc) from [<80108a58>] (arch_cpu_idle+0x34/0x4c)
[25134.479442] [<80108a58>] (arch_cpu_idle) from [<80719bc4>] (default_idle_call+0x34/0x48)
[25134.482789] [<80719bc4>] (default_idle_call) from [<80161df0>] (cpu_startup_entry+0xe4/0x160)
[25134.486143] [<80161df0>] (cpu_startup_entry) from [<80714450>] (rest_init+0x84/0x88)
[25134.489488] [<80714450>] (rest_init) from [<80b00d28>] (start_kernel+0x3b4/0x3c0)
[25134.492827] Code: e594215c e594c158 e1a03005 e3a0ec01 (e58c2004)
`