raspberrypi/linux #1273: missing device node for cpu ? But it still works

microaaron, 2016-01-28:
I compiled the kernel with bcm2709_defconfig. But when I boot the RPi2 with the new kernel, It appeared: 
[    0.053394] missing device node for CPU 0
[    0.053470] missing device node for CPU 1
[    0.053498] missing device node for CPU 2
[    0.053525] missing device node for CPU 3
But it looks still works. It still can enter the desktop.
I found this kernel I build is smaller than another kernel in github.com/raspberrypi/firmware, that kernel won't miss device node for CPU.
So I  guess that kernel was compiled with a different config. But how to config the kernel to let it not missing device node for CPU?

popcornmix, 2016-01-29:
First did you follow: https://www.raspberrypi.org/documentation/linux/kernel/building.md ?
That should produce a kernel with same behaviour as one from firmware repo.

microaaron, 2016-01-30:
Thanks for your help. I followed the way and tried again. But it still happeded.
[image]

popcornmix, 2016-01-30:
Are you using the rpi-4.1.y tree?
what does ``knlinfo kernel.img`` report?
Is device-tree present when you have booted (i.e. does ``/proc/device-tree`` exist)?

microaaron, 2016-01-30:
Sorry, I forgot to mkknlimg the zImage, so knlinfo report no trailer. Thak you. :smile:

closed 2016-01-30 by microaaron