raspberrypi/linux #1275: Please add realtime support

unkissedfrog, 2016-01-31:
Hello, 

please add the possibility to get a realtime (full preemtive) kernel for the raspberry. That would be great to get a low latency audio player :-)

popcornmix, 2016-07-26:
Not going to happen with official kernel.
I'd suggest using a custom kernel build. E.g. http://www.frank-durr.de/?p=203

closed 2016-07-26 by popcornmix

tx0h, 2016-07-27:
yepp, but instead that everybody do it on their own it would be great to have that done right once and put it in the repository. that's the reason for repositories, i think.

Ferroin, 2016-07-27:
If by fullly preemptible kernel you mean one built with CONFIG_PREEMPT, then I'm pretty certain the official firmware already has that.

If you mean one using the -RT sources, or even some other third-party patches, then it's not going to happen because the overhead to maintain that far outweighs the benefit it would have to the primary target users.

Also, once you start talking real-time, 'done right' becomes heavily dependent on the use case.  A real-time kernel 'done right' for a low latency audio player is not the same as one 'done right' for a robot.

tx0h, 2016-07-27:
the kernel i run (raspbian testing) is 4.4.13-v7+ and has only CONFIG_PREEMPT_VOLUNTARY set, CONFIG_PREEMPT is unset. CONFIG_PREEMPT_RT wasn't demanded, yet.

popcornmix, 2016-07-27:
Provide some evidence of the benefits of ``CONFIG_PREEMPT=y`` (e.g. measured average latency before and after adding it) and it may be considered. But I suspect that setting alone will have little effect.
The -RT patches will never be included.

clivem, 2016-07-27:
Here we go again..... OP, read #1216, this resulted in the default config changing from PREEMPT to VOLUNTARY_PREEMPT ISTR.

tx0h, 2016-07-27:
that's why i opt for a second kernel package. some need more throughput, some need more responsiveness. like, if you play with electronic circuits it may be time critical but you don't need to pass a lot of data through the bus.

clivem, 2016-07-27:
> that's why i opt for a second kernel package ...

It's really simple. If you need something other than a kernel built with the Raspberry Pi default configs, then you build your own kernel with the options you want. The RPi guys have a "sane" default config. If you need something specialized for a particular use case, (like I do), you build your own, to your own preferences. It's not hard. ;)