raspberrypi/linux #1276: sdhost breaks after some heavy usb usage on at least 4.3 and 4.4

l29ah, 2016-02-01:
I attached an external hdd to a raspi and ran badblocks(1) on it. After several minutes/hours of running i keep getting this:
```
Feb 01 06:32:19 [kernel] [21249.788691] kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:887!
Feb 01 06:32:19 [kernel] [21249.800179] Modules linked in: lm75 hwmon bcm2835_gpiomem bcm2835_wdt bcm2835_rng w1_therm w1_gpio wire cn uio_pdrv_genirq uio
Feb 01 06:32:19 [kernel] [21249.811665] CPU: 0 PID: 69 Comm: mmcqd/0 Not tainted 4.4.0+ #10
Feb 01 06:32:19 [kernel] [21249.817580] Hardware name: BCM2708
Feb 01 06:32:19 [kernel] [21249.820984] task: ddf92940 ti: ddfa4000 task.ti: ddfa4000
Feb 01 06:32:19 [kernel] [21249.826406] PC is at bcm2835_sdhost_transfer_complete+0xec/0xf8
Feb 01 06:32:19 [kernel] [21249.832328] LR is at bcm2835_sdhost_finish_command+0x264/0x458
Feb 01 06:32:19 [kernel] [21249.838161] pc : [<c04637c8>]    lr : [<c04634e8>]    psr: a0000113
Feb 01 06:32:19 [kernel] [21249.838161] sp : ddfa5d90  ip : ddfa5da8  fp : ddfa5da4
Feb 01 06:32:19 [kernel] [21249.849626] r10: 60000113  r9 : ddfa5ebc  r8 : ddfa0534
Feb 01 06:32:19 [kernel] [21249.854847] r7 : c08bf15c  r6 : 0000270f  r5 : 00000000  r4 : ddf7d260
Feb 01 06:32:19 [kernel] [21249.861367] r3 : 00000007  r2 : ddfa04b8  r1 : 00000000  r0 : ddf7d260
Feb 01 06:32:19 [kernel] [21249.867889] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Feb 01 06:32:19 [kernel] [21249.875192] Control: 00c5387d  Table: 1c270008  DAC: 00000055
Feb 01 06:32:19 [kernel] [21250.054799] [<c04637c8>] (bcm2835_sdhost_transfer_complete) from [<c04634e8>] (bcm2835_sdhost_finish_command+0x264/0x458)
Feb 01 06:32:19 [kernel] [21250.065764] [<c04634e8>] (bcm2835_sdhost_finish_command) from [<c04643e0>] (bcm2835_sdhost_request+0x1a0/0x288)
Feb 01 06:32:19 [kernel] [21250.075865] [<c04643e0>] (bcm2835_sdhost_request) from [<c044897c>] (__mmc_start_request+0x50/0xe0)
Feb 01 06:32:19 [kernel] [21250.084919] [<c044897c>] (__mmc_start_request) from [<c0448b0c>] (mmc_start_request+0x100/0x134)
Feb 01 06:32:19 [kernel] [21250.093711] [<c0448b0c>] (mmc_start_request) from [<c0449a80>] (mmc_start_req+0x30c/0x454)
Feb 01 06:32:19 [kernel] [21250.101987] [<c0449a80>] (mmc_start_req) from [<c045989c>] (mmc_blk_issue_rw_rq+0xe0/0xbc4)
Feb 01 06:32:19 [kernel] [21250.110346] [<c045989c>] (mmc_blk_issue_rw_rq) from [<c045a488>] (mmc_blk_issue_rq+0x108/0x574)
Feb 01 06:32:19 [kernel] [21250.119051] [<c045a488>] (mmc_blk_issue_rq) from [<c045c028>] (mmc_queue_thread+0xe0/0x1a8)
```

pelwell, 2016-02-01:
Thanks - we're aware of this. A fix is being worked on.

jkuek, 2016-02-04:
This just happened to me:

`[   98.207964] kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:927!
[   98.218541] Internal error: Oops - BUG: 0 [#1] ARM
[   98.227765] Modules linked in: cfg80211 rfkill evdev joydev snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi snd_seq_device sierra usbserial sierra_net bcm2835_gpiomem snd_bcm2835 bcm2835_wdt snd_pcm snd_timer snd uio_pdrv_genirq uio
[   98.259024] CPU: 0 PID: 64 Comm: mmcqd/0 Tainted: G        W       4.4.1+ #835
[   98.271348] Hardware name: BCM2708
[   98.279730] task: dafc0da0 ti: dafb4000 task.ti: dafb4000
[   98.290319] PC is at bcm2835_sdhost_finish_command+0x3c0/0x468
[   98.301382] LR is at bcm2835_sdhost_transfer_complete+0xdc/0xf8
[   98.312396] pc : [<c0433d8c>]    lr : [<c0433f10>]    psr: 60000113
[   98.312396] sp : dafb5d50  ip : dafb5d78  fp : dafb5d74
[   98.334234] r10: dafdd488  r9 : dafae42c  r8 : 00000000
[   98.344648] r7 : daf90e60  r6 : 60000113  r5 : daf90c00  r4 : daf90e60
[   98.356489] r3 : 00000000  r2 : 00000008  r1 : daf90e90  r0 : daf90e60
[   98.368317] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[   98.380883] Control: 00c5387d  Table: 18a60008  DAC: 00000055
[   98.392050] Process mmcqd/0 (pid: 64, stack limit = 0xdafb4188)
[   98.403416] Stack: (0xdafb5d50 to 0xdafb6000)
[   98.413102] 5d40:                                     dafdd488 dafb5d8c daf90e60 daf90c00
[   98.426850] 5d60: 60000113 daf90e60 dafb5d8c dafb5d78 c0433f10 c04339d8 00000001 daf90e60
[   98.440640] 5d80: dafb5db4 dafb5d90 c0433bec c0433e40 dafdd488 dafb5de4 dafae42c daf90c00
[   98.454473] 5da0: 60000113 daf90e60 dafb5de4 dafb5db8 c0434a10 c04339d8 00000000 dafae42c
[   98.468200] 5dc0: dafdd488 dafb5e04 dafae42c daf90c00 dafae534 dafae42c dafb5e04 dafb5de8
[   98.481952] 5de0: c041a0d8 c0434894 dafae42c daf90c00 dafae534 dafb5eac dafb5e1c dafb5e08
[   98.495824] 5e00: c041a268 c041a094 daf90c00 00000000 dafb5e6c dafb5e20 c041b0d4 c041a174
[   98.509706] 5e20: dafae404 00000001 dafb5e6c dafb5e38 c042a680 c042db20 00000000 dafae400
[   98.523635] 5e40: dafae404 dafdd488 daf91400 dafdd488 dafae42c dafae404 daf91400 dafdd488
[   98.537610] 5e60: dafb5edc dafb5e70 c042a888 c041ae08 dafb5edc dafb5e80 c059a6a8 c00b3ae4
[   98.551478] 5e80: 00000000 00000200 daf90da8 00000000 00000000 00000000 00000000 dafae400
[   98.565415] 5ea0: 00000000 dafdd488 24590411 dafb5edc dafb5e70 dafae404 daf91400 dafdd488
[   98.579518] 5ec0: dafae400 dafae400 dae7f338 24590411 dafb5f2c dafb5ee0 c042b400 c042a7c4
[   98.593649] 5ee0: 00000000 00000000 00000000 c084af74 dafae404 dafb4000 00000000 daf90c00
[   98.607845] 5f00: 00000001 dafae404 dafb4000 00000000 24590411 00000001 dae7f338 dafdd488
[   98.622065] 5f20: dafb5f64 dafb5f30 c042cee8 c042b308 c042ce30 dafae40c 00000000 daf86340
[   98.636263] 5f40: 00000000 dafae404 c042ce30 00000000 00000000 00000000 dafb5fac dafb5f68
[   98.650476] 5f60: c003c6e8 c042ce3c 00000000 00000000 c003c608 dafae404 00000000 dafb5f7c
[   98.664758] 5f80: dafb5f7c 00000000 dafb5f88 dafb5f88 daf86340 c003c608 00000000 00000000
[   98.679219] 5fa0: 00000000 dafb5fb0 c000f608 c003c614 00000000 00000000 00000000 00000000
[   98.693667] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   98.708107] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   98.722495] [<c0433d8c>] (bcm2835_sdhost_finish_command) from [<c0433f10>] (bcm2835_sdhost_transfer_complete+0xdc/0xf8)
[   98.739598] [<c0433f10>] (bcm2835_sdhost_transfer_complete) from [<c0433bec>] (bcm2835_sdhost_finish_command+0x220/0x468)
[   98.756960] [<c0433bec>] (bcm2835_sdhost_finish_command) from [<c0434a10>] (bcm2835_sdhost_request+0x188/0x234)
[   98.773531] [<c0434a10>] (bcm2835_sdhost_request) from [<c041a0d8>] (__mmc_start_request+0x50/0xe0)
[   98.789119] [<c041a0d8>] (__mmc_start_request) from [<c041a268>] (mmc_start_request+0x100/0x134)
[   98.804424] [<c041a268>] (mmc_start_request) from [<c041b0d4>] (mmc_start_req+0x2d8/0x404)
[   98.819176] [<c041b0d4>] (mmc_start_req) from [<c042a888>] (mmc_blk_issue_rw_rq+0xd0/0xb44)
[   98.834019] [<c042a888>] (mmc_blk_issue_rw_rq) from [<c042b400>] (mmc_blk_issue_rq+0x104/0x508)
[   98.849187] [<c042b400>] (mmc_blk_issue_rq) from [<c042cee8>] (mmc_queue_thread+0xb8/0x15c)
[   98.864062] [<c042cee8>] (mmc_queue_thread) from [<c003c6e8>] (kthread+0xe0/0xfc)
[   98.878112] [<c003c6e8>] (kthread) from [<c000f608>] (ret_from_fork+0x14/0x2c)
[   98.891885] Code: 05931010 ebf287ae e59420d8 eaffff70 (e7f001f2)
[   98.904513] ---[ end trace ffd2d0f78e36e75d ]---`

popcornmix, 2016-02-13:
Can you test again with lastest kernel? (or using BRANCH=next rpi-update)?

l29ah, 2016-02-13:
Will rpi-4.4.y do?

popcornmix, 2016-02-13:
Yes, fix is on rpi-4.4.y tree.

l29ah, 2016-02-14:
Ran badblocks, got
```
Feb 14 01:11:04 [kernel] [19262.384853] usb 1-1.2: USB disconnect, device number 6
Feb 14 01:11:04 [kernel] [19262.403551] scsi 0:0:0:0: rejecting I/O to offline device
Feb 14 01:11:04 [kernel] [19262.403607] scsi 0:0:0:0: [sda] killing request
Feb 14 01:11:04 [kernel] [19262.403744] scsi 0:0:0:0: [sda] UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
Feb 14 01:11:04 [kernel] [19262.403788] scsi 0:0:0:0: [sda] CDB: opcode=0x28 28 00 0c 9f 9b 00 00 00 80 00
Feb 14 01:11:04 [kernel] [19262.403817] blk_update_request: I/O error, dev sda, sector 211786496
Feb 14 01:11:04 [kernel] [19262.404143] Buffer I/O error on dev sda, logical block 26473312, async page read
Feb 14 01:11:04 [kernel] [19262.404672] Buffer I/O error on dev sda, logical block 26473312, async page read
Feb 14 01:11:04 [kernel] [19262.404845] Buffer I/O error on dev sda, logical block 26473312, async page read
Feb 14 01:11:04 [kernel] [19262.404989] Buffer I/O error on dev sda, logical block 26473312, async page read
Feb 14 01:11:04 [kernel] [19262.405155] Buffer I/O error on dev sda, logical block 26473313, async page read
Feb 14 01:11:04 [kernel] [19262.405304] Buffer I/O error on dev sda, logical block 26473313, async page read
Feb 14 01:11:04 [kernel] [19262.405446] Buffer I/O error on dev sda, logical block 26473313, async page read
Feb 14 01:11:04 [kernel] [19262.405585] Buffer I/O error on dev sda, logical block 26473313, async page read
Feb 14 01:11:04 [kernel] [19262.405744] Buffer I/O error on dev sda, logical block 26473314, async page read
Feb 14 01:11:04 [kernel] [19262.405894] Buffer I/O error on dev sda, logical block 26473314, async page read
Feb 14 01:11:07 [kernel] [19264.653447] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
Feb 14 01:11:07 [kernel] [19264.755900] usb 1-1.2: New USB device found, idVendor=152d, idProduct=0551
Feb 14 01:11:07 [kernel] [19264.755946] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=5
Feb 14 01:11:07 [kernel] [19264.755971] usb 1-1.2: Product: USB to ATA/ATAPI Bridge
Feb 14 01:11:07 [kernel] [19264.755992] usb 1-1.2: Manufacturer: JMicron
Feb 14 01:11:07 [kernel] [19264.756011] usb 1-1.2: SerialNumber: 9E0ACFFFFFFF
Feb 14 01:11:07 [kernel] [19264.762973] usb-storage 1-1.2:1.0: USB Mass Storage device detected
Feb 14 01:11:07 [kernel] [19264.779227] scsi host1: usb-storage 1-1.2:1.0
Feb 14 01:11:09 [kernel] [19267.413482] buffer_io_error: 44903 callbacks suppressed
```
after several hours. Not sure if it's the kernel to blame tho as i've experienced the device going offline on an x86 usb host. Any ideas on handling such failures?

popcornmix, 2016-02-15:
This seems unrelated to the sdhost bug reported.
Please close this issue if you believe sdhost bug is resolved.

If you have a different problem then create a new issue.

l29ah, 2016-02-20:
Other devices seem to work fine. Thank you!

closed 2016-02-20 by l29ah

Tangerino, 2020-07-13:
Just started a new Focal 20 and ran this program

```python
import minimalmodbus

instrument = minimalmodbus.Instrument('/dev/ttyUSB0', 2)
instrument.debug = False
instrument.serial.baudrate = 9600
instrument.serial.timeout = 1
while True:
    instrument.read_long(229)
```

It crashes all the time in not regular intervals.

I'm not sure if this place is the right one to report this

The first error in the was
[image]

And later got this

[image]