raspberrypi/linux #1278: (testing) libldb1_1.1.24-1 breaks samba packages

ElTopo, 2016-02-01:
On raspbian "testing": recently (Jan 30, 2016), libldb1 upgraded to 1.1.24, unfortunately, most samba package still depend on 1.1.20:

    # sudo apt-get install -s samba-libs
    Reading package lists... Done
    Building dependency tree
    Reading state information... Done
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:
    
    The following packages have unmet dependencies:
     samba-libs : Depends: libldb1 (< 2:1.1.21~) but 2:1.1.24-1 is to be installed
    E: Unable to correct problems, you have held broken packages.

Please release new samba packages based on libldb1-1.1.24-1, right now there's no working samba can be installed on a raspbian testing system

XECDesign, 2016-02-01:
Hello, this repo is for the linux kernel. This sounds like a Debian problem, which can be reported here.
https://www.debian.org/Bugs/Reporting

closed 2016-02-01 by XECDesign

ElTopo, 2016-02-01:
@XECDesign I checked Debian testing, the official repository still use libldb1-1.1.20, and the official site does NOT have reaspbian repository.

XECDesign, 2016-02-01:
Raspbian bugs can be reported here https://www.raspbian.org/RaspbianBugs