raspberrypi/linux #1283: Renaming file to its own name raises error.

The--Joke, 2016-02-06:
Trying to rename a file, and giving it its original name and accepting will raise an error; "could not rename file; file already exists". This is unnecessary.

JamesH65, 2016-02-06:
How are you trying to do the rename? Using the file manager or command line?

To be honest, whichever you are using, neither are down to the Raspberry Pi foundation to fix - they are standard linux commands (command line) or LXDE functionality (desktop). This github project is for issues with the Linux OS on the Raspberry Pi.

closed 2016-02-06 by JamesH65

Ferroin, 2016-02-08:
Add to that, this is standard behavior on every UNIX derived OS in existence, and at a low level, on Windows as well.