raspberrypi/linux #1285: bcm2708_wdog.ko module not compiled since 4.3.3

magarto, 2016-02-10:
Hi,

Since 4.3.3 kernel release, bcm2708_wdog.ko is not compiled. The only compiled watchdog is for poweroff (bcm2835_wdt) for for reboot (bcm2708_wdog).

There is only a CONFIG_BCM2835_WDT=m in the 4.3 configuration.
https://github.com/raspberrypi/linux/blob/rpi-4.3.y/arch/arm/configs/bcm2709_defconfig#L634

I tried adding CONFIG_BCM2708_WDOG=m to the .config but it doesn't work.

How can I compile it? My goal is to apply the following rebooting watchdog:
http://blog.ricardoarturocabral.com/2013/01/auto-reboot-hung-raspberry-pi-using-on.html

pelwell, 2016-02-10:
Have you tried adding `dtparam=watchdog=on` to your config.txt? That will create /dev/watchdog without the need to edit /etc/modules. The rest of the instructions should then work.

Give it a try, then report back.

magarto, 2016-02-10:
So, it will create a /dev/watchdog, but with that I can use the hardware watchdog (not a software one)?

If so, the only thing I need is to edit /etc/watchdog adding?:
watchdog-device = /dev/watchdog

instead of:
watchdog-device

Thanks!

pelwell, 2016-02-10:
The watchdog is always a hardware device - you can't rely on software to run when the system gets stuck.

We are switching from a Pi-specific driver (bcm2708_wdog) to the Linux standard driver (bcm2835-wdog), but they do the same job. We have also changed to using Device Tree to describe the watchdog hardware (and cause the driver module to be loaded), rather than using Pi-specific initialisation code and /etc/modules.

Although the watchdog is hardware you still need to run some software to activate it, and to keep refreshing it to stop it resetting the board (like the movie Speed you can't go below 50mph). So install the watchdog package, edit /etc/watchdog and uncomment the watchdog-device line and then arrange that watchdog process is run at startup - on Jessie that should be a systemctl operation.

magarto, 2016-02-10:
Thanks for all!
I have problems with new OpenGL drivers but I will post in another thread.
One question: Linux 4.5 will include original Linux drivers instead of your patches?
Thanks!

pelwell, 2016-02-10:
Report OpenGL driver bugs here: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=135790

We will be using fewer and fewer downstream drivers - our rpi-4.4.y branch has already moved a long way towards upstream only, thanks to people like @notro and @msperl.

magarto, 2016-02-10:
Thanks for all your work and efficiency.
I close this issue

closed 2016-02-10 by magarto