raspberrypi/linux #1287: UNKNOWN(0x2003) SCSI Error

rickysarraf, 2016-02-10:
I've recently staretd seeing this error with my USB attached SATA 3.5" HDD. The device has been under use for more than a year. This error has been seen very recently. My guess, given the error code, and what I've understood from the conversation on debian-user [1], this could be a generic problem than being a hardware malfunction.

```
[62711.477903] sd 0:0:0:0: [sda] UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x00
[62711.485701] sd 0:0:0:0: [sda] CDB: opcode=0x28 28 00 9d 40 01 47 00 00 08 00
[62711.492910] blk_update_request: I/O error, dev sda, sector 2638217543
[84370.313684] sd 0:0:0:0: [sda] UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x00
[84370.321532] sd 0:0:0:0: [sda] CDB: opcode=0x28 28 00 a1 40 01 47 00 00 08 00
[84370.328721] blk_update_request: I/O error, dev sda, sector 2705326407
Reporting here hoping the latter.
```

[1] https://lists.debian.org/debian-user/2016/02/msg00194.html

popcornmix, 2016-02-10:
What does ``uname -a`` report?
If you believe it used to be more reliable, then find an older kernel that does work better.
You can find older kernel/firmware releases here:
https://github.com/Hexxeh/rpi-firmware/commits/master

click on a commit and you will the git hash at the end of the URL.
``sudo rpi-update <git hash>``
will get you that version.

If you can find an older version that doesn't have this issue that may give a starting point to diagnose.

Is this Pi1 or Pi2? How are Pi and disk powered? Insufficient power supply would be one possible cause of these errors. As would a failing disk drive.

rickysarraf, 2016-02-11:
Thank you for your reply.

On Wed, 2016-02-10 at 10:44 -0800, popcornmix wrote:
> What does uname -a report?

pi@pi:~$ uname -a
Linux pi 4.1.16-v7+ #833 SMP Wed Jan 27 14:32:22 GMT 2016 armv7l
GNU/Linux

> If you believe it used to be more reliable, then find an older kernel
> that does work better.

I'm trying to recollect what all was changed recently. One item that
came to mind was smartmontools. I've purged it now and am waiting
(24+hrs) to see if the error pops up again.

> You can find older kernel/firmware releases here:
> https://github.com/Hexxeh/rpi-firmware/commits/master
> click on a commit and you will the git hash at the end of the URL.
> sudo rpi-update <git hash>
> will get you that version.

This is good. Wasn't aware of it. Thanks for sharing this information.

> If you can find an older version that doesn't have this issue that
> may give a starting point to diagnose.
> Is this Pi1 or Pi2? How are Pi and disk powered? Insufficient power
> supply would be one possible cause of these errors. As would a
> failing disk drive.

This is an RPi2. The Pi is powered with its dedicated power cord, which
further is connected to my UPS. The USB HDD Drives are 3.5" disks and
they too have dedicated external power connected to the UPS.

With smartmontools purged, I'm still monitoring. I'll post my findings
here on this tracker.

-- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."

mathieugouin, 2016-07-21:
Hello,
I was wondering if you have updates for this issue because I'm experiencing very similar symptoms.

If it can help, I'm running a pi 3 with Seagate backup plus slim 2TB.

`
[75617.284826] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[75617.284849] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] [descriptor]
[75617.284857] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0
[75617.284869] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0xa1 a1 06 20 da 00 00 4f c2 00 b0 00 00
[77417.115145] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[77417.115163] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] [descriptor]
[77417.115172] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0
[77417.115184] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0xa1 a1 06 20 00 00 00 00 00 00 e5 00 00
`

rickysarraf, 2016-07-21:
@mathieugouin  There are a couple of things I did.

1. Ensured that the USB cable from the HDD to the RPi was not on top of the device. Turns out, both the devices generate enough heat.

2. Make some air vent for the HDD device.

Point no 1 and 2 are just physical precautions. I don't have data to prove if they have made any difference.

3. Given that my HDD and RPi had only 5 mins of UPS Backup, I ended up writing a vUPS (Virtual UPS) script to detect and cleanly unmount my HDD before the battery ran out and the RPi (and HDD) got off.

Maybe point no. 3 has made the difference. I haven't seen those errors since these changes, which now is around 4+ months.

rickysarraf, 2016-07-21:
I just did a quick 10 GiB write on the drive. And no SCSI errors. So it could very well be one of the points above, or just a fix with the newer kernel.

@mathieugouin What kernel are you on? And what kernel did you see the errors on?

```
pi@pi:/media/SEAGATE$ dd if=/dev/zero of=foo.img bs=1M count=10000; rm -f foo.img
10000+0 records in
10000+0 records out
10485760000 bytes (10 GB) copied, 404.873 s, 25.9 MB/s

pi@pi:/media/SEAGATE$ dmesg | tail 
[46982.945998] smsc95xx 1-1.1:1.0 eth0: link down
[46984.522834] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133370.215425] smsc95xx 1-1.1:1.0 eth0: link down
[133374.311640] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133375.839476] smsc95xx 1-1.1:1.0 eth0: link down
[133377.311895] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x5DE1
[133379.336016] smsc95xx 1-1.1:1.0 eth0: link down
[133381.600670] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133382.920601] smsc95xx 1-1.1:1.0 eth0: link down
[133384.496955] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
```

mathieugouin, 2016-07-21:
I'm using

Linux xbian 4.4.14+ #1 SMP PREEMPT Wed Jun 29 13:04:58 CEST 2016 armv7l GNU/Linux

rickysarraf, 2016-07-23:
I am not sure. I do have my router rebooting once a day, but these links falling are much more frequent. And that many link changes have happened in the uptime of 3 days, so there definitely is another bug too.

But, btw, I did see the SCSI error in the logs again.

```
pi@pi:~$ uptime
 14:52:13 up 3 days, 22:53,  1 user,  load average: 0.92, 0.90, 0.89

[46970.248030] smsc95xx 1-1.1:1.0 eth0: link down
[46974.368925] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[46975.881075] smsc95xx 1-1.1:1.0 eth0: link down
[46977.345893] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x5DE1
[46979.385132] smsc95xx 1-1.1:1.0 eth0: link down
[46981.617818] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[46982.945998] smsc95xx 1-1.1:1.0 eth0: link down
[46984.522834] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133370.215425] smsc95xx 1-1.1:1.0 eth0: link down
[133374.311640] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133375.839476] smsc95xx 1-1.1:1.0 eth0: link down
[133377.311895] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x5DE1
[133379.336016] smsc95xx 1-1.1:1.0 eth0: link down
[133381.600670] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[133382.920601] smsc95xx 1-1.1:1.0 eth0: link down
[133384.496955] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[219770.013955] smsc95xx 1-1.1:1.0 eth0: link down
[219774.078746] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[219775.638209] smsc95xx 1-1.1:1.0 eth0: link down
[219777.126994] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x5DE1
[219779.159523] smsc95xx 1-1.1:1.0 eth0: link down
[219781.367303] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[219782.687242] smsc95xx 1-1.1:1.0 eth0: link down
[219784.247568] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[219854.463396] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: delalloc,lazytime
[219855.590835] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: delalloc,lazytime
[280683.236435] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x00
[280683.245048] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x28 28 00 12 00 01 3f 00 00 08 00
[280683.252948] blk_update_request: I/O error, dev sdb, sector 301990207
[306168.788142] smsc95xx 1-1.1:1.0 eth0: link down
[306172.972925] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[306174.412760] smsc95xx 1-1.1:1.0 eth0: link down
[306175.901305] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x5DE1
[306177.901347] smsc95xx 1-1.1:1.0 eth0: link down
[306180.126169] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1
[306181.461791] smsc95xx 1-1.1:1.0 eth0: link down
[306183.022949] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x4DE1

pi@pi:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  5.3G  8.6G  38% /
devtmpfs        396M     0  396M   0% /dev
tmpfs           400M   41M  359M  11% /run
tmpfs           400M     0  400M   0% /sys/fs/cgroup
tmpfs            80M     0   80M   0% /run/user/1000
/dev/mmcblk0p1   57M   20M   37M  36% /boot
/dev/sdb1       1.8T  102G  1.7T   6% /media/SEAGATE
/dev/sda1       3.6T  2.9T  655G  82% /media/4TBWD
pi@pi:~$ 
```

mathieugouin, 2016-07-26:
I just upgraded to  Linux xbian 4.4.15+ and the issue is still present.

ezar, 2016-08-06:
Same here!
Using minibian 4.4.16+ 

[   12.255296] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[   12.256967] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] [descriptor] 
[   12.258878] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0 
[   12.260598] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
[   12.399218] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[   12.400713] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] [descriptor] 
[   12.402132] sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0 
[   12.403541] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x85 85 06 2c 00 00 00 00 00 00 00 00 00 00 00 e5 00

mathieugouin, 2016-08-08:
@ezar, just by curiosity what is the make and model of your USB drive?
Mine is Seagate Backup Slim 2TB.  It is currently formatted as NTFS.  Let's see if we can find commonalities...

ezar, 2016-08-08:
@mathieugouin mine is Seagate 3TB.
Model Family:     Seagate Barracuda 7200.14 (AF)
Device Model:     ST3000DM001-1ER166

It's formatted as EXT4.

mathieugouin, 2016-08-09:
I just found out by analyzing more the dmesg message that I always have the following 4 lines in block that are logged:

```
sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
sd 0:0:0:0: [sda] tag#0 Sense Key : 0x4 [current] [descriptor]
sd 0:0:0:0: [sda] tag#0 ASC=0x0 ASCQ=0x0
sd 0:0:0:0: [sda] tag#0 CDB: opcode=0xa1 a1 06 20 00 00 00 00 00 00 e5 00 00
```

By enabling the "human readable output", I was able to see that the blocks repeat roughly every 30 minutes...

`dmesg -T | grep tag#0 | grep UNKNOWN`

Could you check if you have something similar?

lowphas, 2016-08-09:
Hm, I am seeing in my rpi3-s dmesg the same lines for a while. (and never before)
My root fs is on an usb drive. Slightly overclocked, but with a dektop cpu cooler, with the official rpi3 power supply, with raspbian. I use it for light desktop workloads every day, plus as a wireless access point (via the onboard chip, bt disabled).

I have two and two "half" setup to reproduce this:

1. big old wd mybook (2?) with separate power adaptor (WD5000AAVS inside)
2. samsung 840 evo in a cheap noname (sharkoon, I think) bus powered case
3. samsung ssd with WDmybook's sata-to-usb controller (disassembled, then reassembled with the ssd)
4. samsung ssd with a slight modification on the bus powered case (usb cable extended with power and ground wires fed with an external power source /ssd side/ )

I cannot report any problems, data corruptions/loss, slowdowns or crashes.
From my dmesg: http://pastebin.com/atRCBkRQ

Linux ***** 4.4.16-v7+ #899 SMP Thu Jul 28 12:40:33 BST 2016 armv7l GNU/Linux
Bus 001 Device 005: ID 1058:1100 Western Digital Technologies, Inc. 

I hope it helps somewhat, maybe, I don't know.

ps: sorry for my English

Ruffio, 2016-08-17:
@rickysarraf has your issue been resolved? If so, please close this issue. Thanks.

rickysarraf, 2016-08-18:
@Ruffio Resolved? No. But is it really a bug? I don't know.

My inclination is that it is just may be a faulty hardware. But then, like I mentioned in the comments above, I've been able to avoid the scsi errors most of the times.

But given the number of users that have reported this issue, I'm not sure if this is a sign of faulty hardware.

rickysarraf, 2016-08-18:
I just wrote 80 GiB of Data to the drive, without any SCSI errors.
Given that it connects over USB to the RPi, I have my own hunch on what may have been the cause.

Feel free to close the bug if you want to, but it'd be nice if others could also comment first.

mathieugouin, 2016-08-21:
Please don't close the issue.  I still have the problem.  I'm glad to help but would require further guidance on what commands to run.  Thanks.

Kevin-McIsaac, 2016-08-21:
I'm seeing the same problem (output of log below)  on a new PI 3 running Linux dozer 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux

My device works fine for a day or two then the USB drive mount changes from RW to RO. When I umount the process hangs and I have to reboot. 

On reboot I run efsck and get this error "Inodes that were part of a corrupted orphan linked list found", which I then fix. Remounting etc, the machine runs fine for a few days then the problem repeats. 

**/var/log/messages** 
Aug 22 03:59:17 dozer kernel: [74516.369314] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Aug 22 03:59:17 dozer kernel: [74516.369358] sd 0:0:0:0: [sda] tag#0 Sense Key : 0x3 [current] 
Aug 22 03:59:17 dozer kernel: [74516.369379] sd 0:0:0:0: [sda] tag#0 ASC=0x11 ASCQ=0x0 
Aug 22 03:59:17 dozer kernel: [74516.369403] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 0a f2 24 c5 00 00 01 00

rickysarraf, 2016-08-22:
@mathieugouin If you really want to narrow down the bug, you may want to try out @popcornmix's suggestion about trying different revisions of the kernel. The details are mentioned in comment 1 of this bug.

Kevin-McIsaac, 2016-08-22:
After upgrading to the latest firmware I had the same problem about 12 hrs later.  I installed 0a3df181c71b02d933f4431e97a5ae7933041427 (committed on 24 Oct 2015) but by system now won't boot. I get the splash screen, and can get to NOOB. However when I exit the consol is blank and the system does not boot.

Any way to upgrade the firmware on a machine that only boots to the NOOB screen?

rickysarraf, 2016-08-22:
Sorry that you have an unbootalble state now.

In the past, I've done is to plug the sdcard into my laptop and them chroot
into it and then fix. May not be very straightforward as laptops usually
are x86.

s3nt fr0m a $martph0ne, excuse typ0s

Kevin-McIsaac, 2016-08-22:
All my fault :-) 

I found [this](http://elinux.org/R-Pi_Troubleshooting#Get_the_latest_firmware_version) but it seemed easier and more reliable to reinstall from NOOB.

How do I know which version of the firmware are compatable with my kernel version?

Kevin-McIsaac, 2016-08-22:
@mathieugouin my drive reports its a  ST2000DM001-1E6164, which I think is a Seagate Barracuda 2tb

Kevin-McIsaac, 2016-08-22:
This [thread](https://lists.debian.org/debian-user/2016/02/msg00208.html) discusses the same issue on a PI 2. 

Look like code 0x2003 should be FAILED, rather than UNKNOWN.

Still no resolution, but interesting to see this person's drive is also Seagate

ezar, 2016-08-23:
@Kevin-McIsaac Works with libata.force=1.5Gbp?

rickysarraf, 2016-08-23:
@Kevin-McIsaac  That person is me. :-)

 I'm not sure if that'd help but it'd be nice to get some users to confirm it. RPi uses USB1/2 which, by spec, should only be able to do 480 Mbps.

I've been using the workaround you've mentioned, for a bug in the sata driver, for my laptop.
https://www.researchut.com/blog/lenovo-yoga-2-13-debian

Kevin-McIsaac, 2016-08-23:
@ezar I'm not sure at this stage. The system just hung last night and the disk need fsck repairs. So probably not. 

I've placed an order for a new drive and will try that when it arrives

kristofejro, 2016-08-24:
I have the same problem in my RPi2

In syslog every 5 minutes there is:
```
Aug 24 14:23:09 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
Aug 24 14:23:09 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x4 [current] [descriptor]
Aug 24 14:23:09 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 ASC=0x0 ASCQ=0x0
Aug 24 14:23:09 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
Aug 24 14:23:10 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
Aug 24 14:23:10 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x4 [current] [descriptor]
Aug 24 14:23:10 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 ASC=0x0 ASCQ=0x0
Aug 24 14:23:10 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x4 [current] [descriptor]
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 ASC=0x0 ASCQ=0x0
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x4 [current] [descriptor]
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 ASC=0x0 ASCQ=0x0
Aug 24 14:28:17 raspberrypi kernel: sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
```

I am sure it was not showing earlier. 
I have 2.5 USB attached WD HDD and there is enough current because I have dedicated powered USB HUB.
HDD itself is new and there is no problem with it (no errors, SMART is clean)
I have just upgraded to 4.4.19-v7+ and problem still persits.

mathieugouin, 2016-08-24:
@kristofejro Out of curiosity, what OS are you using for your Pi2.  I'm using xbian.

It's interesting because you have the same repeating 4 lines as I do, but mine are every 30 min.

Could you try this and see the output:

`dmesg -T | grep tag#0 | grep UNKNOWN`

kristofejro, 2016-08-24:
@mathieugouin 
I run: Raspbian GNU/Linux 8.0 (jessie) 4.4.19-v7+
Just part of dmesg query:
```
[Wed Aug 24 18:13:39 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:13:39 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:18:49 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:18:49 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:23:59 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:23:59 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:29:01 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:29:02 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:34:10 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:34:10 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:39:20 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[Wed Aug 24 18:39:20 2016] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08

```

rickysarraf, 2016-08-24:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Wed, 2016-08-24 at 06:41 -0700, kristofejro wrote:
> I am sure it was not showing earlier. 
> I have 2.5 USB attached WD HDD and there is enough current because I have
> dedicated powered USB HUB.
> HDD itself is new and there is no problem with it (no errors, SMART is clean)
> I have just upgraded to 4.4.19-v7+ and problem still persits.

Honestly, I'm glad more and more people are hitting this bug. The best would be
to take this up on linux-scsi mailing list and use this bug report as a
reference.

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJXvdsNAAoJEKY6WKPy4XVpvocP+gMWoObesuNQRIKFWnozWuEq
t4R9NSpsETFAfJ3CHyX0W+uyjei4+btIyMeraDTUsE6oIZr/7xdLBLHJV8Kd2h9v
J1dO9omIor+nau6uCVmNDJQXIXfORj3IAmbOOywX0mkhHHNGCo+0/6ixcAKcJ16K
7iUWlEqylA0qfyLN/ZZGlspXii7I5QBzEsytOSBshKjNs94D55oaot8aOmMXNehY
kmVjNIGjHTAITKrpzlKSx27t7qAbqB5i09BIlZP75gqAggQuI5/1nKi0mlW9ZWzl
3rx1uHPpspCcg0E5YrY47ojm4tXQVvYcnvOPvsJ9HP+9F5y4Y0G4xXbVN14OcVxu
IxRaw/sznOMZlNIiUfpOaoM1byqlToHym5YxWDKXbWOrdXPz2QcCZpu5E95uXHzs
ENwcjHww5oA9RnpIQI3I8GfUQL87wZIvHhFtT59c7diDpEzLggXLhpwXKQnJ67Js
4fKw/goaJ4ETmxru4R9me3X1bs+SYZM5C20PwEDINjyFZ/uSCWyVqjxWRL3Etrwb
HpoG3fAlLMXCjqNWXtp2K/Y6Ir7TvE9TecxrgulUVBEyVAj99bHvMwm141FUXuk5
DIDwgG6xVDXIZ7rlzRMa6NumxugkrFnGwLgqq2OTxxe6cqe4CdUPZexvUTS3esdT
iiYxxYAXJc4tPx5ULusb
=dVry
-----END PGP SIGNATURE-----

kristofejro, 2016-08-24:
I have downgraded kernel to 4.4.13 and problem is gone (so far)
Mabe someone else can confirm (perform kernel downgrade).

Kevin-McIsaac, 2016-08-24:
@kristofejro , I'm running 4.4.13-v7+ #894 and still get the messages a few times a day.

@ezar , using libata.force=1.5Gbp did not eliminate the issue

kristofejro, 2016-08-30:
Isn't it very similar to: https://github.com/raspberrypi/linux/issues/1549 ?

ezar, 2016-09-11:
Same problem in 4.4.20+
Any news?

Dmole, 2016-11-04:
This is not raspberrypi specific.

AFAIK it's just that some drives do things the kernal is not expecting (someone should update a driver).

I get this (non-fatal warning) on a XU4 with a 5TB Seagate drives;
```
> uname -r
4.8.6+
> lsusb | grep Seagate
Bus 002 Device 006: ID 0bc2:3312 Seagate RSS LLC SRD00F2 Expansion Desktop Drive (STBV)
Bus 002 Device 005: ID 0bc2:3312 Seagate RSS LLC SRD00F2 Expansion Desktop Drive (STBV)
Bus 002 Device 004: ID 0bc2:3312 Seagate RSS LLC SRD00F2 Expansion Desktop Drive (STBV)
> dmesg |tail -n 4
[216006.391906] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x08
[216006.391924] sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x4 [current] [descriptor] 
[216006.391932] sd 2:0:0:0: [sdc] tag#0 ASC=0x0 ASCQ=0x0 
[216006.391945] sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x85 85 06 2c 00 da 00 00 00 00 00 4f 00 c2 00 b0 00
```

rickysarraf, 2016-11-04:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA512

On Thu, 2016-11-03 at 19:45 -0700, Tim wrote:
This is not raspberrypi specific.

AFAIK it's just that some drives do things the kernal is not expecting (someone should update a driver).

If you are certain this is a generic SCSI issue in the Linux kernel, could you
please report it upstream ?

- -- 
Ritesh Raj Sarraf
RESEARCHUT - http://www.researchut.com
"Necessity is the mother of invention."
-----BEGIN PGP SIGNATURE-----

iQIcBAEBCgAGBQJYHGEXAAoJEKY6WKPy4XVprGcP/3RH02JBNBM2GgSJkvLOVm9w
1LcSkPFmbK+QjrAXfyVDud7YCV1C98Ii9+ILca6ludfx1LxHCxPzJU5HyhWTjR+G
pdzexw2O32b+koUFDQ9oY1x4OqmzQo0iXRdqPY0dMurVy3ucMKXnsd36s0L5s9Va
XQxQvZcPaIXArkVi9nAiOTq2QjMZgXS7QP3KS9sxDqlq7+X0btVJiA8V3/904YG7
mUJ4a3kQSko6/ijREJS/C56VXe1cP6aFloy6e3QJdK89gyY5ADVgaKbFGJH5R3e3
BqrQIW6wtP+ihx+pS/THLwFA0OuqD9zOSHfeZZLU0JbrLt3T8Pnqqfa+sUSd5wlJ
cul59LynVXwiogOK4dfqZ47/QuzTd1ABlgXE6hIzwfzdlLTJ2BYBWJHz9kp5y2Yj
wXqA/YMEwz+3s2xH+8TJjVKq/PcV3JlN3EgaLvuipP3mS5y8+lLWfyoBLTB4VMQ9
ZZgBeRLo4Uz6kfE/Z3ssi2H2+1LgKbtGg9Rth+YBcIBb5Iw+WXJFgfM3dvGX8rS5
btcHQuSB5SXVsAy2LVupQT6SZ4XqZ82dCFsGy4GJQfFvhiIIZ3K3j1ReZtAyI/Ct
OZAZaqfzfnSdCbTmu9GygwiJgkUtKhHd67od0hrTyUEfvDe1q+b//H5Sz1Tt/S/p
3OQlbDuRhK66TvPYuDns
=T4o5
-----END PGP SIGNATURE-----

NewToTheStory, 2016-11-20:
Hi Guys,

i'am not sure, but i think i have the same or at least similiar problem,

I have two Hitachi Travelstar 5K1000 HTS541010J22413, installed in 2 cases, which also are the same models.
On first hdd i have bootpartition and also seafile-data. The second hdd is planned as backup for hdd1. All worked fine, till today, when i shutdown raspi, siwtched of power, replaced hardware to another place in flat and powerd on. Since i connected both hdd i get an I/O Error on SDA1 (non-backup HDD) and OS run in RO. Currently i only can use the system with unplugged second hdd, thats really annoying

`sebastian@raspberrypi:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.26-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #915 SMP Thu Oct 20 17:08:44 BST 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808c2f80, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f62000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa02082 bcm2709.serial=0x97c05755 smsc95xx.macaddr=B8:27:EB:C0:57:55 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/sda1 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939072K/966656K available (6348K kernel code, 432K rwdata, 1716K rodata, 476K init, 764K bss, 19392K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807e8584   (8066 kB)
      .init : 0x807e9000 - 0x80860000   ( 476 kB)
      .data : 0x80860000 - 0x808cc290   ( 433 kB)
       .bss : 0x808cf000 - 0x8098e1ec   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000293] Console: colour dummy device 80x30
[    0.001345] console [tty1] enabled
[    0.001390] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001459] pid_max: default: 32768 minimum: 301
[    0.001798] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001842] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002800] Disabling cpuset control group subsystem
[    0.002858] Initializing cgroup subsys io
[    0.002910] Initializing cgroup subsys memory
[    0.002975] Initializing cgroup subsys devices
[    0.003019] Initializing cgroup subsys freezer
[    0.003062] Initializing cgroup subsys net_cls
[    0.003135] CPU: Testing write buffer coherency: ok
[    0.003224] ftrace: allocating 21225 entries in 63 pages
[    0.052930] CPU0: update cpu_capacity 1024
[    0.053000] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.053033] [bcm2709_smp_prepare_cpus] enter
[    0.053198] Setting up static identity map for 0x8240 - 0x8274
[... 345 lines cut ...]
[  288.391707] sd 0:0:0:0: [sda] tag#0 ASC=0x47 ASCQ=0x3 
[  288.391739] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 00 00 08 00 00 00 08 00
[  288.391766] blk_update_request: I/O error, dev sda, sector 2048
[  288.391787] Buffer I/O error on dev sda1, logical block 0, lost sync page write
[  288.981051] .ready
[  288.981583] sd 1:0:0:0: [sdb] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[  288.981989] sd 1:0:0:0: [sdb] Write Protect is off
[  288.982008] sd 1:0:0:0: [sdb] Mode Sense: 33 00 00 08
[... cut ...]
[  293.081785] blk_update_request: I/O error, dev sda, sector 8394240
[  293.081805] Buffer I/O error on dev sda1, logical block 1049024, lost async page write
[  306.431101] EXT4-fs (sda1): error count since last fsck: 21
[  306.431135] EXT4-fs (sda1): initial error at time 1479652602: ext4_journal_check_start:56
[  306.431151] EXT4-fs (sda1): last error at time 1479659146: ext4_journal_check_start:56
`

`sebastian@raspberrypi:~$ uname -a
Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux`

NicoHood, 2016-12-18:
I've got the same issue every few minutes. I am using ArchLinux with:
```
Linux alarmpi 4.4.37-1-ARCH #1 SMP Fri Dec 9 19:03:41 MST 2016 armv7l GNU/Linux
```
What is weird is, that i am using 2 identical usb 2tb (seagate expansion) drives with btrfs raid0. But only sda gives those warnings. The music hangs for a short time when the bug/error occurs.

NewToTheStory, 2016-12-18:
I found out that mounting the second hdd, caused this problem.
I deleted the hdd completely with dd and after this the problem was gone. But I had to restore the hdd anyway, which caused the problem, because after cleaning I had (or maybe also before) an hardware issue with her (clicking). Now I have again to similar hdd's mounted without this problem...
Maybe this helps someone

NicoHood, 2016-12-18:
Is it possible that the internal hub of the raspi causes problems? Or undervoltage? I have the official 5.1V 2,5A PSU, so that should be fine. The hdds are self powered. But I remember that one of my hdd crashed 2-3 times and I had to unplug it from power to get it spin up again. I am not sure if this is a driver fault or if my hardware is defective.

Dmole, 2016-12-20:
Not likely I put paper + tape over the power pin to isolate the pi from powered USB devices... have yet to test with 4.8.15 or 4.9...

JamesH65, 2017-05-18:
I suspect there a number of issues at play, bad power supply and perhaps an upstream driver issue. neither of which should be dealt with here, so will close. If anyone requires it or has more Raspberry specific information, feel free to reopen.

closed 2017-05-18 by JamesH65

marcotini, 2020-06-12:
Just happen to me on Raspberry Pi 4 and my WD My Passport 4TB when I've plugged an ssd on the second usb 3.0 port:

```
[Thu Jun 11 11:27:05 2020] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[Thu Jun 11 17:10:17 2020] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[Thu Jun 11 21:11:13 2020] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[Fri Jun 12 10:40:17 2020] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
```

The hard drive made a click (like a bad click for hdd) and I assume that it's a power issue. Even if I use a 30W Apple original power supply.

What can you suggest me to do?

```
[134073.824203] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[134073.824224] sd 0:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 c5 07 8b 00 00 00 01 00 00 00
[134073.824239] print_req_error: I/O error, dev sdb, sector 3305605888
[182617.523776] sd 0:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
[182617.523795] sd 0:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 29 08 97 f8 00 00 00 08 00 00
[182617.523809] print_req_error: I/O error, dev sdb, sector 688429048
```

And I'm on:

`Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux`

JamesH65, 2020-06-12:
This is very old issue, and closed, but, this is purely a power problem. The Pi can provide 1.2A total across all the USB ports. You are likely exceeding that. Use a powered hub.

marcotini, 2020-06-12:
> This is very old issue, and closed, but, this is purely a power problem. The Pi can provide 1.2A total across all the USB ports. You are likely exceeding that. Use a powered hub.

And a cable like [this one](https://images-na.ssl-images-amazon.com/images/I/51LmgwQ7EdL._SL1000_.jpg) can make a difference (plugging the other one on a different power supply)?

timstanley1985, 2020-06-12:
I have been having this issue for 2+ years with a usb hard drive via a powered hub so not sure it is a power issue.  I've never lost any data so just ignore the errors

3dr14n, 2020-07-05:
Same issue with my powered (12V2A) external drive bay Orico NS1066 for my WD80EZAZ-11T connected to Raspberry Pi 4. The behavior is repeatable because as soon as iowait become high while playing 60GB H265 video via Kodi installed in Raspbian, the HDD disconnects from, for example `sdb` then reconnects to `sdc` .

`Jul  5 19:04:04 megaserver kernel: [81232.365054] usb 2-2: Disable of device-initiated U1 failed.
Jul  5 19:04:09 megaserver kernel: [81237.393757] usb 2-2: Disable of device-initiated U2 failed.
Jul  5 19:04:18 megaserver kernel: [81246.141910] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
Jul  5 19:04:18 megaserver kernel: [81246.141920] usb 2-2: USB disconnect, device number 3
Jul  5 19:04:18 megaserver kernel: [81246.141930] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 a4 ee 3e e0 00 00 04 00 00 00
Jul  5 19:04:18 megaserver kernel: [81246.142295] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=0x00
Jul  5 19:04:18 megaserver kernel: [81246.142313] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x88 88 00 00 00 00 00 a4 ee 4a e0 00 00 02 00 00 00`

The only solution is to restart, for the hdd to reconnect again via its default mount-point in fstab.

pelwell, 2020-07-05:
Try disabling UAS by following this guide: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

3dr14n, 2020-07-06:
> Try disabling UAS by following this guide: https://www.raspberrypi.org/forums/viewtopic.php?t=245931

UAS was already disabled when issuing a `lsusb -t`  the driver was already in `usb-storage` even without applying quirks in cmdline.txt. Sure enough, I tried it anyway as instructed in the link provided for curiosity's sake but still the same behavior, disconnection and reconnection. 

My main issue now could be because Norelsys NS1066 could be faulty but this issue does not happen on the earlier days of Pi4. Before I can copy 100GB of zip file to the same drive without any problem (via ethernet) but now I can't do the same, even media playback via Kodi makes the drive disconnect and reconnect afterwards.

I won't associate this with power issue because the dock was already powered via 12V2A.
The drive disconnects as soon as iowait became red while being monitored via terminal using glances.
Thermal is not an issue either because the Pi and the HDD are cooled via 120mm fan intake. The Pi4 were kept cool below 55°C while playing Kodi.
I'm not doing any Overclocking with the Pi4.

I'm wondering if there were changes with the USB 3 chip in Pi4 that could cause this issue?

andrewufrank, 2020-12-07:
I had exectly the same error msg - with a very different setup (odroid xu4 - armbian debian buster) and remembered having seen it before: it was caused by insufficient power supply (as suggested above by others). I changed the powersupply from 4A to 6A and the problem is gone.  -- This mostly as a message to others!

CoiaPrant, 2022-02-20:
I have the same problem!

cslev, 2024-01-04:
While the thread's last comment is more than 3 years old, the problem still exists :)
I am having a raspberry pi 4 enclosed in a Geekworm Naspi Gemini [https://geekworm.com/collections/raspberry-pi/products/x882?_pos=4&_ss=r](https://geekworm.com/collections/raspberry-pi/products/x882?_pos=4&_ss=r).
While I have indeed two SSDs connected to the pi, I encounter this error also when writing/reading from only one of them. And because there are occasional writings/readings (e.g., media streams, downloads, nextcloud), I face this problem every day and only reboot helps.
Power and heating are not issues for me either. I purposely use a 12V5A charger to provide sufficient power. A fan is also inside the case, and has never been above 55 C (I monitor it with Prometheus and Grafana).
I believe that maybe the USB cable or the NASpi Gemini board causes the issue, but did not bother yet to open the case, and connect the disks by other means (e.g., powered USB SSD enclosure, other geekworm naspi products) before someone can confirm similar findings :)

andrewufrank, 2024-01-04:
i have now several raspi4 running the debian raspi images from https://raspi.debian.net/tested-images/ and have no problems  - but my power supply is 6A. I saw the issue when i had a 4A power supply. perhaps the 5A are  not enough to  deal with the spikes when writing. before anything else, I would put in a larger powersupply and change the usb cable!

cslev, 2024-01-04:
I actually use a step-down converter which is precisely set to 5V from 12V. Since the charger can deliver 5A on 12V, after reducing the consumption to 5V only, I would not expect power issue. 
But you are right, I should give it a try...however, I find it quite difficult to get 5V charger with more than 4A support. Do you have any good source for buying something like yours?

pelwell, 2024-01-04:
Be careful using chargers as power supplies - although they do the same job, the requirements for a charger are a bit more relaxed; a momentary loss of charging probably won't affect your device whereas a brief loss of power probably will.

cslev, 2024-01-04:
oh sorry, i am using power supply, not a charger. Just usually interchange those words, but thanks for pointing this out :+1:

andrewufrank, 2024-01-04:
my 5v6a power supply i bought for another cpu. it is meanwell sga60e05 and then have a converter from barel connector to usb. i see it still on offer for $28 (I hope I paid less five years ago, I bought several!). it worked well with the odroid-xu4 and now with the raspi4.  @pelwell Thank you for the hint. I had naively assumed they would be to the same specs.

andrewufrank, 2024-01-09:
I remember that I have troubles if added two hard drives (the second with its own power supply, the first is a regular 2.5" 3TB drive). I never investigated what was the cause; there were simpler ways to make a backup.

cslev, 2024-01-11:
I use one harddrive for media, and the other one for nextcloud. But I will merge them into the same drive and use only a USB enclosure with external power instead of buying the whole gemini board. Or maybe just buy a smaller board with a single SATA slot. 
Thanks for the comments, though