raspberrypi/linux #1288: g_ether

Darkmelman, 2016-02-12:
Hello,

i have updated to the new raspbian release of 2016-02-09 (4.4.1+) and these steps:

1. First, flash Jessie (full, not lite) onto a blank microSD card.
2. Boot it up and resize the root partition (open a terminal and enter sudo raspi-config, then select        expand and reboot).
3. Once it starts up again, run sudo BRANCH=next rpi-update. This will take a while.
4. Next we need to make sure we are using the dwc2 USB driver sudo echo "dtoverlay=dwc2" | tee /boot/config.txt.
5. And enable it in Raspbian sudo echo "dwc2" | tee /etc/modules
6. Need to now pick which module you want to use from the list above, for example for ethernet sudo echo "g_ether" >> /etc/modules. You can only pick one of the above modules to use at a time.

but USB- Gadget (g_ether) isn’t working now! Before the update, it was working properly…
can anyone confirm that or hast anyone a solution?

Thank you verry much!

ersinpw, 2016-05-07:
Same problem here with version: 4.4.6+

popcornmix, 2016-05-07:
Don't use BRANCH=next. That is used when testing specific experimental features, and hasn't been updated for a long time. The default branch is also 4.4. kernel and is much newer.

Ruffio, 2016-08-17:
@Darkmelman has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-18 by JamesH65