raspberrypi/linux #1290: __DWC_DMA_FREE warning in dwc_otg (kernel 4.4)

jkuek, 2016-02-15:
This was discovered while trying to investigate https://github.com/raspberrypi/linux/issues/1272, but I believe it is a separate issue. A couple more relevant comments are included in that issue.

A Raspberry Pi B+ was loaded with Raspbian Jessie, and then upgraded to the latest kernel (sudo BRANCH=next rpi-update)
When disconnecting and reconnecting USB devices, it is common to see the following in syslog:

```
Feb 14 22:42:18 raspberrypi kernel: [240747.759064] WARNING: CPU: 0 PID: 2955 at include/asm-generic/dma-mapping-common.h:274 __DWC_DMA_FREE+0xac/0xc8()
Feb 14 22:42:18 raspberrypi kernel: [240747.759080] Modules linked in: cfg80211 rfkill joydev evdev snd_usb_audio snd_usbmidi_lib snd_hwdep snd_rawmidi sierra snd_seq_device sierra_net usbserial bcm2835_gpiomem snd_bcm2835 bcm2835_wdt snd_pcm snd_timer snd uio_pdrv_genirq uio
Feb 14 22:42:18 raspberrypi kernel: [240747.759187] CPU: 0 PID: 2955 Comm: kworker/0:1 Not tainted 4.4.1+ #837
Feb 14 22:42:18 raspberrypi kernel: [240747.759201] Hardware name: BCM2708
Feb 14 22:42:18 raspberrypi kernel: [240747.759236] Workqueue: usb_hub_wq hub_event
Feb 14 22:42:18 raspberrypi kernel: [240747.759302] [<c00169b8>] (unwind_backtrace) from [<c0013944>] (show_stack+0x20/0x24)
Feb 14 22:42:18 raspberrypi kernel: [240747.759336] [<c0013944>] (show_stack) from [<c02df2f4>] (dump_stack+0x20/0x28)
Feb 14 22:42:18 raspberrypi kernel: [240747.759369] [<c02df2f4>] (dump_stack) from [<c0021a28>] (warn_slowpath_common+0x8c/0xc4)
Feb 14 22:42:18 raspberrypi kernel: [240747.759400] [<c0021a28>] (warn_slowpath_common) from [<c0021b1c>] (warn_slowpath_null+0x2c/0x34)
Feb 14 22:42:18 raspberrypi kernel: [240747.759434] [<c0021b1c>] (warn_slowpath_null) from [<c03f8368>] (__DWC_DMA_FREE+0xac/0xc8)
Feb 14 22:42:18 raspberrypi kernel: [240747.759467] [<c03f8368>] (__DWC_DMA_FREE) from [<c03f1e20>] (dwc_otg_hcd_qh_free+0xbc/0xec)
Feb 14 22:42:18 raspberrypi kernel: [240747.759496] [<c03f1e20>] (dwc_otg_hcd_qh_free) from [<c03ea6bc>] (dwc_otg_hcd_endpoint_disable+0xac/0xc0)
Feb 14 22:42:18 raspberrypi kernel: [240747.759525] [<c03ea6bc>] (dwc_otg_hcd_endpoint_disable) from [<c03edca8>] (endpoint_disable+0x28/0x34)
Feb 14 22:42:18 raspberrypi kernel: [240747.759560] [<c03edca8>] (endpoint_disable) from [<c03c4148>] (usb_hcd_disable_endpoint+0x3c/0x40)
Feb 14 22:42:18 raspberrypi kernel: [240747.759596] [<c03c4148>] (usb_hcd_disable_endpoint) from [<c03c66c8>] (usb_disable_endpoint+0x70/0x8c)
Feb 14 22:42:18 raspberrypi kernel: [240747.759627] [<c03c66c8>] (usb_disable_endpoint) from [<c03c6904>] (usb_disable_device+0x1bc/0x1d8)
Feb 14 22:42:18 raspberrypi kernel: [240747.759658] [<c03c6904>] (usb_disable_device) from [<c03bec10>] (usb_disconnect+0x78/0x208)
Feb 14 22:42:18 raspberrypi kernel: [240747.759687] [<c03bec10>] (usb_disconnect) from [<c03bf85c>] (hub_event+0x300/0xe4c)
Feb 14 22:42:18 raspberrypi kernel: [240747.759715] [<c03bf85c>] (hub_event) from [<c0036b84>] (process_one_work+0x110/0x39c)
Feb 14 22:42:18 raspberrypi kernel: [240747.759741] [<c0036b84>] (process_one_work) from [<c0036f6c>] (worker_thread+0x15c/0x4e4)
Feb 14 22:42:18 raspberrypi kernel: [240747.759774] [<c0036f6c>] (worker_thread) from [<c003c6e8>] (kthread+0xe0/0xfc)
Feb 14 22:42:18 raspberrypi kernel: [240747.759806] [<c003c6e8>] (kthread) from [<c000f608>] (ret_from_fork+0x14/0x2c)
Feb 14 22:42:18 raspberrypi kernel: [240747.759822] ---[ end trace 01333adc69252151 ]---

```
After this the system "seems" to run normally. It's not clear whether there are long-term problems

P33M, 2016-02-15:
http://lxr.free-electrons.com/source/include/asm-generic/dma-mapping-common.h#L274

It's a warning for having IRQs disabled. It appears the DMA API has been reworked between 4.1 and 4.4 - apparently doing a dma_free_coherent can no longer be done with interrupts disabled. There should be no impact on functionality.

popcornmix, 2016-03-18:
Latest ``BRANCH=next rpi-update`` kernel includes a fix for the dmesg warning involving dwc_otg_hcd_qh_free. Can you update and test?

jkuek, 2016-03-21:
I updated to 4.4.6+ #861, and ran some tests. I was not able to reproduce the warning. I think it's safe to close this one off.

Note that I was still able to reproduce issue #1272 after updating

closed 2016-03-21 by popcornmix