raspberrypi/linux #1291: About rpi-firmware repo

bigbuglamer, 2016-02-15:
After latest commit of rpi-firmware (rebuild with linaro gcc 4.9.3) omxplayer says "Illegal instruction", problem disappears when i downgrade to any previous build.

Platform is:
RPI B+
Hardware : BCM2708
Revision : 0010
CPU variant : 0x0
CPU part : 0xb76
CPU revision : 7

Also vcgencmd stops working, looks like same issue.

P.S.: reposting there because i have found that rpi-firmware repo is just for hosting binaries, not for development/etc.

pelwell, 2016-02-15:
This is a build issue and not a source issue, so firmware is arguably the correct place. Either way, we're working hard behind the scenes to get Pi1-compatible build.

popcornmix, 2016-02-15:
Duplicate of https://github.com/raspberrypi/firmware/issues/544

closed 2016-02-15 by popcornmix