raspberrypi/linux #1296: LUKS / cryptsetup / dm-crypt not working anymore

saddy001, 2016-02-20:
Since the last upgrade I cannot mount my LUKS encrypted device anymore. It says

Geben Sie die Passphrase für »/dev/sda2« ein: 

> Einrichten der dm-crypt-Schlüsselzuordnung für Gerät »/dev/sda2« fehlgeschlagen.
> Stellen Sie sicher, dass der Kernel die aes-xts-plain64-Verschlüsselung unterstützt.
> (Sehen Sie im System-Log nach, ob sich dort Hinweise finden.)

The following packages were upgraded:
`cpio libc-bin libc-dev-bin libc6 libc6-dbg libc6-dev libgcrypt20 libgcrypt20-dev libgraphite2-3 libpq5 libtiff5 locales multiarch-support nginx nginx-common nginx-full`

popcornmix, 2016-02-20:
See here for info:
https://github.com/Hexxeh/rpi-firmware/issues/109

saddy001, 2016-02-20:
Now THIS was a fast reply, thank you. I hope it will be fixed soon.

FuturePilot, 2016-02-29:
4.1.18 just hit the official raspberrypi-bootloader package. I'm guessing a lot more people are going to get bitten by this bug other than those using rpi-update.

dh-connect, 2016-03-01:
I had the same issue & posted it here, sorry didn't see this one
https://github.com/raspberrypi/firmware/issues/552

7hunderbug, 2016-03-01:
I just got hit with this bug on a headless RPi zero. I've got my encrypted volume set up to auto-mount with /etc/cryptab and /etc/fstab and due to this bug, the Pi will not complete the boot process and it hangs - wireless networking doesn't start (so no IP address) and so is completely inaccessible via SSH. The only way to fix it was to connect a screen and keyboard, interrupt the boot process with <Ctrl>+<c> and comment out the relevant entries in the aforementioned files so that I can reboot and downgrade the kernel/firmware.

For future reference and others visiting this bug (from https://github.com/raspberrypi/linux/issues/1312):

`sudo rpi-update 0764e7d`

popcornmix, 2016-03-05:
I've pushed an update to 4.1.19 which looks to contain a fix for this. Please update and test.

dh-connect, 2016-03-06:
Yes update 4.1.19 worked for me, thank you!

saddy001, 2016-03-08:
Worked here too. Thanks!

closed 2016-03-08 by popcornmix