raspberrypi/linux #1302: rpi-proto audio codec on Raspberry Pi v2 model B

Shibonja, 2016-02-24:
Hi,

First I want to say great work. I recently tried to get I2S sound working on the Raspberry Pi v2 model B with rpi-proto board purchased on mikroelektronika:
http://www.mikroe.com/add-on-boards/audio-voice/audio-codec-proto/

I enabled the codec in /boot/config and it is displayed when listing audio devices. I connected the board like this:
SCK -> J8-Pin12
MISO -> J8-Pin38
MOSI -> J8-Pin40
ADCL,DACL -> J8-Pin35
SDA -> J8-Pin3
SCL -> J8-Pin5
3.3V -> J8-Pin17
GND -> J8-Pin39

I alos loaded the modules: 
snd_soc_bcm2708_i2s
bcm2708_dmaengine
snd_soc_wm8731
snd_soc_rpi_proto

jet, I can not get any sound recording and also sound output. Am I missing some settings. My kernel version is 4.1.7-v7+. 

Regards, Haris.

popcornmix, 2016-02-24:
You might have better luck contacting mikroelektronika.

Shibonja, 2016-02-25:
Hi, @popcornmix 

I got playback working, even with the errors on dmesg log, except that I am getting a new one every time I play something on the audio codec:
```bash
[   59.514601] bcm2708-i2s 3f203000.i2s: I2S SYNC error!
```

Shibonja, 2016-03-04:
Hi, @popcornmix 

I made recording kind of work. I found a code example for testing the microphone with Arduino and wm8721. I Uploaded the code to arduino and connected only arduinos I2C pins to wm8731. Run the code once, and connected the wm8731 I2C pins back to raspberry. And after this I was ableto record sound and also play sound as well. After a reboot with the codec connected to raspberry pi sound recording was gone, I needed to repeat the procedure to get it work again. It seem that the raspberry pi codec driver writes some registers in wm8731 which are disabling sound recording. The Arduino code i found is on this link: https://www.sparkfun.com/products/retired/11290

Regards, Haris.

klausro, 2016-03-04:
Hi @Shibonja,

how did you manage to get at least playback running? I didn't hear anything. The errors in dmsg are
the same as yours during startup, and during playback I get

bcm2708-i2s 3f203000.i2s: I2S SYNC error!
bcm2708-dmaengine 3f007000.dma: DMA transfer could not be terminated

I try playback with 
mplayer -ao alsa:device=hw=1,0 file.mp3

Regards, Klaus

pelwell, 2016-03-04:
Why are you not using the rpi-proto overlay?
```
dtoverlay=rpi-proto
```
It enables i2c and i2s, loads the drivers and configures the ALSA interface for you. You shouldn't need to load any modules or put anything else in config.txt.

pelwell, 2016-03-04:
Note that the -517 errors are an artifact of the load order (they are components indicating that something they need isn't loaded yet and to try again later, also known as EPROBEDEFER) and aren't necessarily a sign of a problem.

klausro, 2016-03-04:
Hi @pelwell ,
thanks for the information. I use the dtoverlay=rpi-proto. In his 3rd post, @Shibonja  said he got audio out working. I just want to know, what he has done!

clivem, 2016-03-04:
@pellwell Phil, just a heads-up, I haven't had any time today to look at this, but they are using SND_SOC_DAIFMT_CBM_CFM with this DAC driver. I wonder if the problem reported by the HiFiBerry people, with their DAC in clock master mode on Pi3B, might also be applicable to Pi2B.... hence the I2S SYNC error here!

klausro, 2016-03-04:
Ok, got it running. Simple solution, audio out was disabled in mixer.

meerd, 2016-04-29:
Hello,

Is anyone able to record sound with rpi-proto? The board works perfectly when it comes to play something, but I haven't been able to record any sound yet. I tried changing alsa's settings but this did not work as well. Any help is appreciated.

Thanks.

kweiwen, 2016-05-30:
hi @meerd, i have same problem as you first, but finally i can use mic input to record sound now :D

here's my setting for your information:
i use pi2 and wm8731 proto board the setting were the same as @Shibonja 
use `amixer cset numid=13 on` to unmute Output Mixer HiFi Playback Switch
i manually use `alsamixer` to select input as "Mic Input" (this is very important, cause after checked out the schematic of mikroelektronika they use mic input on the wm8731 and line input was unconnected). now i  can use `arecord` to record something.

Ruffio, 2016-08-17:
@Shibonja has your issue been resolved? If so, please close this issue. Thanks.

meerd, 2016-08-23:
Hello @kweiwen,

Thanks for your reply. I solved the issue by using alsamixer too.

Shibonja, 2016-08-27:
Hi, 
Sorry for late response. I got sound recording and playback working.

closed 2016-08-27 by Shibonja

05101991, 2016-09-28:
Hi All,
In my audio codec board playback is working fine . But recording is not done . 
arecord -vD hw:2,0 -c 2 -r 48000 -f S16_LE  name.wav 
I used arecord to record an audio using earphones. 
please help me out to record an audio through my audio codec board(wm8731s) . 
Is anything I need to enable before recording ?

Shibonja, 2016-09-28:
Hi, 
Here is my confiugration and changes I made to get codec working.

Change config.txt
``
sudo nano /boot/config.txt
##########################
#uncomment lines:
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
#add lines:
dtoverlay=i2s-mmap
dtoverlay=rpi-proto
``

05101991, 2016-09-28:
Evrything I had enabled previously . (both in .config.txt and etc/modules) . 

For playback  I used   1 ---> amixer cset numid=13 on
                                  2----> mpalyer song.mp3 
Its working fine. 

For recording I used  1 --->  amixer cset numid=14 0n   [ for line in ] 
                                  2 ----> arecord -vD hw:2,0 -c 2 -r 48000 -f S16_LE  name.wav 

I am unable to record audio. The name.wav file is created but,If i play the .wav file Unable to hear recorded voice . 
Please correct If I am Going wrong.

Shibonja, 2016-09-28:
hi @05101991 

Like said maybe are the id's for different settings on amixer are not the same for you, you need to change the following three items: Output Mixer HiFi Playback Switch, Mic Capture Switch and Input Mux.

05101991, 2016-09-28:
Hi Shibonja 

How to change Output Mixer HiFi Playback Switch, Mic Capture Switch and Input Mux .
I am new to this . can you help me ???

05101991, 2016-09-29:
Hi Shibonja

I changed 

pi@raspberrypi:~$ amixer controls
numid=2,iface=MIXER,name='Master Playback ZC Switch'
numid=1,iface=MIXER,name='Master Playback Volume'
numid=4,iface=MIXER,name='Line Capture Switch'
numid=5,iface=MIXER,name='Mic Boost Volume'
numid=6,iface=MIXER,name='Mic Capture Switch'
numid=8,iface=MIXER,name='ADC High Pass Filter Switch'
numid=3,iface=MIXER,name='Capture Volume'
numid=10,iface=MIXER,name='Playback Deemphasis Switch'
numid=14,iface=MIXER,name='Input Mux'
numid=13,iface=MIXER,name='Output Mixer HiFi Playback Switch'
numid=11,iface=MIXER,name='Output Mixer Line Bypass Switch'
numid=12,iface=MIXER,name='Output Mixer Mic Sidetone Switch'
numid=7,iface=MIXER,name='Sidetone Playback Volume'
numid=9,iface=MIXER,name='Store DC Offset Switch'

pi@raspberrypi:~$ amixer cset numid=13 on
numid=13,iface=MIXER,name='Output Mixer HiFi Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on

pi@raspberrypi:~$ amixer cset numid=6 on
numid=6,iface=MIXER,name='Mic Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on

amixer cset numid=14 1                                        
numid=14,iface=MIXER,name='Input Mux'                                           
  ; type=ENUMERATED,access=rw------,values=1,items=2                            
  ; Item #0 'Line In'                                                           
  ; Item #1 'Mic'   

To Record : 
arecord -c 2 -D hw:0,0 -f S16_LE -r48000 song.wav
Recording WAVE 'song.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
overrun!!! (at least 975.543 ms long)

File is created .But, the voice is not recorded . File size is also around 3mb. 

To stop recording do we have any commands  ??

05101991, 2016-09-29:
What may be the problem ? Stiil I cant record an audio.

Shibonja, 2016-09-29:
Hi, I used codec board from link http://www.mikroe.com/add-on-boards/audio-voice/audio-codec-proto/ .
Check if connections are the same.

05101991, 2016-09-29:
Same board I am also using . The connections(audio codec to raspberrpy pi 3)  are same what you specified above (I did not make any changes in connection ). 

I am using Earphones to record my voice (Is anything is required more to record audio) through codec board. 

During booting of my pi board I am Getting this error. 

[   15.993127] snd-rpi-proto soc:sound: ASoC: CPU DAI (null) not registered
[   15.993138] snd-rpi-proto soc:sound: snd_soc_register_card() failed: -517

Shibonja, 2016-09-29:
It worked form me, I recorded using some cheap pc microphone like this: http://www.promo-wholesale.com/Upfiles/Prod_r/Jwin-PC-Microphone_20090763410.jpg

05101991, 2016-09-29:
But I used our mobile Earphones . (will this  work ?)

clivem, 2016-09-29:
There are two 3.5mm jacks on this board. A stereo (green) headphone port for output and a mono input for mic. What do you have plugged into the (pink) microphone 3.5mm jack, anything?

05101991, 2016-09-29:
I used mobile Earphones (there is no seperate channel for mobile earphones )  .In Board I am using input line to record and and output line for playback. 

Guys can we use mobile headset to record an audio through audio codec board . If I am wrong please help me out..

clivem, 2016-09-29:
I do not own this board, but having looked at the circuit diagram, I doubt you are going to get very far trying to record anything from a microphone unless it is plugged into the MIC jack.

[image]

05101991, 2016-09-29:
Hi clivem

I did not get you what you are trying to say . Can you expalin briefly . we are recording by connecting headset through mic jack .

chronopoulos, 2020-05-21:
a mobile headset (combined headphones+microphone in a TRRS plug) will not work with the mic input on the proto board, because only the left heaphone channel will be making contact with the signal line in the mic jack. the order of signals in a headset TRRS is either {left, right, mic, gnd} or {left, right, gnd, mic}. in either case, it's the left channel that's contacting the microphone input, which won't do you much good (actually you can test this by yelling loudly into the left earphone, using it as a microphone in reverse).

a better solution is to either get a mono computer mic and plug that in, or else just run an 1/8" TRS cable from an mp3 player (set on very low volume) into the proto board. you should then be able to record the left channel of the mp3 player using the proto board / arecord.