raspberrypi/linux #1305: Raspbian mirror root.lu down

grueneedv, 2016-02-26:
While trying to download:
http://mirrordirector.raspbian.org/raspbian/pool/main/x/xorg/xserver-xorg_7.7+7+b1_armhf.deb
and other files I get directed to
http://raspbian.mirror.root.lu/raspbian/pool/main/x/xorg/xserver-xorg_7.7+7+b1_armhf.deb

Unfortunately this mirror doesn't contain the directory raspian (butothers) so I get a 404 and can't download any packages at all.

For someone searching for it, go to /etc/apt/sources.list and enter an other mirror by hand from here: http://www.raspbian.org/RaspbianMirrors

mati75, 2016-02-26:
Please run first:
`sudo apt-get update`
and try again.

grueneedv, 2016-02-26:
Have you even tried to click one of the links?

XECDesign, 2016-02-26:
As this is not a raspberrypi.org or a raspberry pi kernel issue, please submit it at https://www.raspbian.org/RaspbianBugs.

closed 2016-02-26 by XECDesign