raspberrypi/linux #1306: --depth 1

ShorTie8, 2016-02-26:
Trying to do a
```
git clone -b rpi-3.18.y --depth 1  https://github.com/raspberrypi/linux.git
```
But it seems I always git there whole tree @ ~370megs
and when I do a 'git branch -a', I see all the branches in stead of just rpi-3.18.y as I would think .. :/~
```
 git branch -a
* rpi-3.18.y
  remotes/origin/HEAD -> origin/rpi-4.1.y
  remotes/origin/rpi-3.18.y
  remotes/origin/rpi-3.18.y-rebase
  remotes/origin/rpi-3.19.y
  remotes/origin/rpi-3.2.27
  remotes/origin/rpi-3.6.y
  remotes/origin/rpi-3.8.y
  remotes/origin/rpi-3.9.y
  remotes/origin/rpi-4.0.y
  remotes/origin/rpi-4.1.y
  remotes/origin/rpi-4.1.y_rebase
  remotes/origin/rpi-4.2.y
  remotes/origin/rpi-4.3.y
  remotes/origin/rpi-4.3.y_rebase
  remotes/origin/rpi-4.4.y
  remotes/origin/rpi-4.4.y_irq
  remotes/origin/rpi-4.5.y
  remotes/origin/rpi-patches
```

dlech, 2016-02-26:
Linux is 15+ million lines of code. That means you have 15MB in newline characters alone.  I think the 370MB sounds right for `--depth 1` (~25 chars/line). Just because `git branch -a` lists remote branches, it does not mean that it downloaded them. The whole tree is probably on the order of several GB.

6by9, 2016-02-26:
Certainly my complete clone is 3.2GB though that includes build artefacts too. The .git directory is 2.0GB.
Even if it were an issue, then it's an issue with git not with the repository.

ShorTie8, 2016-02-27:
but
But
BUT

when I follow the kernel/building.md with a
```
git clone --depth=1 https://github.com/raspberrypi/linux
```
it comes in @ ~ 240megs, which stills seems kinda big, with a git branch of
```
git branch -a
* rpi-4.1.y
  remotes/origin/HEAD -> origin/rpi-4.1.y
  remotes/origin/rpi-4.1.y
```

6by9, 2016-02-27:
And?
```git clone --depth=1 https://github.com/torvalds/linux.git``` (ie the upstream kernel) and I get 881M used - 161MB in git data, and 720MB of source.
Having done it for the Pi 4.1.y tree, I get 150MB of .git and 665MB of source. That makes some sense for 4.5.y vs 4.1.y.

That's the size of the Linux kernel source. If you want to build the kernel then you need all that.

ShorTie8, 2016-02-27:
Ok,
this wasn't really a issue, more of a wonder
popcorn suggested to use the rpi-3.18.y.tar.gz version,
So I'll try that

Thankz for all the play you guys did

Have A Great Day
ShorTie

closed 2016-02-27 by ShorTie8