raspberrypi/linux #1307: unzip -q rpi-3.18.y.zip

ShorTie8, 2016-02-27:
Why when I do a 'unzip -q rpi-3.18.y.zip' do I get 'symlink error: File name too long' ??
Something doesn't sound right to me .. :(~
```
unzip -q rpi-3.18.y.zip
symlink error: File name too long
symlink error: File name too long
symlink error: File name too long
symlink error: File name too long
symlink error: File name too long
symlink error: File name too long
```

popcornmix, 2016-02-27:
Are you running this on Windows?
That will not work. The file system won't support it.
If you are on Linux then download the tarball (replace .zip in url with .tar.gz) or even better just use git clone.

ShorTie8, 2016-02-27:
Nop, it's on a rpi

popcornmix, 2016-02-27:
Well that Linux so download the tarball or use guy clone.

ShorTie8, 2016-02-27:
down load a tarball ??
https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.bz2   or
https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.xz
doesn't seem to work for me
```
wget https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.bz2
--2016-02-27 12:04:19--  https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.bz2
Resolving github.com (github.com)... 192.30.252.130
Connecting to github.com (github.com)|192.30.252.130|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-02-27 12:04:24 ERROR 404: Not Found.
```

ShorTie8, 2016-02-27:
and git clone is giving me
```
scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[4]: *** [silentoldconfig] Error 1
Makefile:541: recipe for target 'silentoldconfig' failed
make[3]: *** [silentoldconfig] Error 2
make[2]: *** No rule to make target 'include/config/auto.conf', needed by 'include/config/kernel.release'.  Stop.
```

pelwell, 2016-02-27:
That error does not come from git. I'm guessing the clone completed without error. What commands are you trying to run?

ShorTie8, 2016-02-27:
it's part of a script, but I believe it's in the 'make prepare' part
```
else ifeq ($(BUILD_PHASE), Final_arm)
	@echo "Here at kernel-headers Makefile Final_arm"
	+mkdir -p $(PKG_ROOT); \
	echo "Here at cd mrproper"
	cd linux; \
	  make mrproper
	echo "Here at sed"
	echo "CFG_TYPE  $(CFG_TYPE)"
	echo "KERNEL_TYPE  $(KERNEL_TYPE)"
	echo "ARM_TYPE  $(ARM_TYPE)"
	cd linux; \
	  sed -e 's/^CONFIG_LOCALVERSION=.*$$/CONFIG_LOCALVERSION="$(KERNEL_TYPE)"/' ../../linux.config$(CFG_TYPE) > .config
	echo "Here at prepare"
	cd linux; \
	  make prepare
	echo "Here at headers check"
	cd linux; \
	  make headers_check
	@echo "Here at make"
	cd linux-$(KERNEL_VER); \
	  make INSTALL_HDR_PATH=$(PKG_ROOT)$(PKG_DIR) headers_install
	@echo "Here at find"
	cd linux-$(KERNEL_VER); \
	  find $(PKG_ROOT)$(PKG_DIR)/include \( -name .install -o -name ..install.cmd \) -deleteelse
```

popcornmix, 2016-02-27:
> down load a tarball ??
https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.bz2 or
https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.xz
doesn't seem to work for me

Those are not the filenames I described. Try:
https://github.com/raspberrypi/linux/archive/rpi-3.18.y.tar.gz

But using git clone is the better option.

ShorTie8, 2016-02-27:
Ok, me sorry, didn't try a gz
Thankz for all your help

closed 2016-02-29 by ShorTie8