raspberrypi/linux #1309: Cx231xx cant detect digital tuner on Hauppauge HVR-850 v1200

Apokalypzx, 2016-02-29:
I have a hauppauge HVR-850 v1200 with the CX231XX chipset. My Raspberry Pi 2 detects the device OK, but when it probes for the digital tuner, it fails and I only get an analog interface. I have tested this device on my x86 PC running the mainline kernel and I've tested it on my Odroid XU4 running kernel 3.10.96. Both of them are able to detect and use the digital tuner and give me a dvb interface. The relevant part of my dmesg is as follows:

[ 1835.682586] usb 1-1.4: new high-speed USB device number 8 using dwc_otg
[ 1835.805478] usb 1-1.4: New USB device found, idVendor=2040, idProduct=b140
[ 1835.814939] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1835.824790] usb 1-1.4: Product: Hauppauge Device
[ 1835.831915] usb 1-1.4: Manufacturer: Hauppauge
[ 1835.838831] usb 1-1.4: SerialNumber: 4034005309
[ 1835.850725] cx231xx 1-1.4:1.1: New device Hauppauge Hauppauge Device @ 480 Mbps (2040:b140) with 7 interfaces
[ 1835.866325] cx231xx 1-1.4:1.1: Identified as Hauppauge EXETER (card=8)
[ 1835.876803] i2c i2c-4: Added multiplexed i2c bus 6
[ 1835.884465] i2c i2c-4: Added multiplexed i2c bus 7
[ 1836.146182] cx25840 3-0044: cx23102 A/V decoder found @ 0x88 (cx231xx #0-0)
[ 1838.127229] cx25840 3-0044: loaded v4l-cx231xx-avcore-01.fw firmware (16382 bytes)
[ 1838.180086] Chip ID is not zero. It is not a TEA5767
[ 1838.187581] tuner 6-0060: Tuner -1 found with type(s) Radio TV.
[ 1838.211385] tda18271 6-0060: creating new instance
[ 1838.220558] Unknown device (110) detected @ 6-0060, device not supported.
[ 1838.229843] tda18271_attach: [6-0060|M] error -22 on line 1285
[ 1838.238188] tda18271 6-0060: destroying instance
[ 1838.250589] tuner 6-0060: Tuner has no way to set tv freq
[ 1838.259538] cx231xx 1-1.4:1.1: v4l2 driver version 0.0.3
[ 1838.412119] tuner 6-0060: Tuner has no way to set tv freq
[ 1838.420310] cx231xx 1-1.4:1.1: Registered video device video0 [v4l2]
[ 1838.429346] cx231xx 1-1.4:1.1: Registered VBI device vbi0
[ 1838.437803] cx231xx 1-1.4:1.1: audio EndPoint Addr 0x83, Alternate settings: 3
[ 1838.502741] cx231xx 1-1.4:1.1: dvb_init: looking for tuner / demod on i2c bus: 6
[ 1838.515054] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1838.527254] lgdt3305_read_reg: error (addr 0e reg 0001 error (ret == -32)
[ 1838.536656] lgdt3305_attach: error -32 on line 1143
[ 1838.544164] lgdt3305_attach: unable to detect LGDT3305 hardware
[ 1838.552639] cx231xx 1-1.4:1.1: Failed to attach LG3305 front end
[ 1838.561422] cx231xx 1-1.4:1.1: video EndPoint Addr 0x84, Alternate settings: 5
[ 1838.571158] cx231xx 1-1.4:1.1: VBI EndPoint Addr 0x85, Alternate settings: 2
[ 1838.580708] cx231xx 1-1.4:1.1: sliced CC EndPoint Addr 0x86, Alternate settings: 2
[ 1838.593164] cx231xx 1-1.4:1.1: TS EndPoint Addr 0x81, Alternate settings: 6

I believe the issue has to do with the line "Unknown device (110) detected @ 6-0060, device not supported." On my Odroid XU4, that line was replaced with "[c0] TDA18271HD/C2 detected @ 6-0060". For more information, I posted on the archlinuxarm.org forum:  https://archlinuxarm.org/forum/viewtopic.php?f=64&t=9857

Ruffio, 2016-08-17:
@Apokalypzx has your issue been resolved? If so, please close this issue. Thanks.

Apokalypzx, 2016-08-18:
The issue hasn't been resolved. I still get the same errors in dmesg.

6by9, 2016-08-18:
Upstream issue . I don't know if this was you posting to linux-media (doubt it as it references x86_64), but it got an answer.
http://www.spinics.net/lists/linux-media/msg98405.html
http://www.spinics.net/lists/linux-media/msg98406.html
> Return it and get an HVR-950q.  This has been a known issue for more
than 5 years and nobody's gotten around to fixing it (largely because
of limitations inside the V4L2 framework).
Devin
(a.k.a. the guy who added the original support for the HVR-850).

If it's not being fixed upstream, then it won't be fixed here.
You could ask on linux-media for more details of what is broken seeing as it worked on your ODroid 3.10.96 kernel and try fixing it. However if the experts say "no" I'd take it as being broken and not fixable.

closed 2016-08-19 by popcornmix