raspberrypi/linux #1310: Pi3B/BCM2837 support

clivem, 2016-02-29:
Does the Pi3B (BCM2837) require any new code to be committed on the kernel side of things, new dt files, etc. etc. or is the rpi-4.4.y already sufficient to support it?

P33M, 2016-02-29:
An ARMv7 kernel boots on bcm2837. A bcm2710 dtb is required, this is shipped with the Raspbian / NOOBS images released today. Sources will be updated in due course.

clivem, 2016-02-29:
Thanks. A quick look at the raspbian image, and I noticed the bcm2710-rpi-3-b.dtb. What source is the Raspbian kernel being built from. It doesn't appear to be from the RPi public repos?

pelwell, 2016-02-29:
As P33M said, "Sources will be updated in due course."

clivem, 2016-02-29:
Yes, I got that. You mean the RPi github sources will be updated in due course. I'm asking for you to point me at the source that the Raspbian kernel was built from, I'll build from that.

pelwell, 2016-02-29:
> I'm asking for you to point me at the source that the Raspbian kernel was built from, I'll build from that.

rpi-4.1.y has been updated.

clivem, 2016-02-29:
@pelwell Thanks, pulling now.

clivem, 2016-02-29:
@pelwell Phil, I'm confused. I was expecting more than a dtb file. I was expecting a new config file, a new ARCH...... Is this sufficient to take an Aarch64 distribution root filesystem, replace the kernel with one built from rpi-4.1.y with this patch, put the other boot binaries in a first FAT partition and actually boot a 64 bit OS?

pelwell, 2016-02-29:
As Eben has said, we'll be running the Pi3 in 32-bit mode until we have a good reason not to - two kernels is enough to manage. Build using bcm2709_defconfig.

clivem, 2016-02-29:
I'm getting more confused by the minute. The only thing I read that Eben said, is that you will investigate whether to have a 64bit Raspbian image. To start with you wont. Fine! I don't want to run Raspbian. 

This is crazy, if RPi are selling hardware, making a big deal of the 64 bit arch, but there isn't anyway to boot it 64 bit. I don't care about Raspbian. I don't and won't use it. I was expecting to see 64 bit kernel support, so that people who do want to take advantage of the hardware, and are willing to run something other than Raspbian, can! ;)

Are you saying that from the kernel side of things, you have no choice but to run the proc in backward compatible 32 bit mode? It isn't an OS limitation.... It's a kernel limitation. There isn't any 64 bit support kernel side?

pelwell, 2016-02-29:
I don't think we've made a big deal of the 64-bit architecture - 50% faster with WiFi and Bluetooth have been the headlines, all for the same price as a Pi2.

clivem, 2016-02-29:
"At launch, we are using the same 32-bit Raspbian userland that we use on other Raspberry Pi devices; over the next few months we will investigate whether there is value in moving to 64-bit mode."

I thought Eben was talking about userland. Can the BCM2837 even be booted in 64 bit mode, or not?

pelwell, 2016-02-29:
Yes it can be booted into 64-bit mode, but the 64-bit support isn't ready to be released yet.

clivem, 2016-02-29:
@pelwell Thank you. We got there in the end. Can I please suggest you put this info somewhere, blog, FAQ...... somewhere where people will see it. If I'd have bought any brand of 64 bit SBC, to be then told that I can only use it in backward compatible 32 bit mode....... This isn't a simple case of, we only have a 32 bit Raspbian image at the moment, which would imply that you can have 64 bit support if you use another image....

bmauduit, 2016-03-02:
Hi,

I agree with @clivem.
RPI-3 is the first arm64 board available at a small price. A 64bit portage of the current kernel is required.
I suppose some people are working on this, do you have any information about that ? And how we can help ?

Thanks

lgeek, 2016-03-02:
I see I'm not the only one caught by surprise that an AAarch64 machine is shipped without a working arm64 kernel. Has anyone started publicly working on a port? I might be able to spare some time to help.

JamesH65, 2016-03-02:
I think the release of the product caught most people by surprise! One big reason there is no 64bit OS ready, is to ensure the Foundation only has to support one distro, that runs on all Pi's. It's double the effort to support both at this stage, and with an installed base of 8M 32bit machines, you really need to keep everything going for those over spending all your dev time on working on a new product that hasn't hit the market yet. I'm sure 64bit will turn up eventually.

philw38, 2016-03-06:
Hmmmm.
Though I can understand the reason given for not supporting - at least right now - the 64 bit mode, I’m a bit disappointed :-( 
From my point of view the interest I see in the 64 bit mode is not the 64 bit addressing capability but the higher efficiency of NEON intensive Aarch64 application (at least this is true for some ARMV8 processors - the Apple A8 and A9 for instance) and I wanted to bench it on the A53. Maybe a limited distro (ethernet, USB, ALSA) could be made available with a reduced amount of work?

robingroppe, 2016-03-06:
Nobody made a big deal about 64bit.
The new cpu was choosen because its a better 32bit cpu.

clivem, 2016-03-06:
> Nobody made a big deal about 64bit.

Sure, nobody marketing or selling the Pi3B is even mentioning the "64-bit" processor.....

And, it is not as if the launch blog post even tried to highlight the important points, by putting the "**1.2GHz 64-bit quad-core**" in bold, is it? ;) Didn't put "~10x the performance of Raspberry Pi 1" in bold, but highlighted the **1.2GHz 64-bit quad-core** in bold! ;)

robingroppe, 2016-03-06:
Well, that are all facts.
I even got a 50-60% Performance gain over my Pi2.
But think about the efford. We would need a whole new Distribution for 64bit.
And the benefits of that will be not near any of those numbers.
Eben Upton said the Processor was chosen because its a better 32bit cpu. That were not my words.
Anyway, nobody said 64bit will not happen just that it needs investigation.
There will be experiments and then we will see how the effords needed are matching the benefits from switching to 64bit.

philw38, 2016-03-06:
@robingroppe 
I haven’t yet received my PI3 but your numbers are what geekbench-2.4.2 gives (I read that somewhere on the web). And yes that gain is already valuable. Hey, we always want more :-) Maybe someday we’ll see a PI4 (or 5) built around a quad core Cortex-A72 :-)

clivem, 2016-03-06:
I've run a bunch of benchmarks, but not really interested in synthetic numbers. Probably the most telling, from my point of view, (assuming you can keep the Pi3B temps south of 80 degrees and away from throttling the arm core speed), is that a full kernel compile is down from 2h38m on a Pi2B to 1h10m on Pi3B.

robingroppe, 2016-03-06:
Yep, its speed is amazing. 
I also see it running hotter than the Pi2.
I applied a heatsink and now its fine. For some reason the governor only switches to 600MHz or 1200MHz. Nothing in between. 
I am also having an issue with the onboard wifi which i am investigating right now.

lgeek, 2016-03-07:
Nobody reasonable is asking for a stable or  ready to run package like raspbian for AArch64 at this point. A good starting point would be to publish a bootloader which can actually launch the kernel in the AArch64 state, so the community can at least start working on a port. An arm64 kernel which can at least print the boot log to the serial port would, of course, be even better.

philw38, 2016-03-07:
+1.
That would be a good starting point.

pelwell, 2016-03-07:
You can already do that - just add `arm_control=0x200` to config.txt.

lgeek, 2016-03-07:
@pelwell That's great. Just wondering, is that documented somewhere where I've missed it? `arm_control` is not mentioned at all in [the config.txt docs](https://www.raspberrypi.org/documentation/configuration/config-txt.md).

pelwell, 2016-03-07:
Yes - here: https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=137963

If you google "starting raspberry pi in 64-bit mode" it's the second link on the page.

The Arch Linux crowd found it.

Ruffio, 2016-08-17:
@clivem has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-17 by clivem