raspberrypi/linux #1311: Support for more than 2 I2S audio channels with the Cirrus Logic CS42448 codec in "One line mode"

doumdi, 2016-02-29:
Hello I2S/Audio and Codec Experts, 

Do you think it would be possible to connect a multi channel audio codec (like the Cirrus Logic CS42448) to the I2S port of the Raspberry Pi? There is a "one line mode" on the CS42448 codec that would be useful for this kind of configuration by sending multiple channels over the same I2S lines at higher clock rates. However, I am not sure we could modify the kernel driver to support this kind of configuration. I don't know if the soc can handle that also. The intended application is audio beamforming with multiple microphones. 

Any thoughts or suggestions are welcomed.

Thank you!

Ruffio, 2016-08-17:
@doumdi has your issue been resolved? If so, please close this issue. Thanks.

flatmax, 2017-01-13:
It is now possible, [see here](http://audioinjector.net/)

doumdi, 2017-01-13:
Thanks Matt, 

It seems it was a good idea after all! Good luck with your kickstarter campaign. 

You managed to get the I2S code into the kernel ? It seems they reverted it to 2 channels max.

Regards,

Dominic

Ruffio, 2017-01-13:
@doumdi has this been resolved? I think perhaps your should use the forum to ask questions as Github is only for solving bugs...

JamesH65, 2017-05-18:
Closing this issue as questions answered/resolved.

closed 2017-05-18 by JamesH65