raspberrypi/linux #1312: cryptsetup aes-cbc-essiv:sha256 cipher supposedly not supported in kernel

gkroon, 2016-02-29:
Hello,

I'm not sure this is the right place, but I fear I've stumbled upon a bug. I'm running the latest Raspbian Jessie on my Raspbian Pi 2 model B with the latest image I downloaded from raspberrypi.org/downloads/raspbian/ (2016-02-29). I've been able to mount my dm-crypt LUKS disk without any trouble before I flashed the new image, but now it seems my kernel doesn't support the aes-cbc-essiv:sha256 cipher:

`pi@pibox:~ $ sudo cryptsetup -v --key-file <my-key-file> luksOpen /dev/sdb data
Failed to setup dm-crypt key mapping for device /dev/sdb.
Check that kernel supports aes-cbc-essiv:sha256 cipher (check syslog for more info).
Command failed with code 22: Failed to setup dm-crypt key mapping for device /dev/sdb.
Check that kernel supports aes-cbc-essiv:sha256 cipher (check syslog for more info).`

I've inserted numerous crypto kernel modules I could find (aes, sha256, and variations thereof I found on my other systems), but sadly to no avail. Is there any way for me to solve this, or is this really a bug in the kernel, cryptsetup, or possibly my own fault?

Looking forward to everyone's suggestions.

gkroon, 2016-03-01:
Apologies, apparently this was already reported in an earlier issue (https://github.com/Hexxeh/rpi-firmware/issues/109). I downgraded my firmware to 4.1.17 and it works again. I guess we'll just have to wait for a fix.

For other people struggling with this, this is how downgraded my firmware to 4.1.17 (hex value is specific github commit):
`sudo rpi-update 0764e7d`

popcornmix, 2016-03-01:
I'll close this as a duplicate. I suspect this will be fixed in 4.1.19.

closed 2016-03-01 by popcornmix