raspberrypi/linux #1314: How to get BT working on Pi3B.

clivem, 2016-03-01:
I figured this was going to be really simple, with something like using hciattach to the UART, to make the device visible.....
Any instructions anywhere?

JamesH65, 2016-03-01:
You really need to post this sort of question on the forum, but in short

sudo apt-get install pi-bluetooth

should install a load of stuff you need. Not sure how to get the panel icon
in LXDE though.

pelwell, 2016-03-01:
sudo apt-get install blueman gets you the GUI (which is just about functional).

clivem, 2016-03-01:
Guys, I'm not using Raspbian. I never will be using Raspbian, I have no interest in Raspbian.

I can see that the BT section of chip is connected to UART from DT config. I don't want to install Debian packages. I am asking for the technical info to be able to get it working on any distribution.
How do I get the device recognised, so that "hcitool dev" actually shows there is a device available.

Don't want blueman, LXDE, any desktop gui..... blah, blah, blah. Just want the device to be recognised.

pelwell, 2016-03-01:
But Raspbian is the supported way. All the information is out there for smart, motivated people to use, but don't expect to be spoon-fed.

clivem, 2016-03-01:
Sorry, Raspbian may be the only supported OS. I can understand that. What I need is the lower level info to make this work with an OS that is not Raspbian. That's the info I am looking for.

Sorry, I must be really stupid. Would you please educate a stupid person where he should look, to find the information I seek. I don't require spoon-feeding, just a pointer to the info. I have looked!

clivem, 2016-03-01:
@pelwell Phil, at least tell me what I am looking for. A Broadcom technical document? Documentation in the PiF documentation github?

pelwell, 2016-03-01:
You need the modified BlueZ package found in Raspbian (source available) and the systemd script called hciattach.service from Raspbian.

XECDesign, 2016-03-01:
http://archive.raspberrypi.org/debian/pool/main/b/bluez/bluez_5.23-2+rpi1.debian.tar.xz
(debian/patches/ 50-53)
`/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -`

clivem, 2016-03-01:
@XECDesign Thank you for the help. Where are those params documented? 

`/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -`
`bcm43xx_init`
`Initialization timed out.`

I have vague recollections of some sort of firmware upload needing to be done before the attach. (Think that was with Wandboard which used a BRCM chipset for wifi/BT). Does something like that need to happen with the Pi3B?

ghollingworth, 2016-03-01:
Have you tried typing hciattach into google?

clivem, 2016-03-01:
Yes, but it returns rather a lot of results. None of which seem to be telling me where to obtain the firmware file. Is it available from PiF github? Or only buried in a Raspbian package?

Cheong2K, 2016-03-05:
I am using a BT module on Pi2, first I enable the UART flow control by:
      gpio -g mode 16 alt3
      gpio -g mode 17 alt3
and then using the brcm-patchram-plus tool to load the BT firmware to the BT chip.
Try google it.

Cheong2K, 2016-03-05:
@clivem,

I don't have a Pi3 at this moment but I tried my Pi2 with NOOBS version 1.8.0, BT works natively, only need to make some scripts.

davidpronk, 2016-03-05:
@clivem 

Did you manage to get wifi to work on the Pi3B without Raspbian? I'm trying to figure out how to do this but I'm kind of in the dark on this...

clivem, 2016-03-05:
@davidpronk Not yet. I got as far as grabbing those patches Phil made to modify the bluez hciattach from the Raspbian package, and rebuilt my Fedora bluez package..... Got the firmware, BCM43430A1.hcd, from Raspbian image..... At which point I was hoping it would just work, but still getting a timeout.....

$ sudo /usr/bin/hciattach -n /dev/ttyAMA0 bcm43xx 921600 noflow -
bcm43xx_init
Set Controller UART speed to 921600 bit/s
Flash firmware /lib/firmware/brcm/BCM43430A1.hcd
Initialization timed out.

I hope to find some time tomorrow to try and figure it out......

The last time I was fighting with Broadcom bluetooth on another SBC, it was still the "two shot" brcm-patchram-plus to initialize firmware, followed by hciattach..... At least now, it's just hciattach responsible for firmware upload and connecting the UART to the Bluetooth subsystem.

Noltari, 2016-03-06:
@pelwell "All the information is out there for smart, motivated people to use, but don't expect to be spoon-fed".

Completely disagree with this one.
The information isn't provided anywhere, but on several forums where people is asking how to get things working.
Official developers only made Raspbian compatible and the information needed to get things working must be extracted from there.
However, it should be properly documented somewhere...
Firmwares aren't available on linux-firmware...

pelwell, 2016-03-06:
> The information isn't provided anywhere, but on several forums where people is asking how to get things working.
> Official developers only made Raspbian compatible and the information needed to get things working must be extracted from there.

Do you not see the contradiction here?

And to reiterate:
> You need the modified BlueZ package found in Raspbian (source available) and the systemd script called hciattach.service from Raspbian.

Noltari, 2016-03-06:
> Do you not see the contradiction here?
Not at all, because I don't consider that a proper documentation.

clivem, 2016-03-06:
> And to reiterate:

@pelwell Are you familiar with the concept of a BSP/SDK?

pelwell, 2016-03-06:
Do you have any examples of previous RPi BSP/SDKs?

clivem, 2016-03-06:
> Do you have any examples of previous RPi BSP/SDKs?

Did you ever release a board, before now, that provided Bt/wifi, that required proprietary firmware to make it work, (not hosted in linux-firmware), and required modification to userspace software, to make it work? 

Phil, I'm not trying to wind you up. Eben's blog, named you .....

> Phil Elwell developed the wireless LAN and Bluetooth software.

.... so it would seem that you might be the best person to put the information together in a format that would be suitable, for other people, who do not wish to use Raspbian, to be able to get it working on whatever other OS, they choose to use.

Noltari, 2016-03-06:
@clivem 
With 921600 or non defined baud rate I also get "Initialization timed out."
However, if I switch to 115200 I get the following:
```
root@OpenWrt:/# hciattach -n /dev/ttyAMA0 bcm43xx 115200 noflow -
[   29.461777] uart-pl011 3f201000.uart: no DMA platform data
bcm43xx_init
Set Controller UART speed to 115200 bit/s
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 115200 bit/s
Device setup complete
```
And then it hangs...

By the way, try this with fresh boot, since it appears that device reset isn't properly working...

Noltari, 2016-03-06:
Okay, didn't notice the "-n" option prevents detaching from controlling terminal.

It's working now, but only a 115200...
```
root@OpenWrt:/# hciattach /dev/ttyAMA0 bcm43xx 115200 noflow -
[   22.634305] uart-pl011 3f201000.uart: no DMA platform data
bcm43xx_init
Set Controller UART speed to 115200 bit/s
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 115200 bit/s
Device setup complete
root@OpenWrt:/# hciconfig
hci0:   Type: BR/EDR  Bus: UART
        BD Address: B8:27:EB:XX:XX:XX  ACL MTU: 1021:8  SCO MTU: 64:1
        DOWN
        RX bytes:654 acl:0 sco:0 events:33 errors:0
        TX bytes:419 acl:0 sco:0 commands:33 errors:0
root@OpenWrt:/# hcitool scan
Scanning ...
        90:B9:31:XX:XX:XX       NoltariPhone
```

pelwell, 2016-03-06:
Are you using the patched hciattach source from the Raspbian distribution?

clivem, 2016-03-06:
Phil, where does your patch set fit in with what was committed upstream to bluez, from Loic Poulain, back in 2014.

http://git.kernel.org/cgit/bluetooth/bluez.git/commit?id=beb4892e96785ba9ff429fb940126a815efd47fd

Noltari, 2016-03-06:
@pelwell Nope, but I'm using a patched version of the OpenWrt one:
https://github.com/openwrt/packages/pull/2464
As you can see I added the raspbian patches there.
@clivem yeah I had issues with patch #0051 because of that:
https://github.com/openwrt-es/openwrt-packages/blob/bcd5ea4533f5ddb650b3a413152b03a63d6ab09c/utils/bluez/patches/301-bcm43xx-The-UART-speed-must-be-reset-after-the-firmw.patch

clivem, 2016-03-06:
This is madness..... If we need to modify userspace code, can we please have "reference" patches against current upstream bluez, not some antique version that Debian/Raspbian is using..... Pretty please?

pelwell, 2016-03-06:
Try these: https://gist.github.com/pelwell/c8230c48ea24698527cd

pelwell, 2016-03-06:
Since the firmware is the most important data for a modem, increasing the speed before uploading the firmware on a modem without flow control is not a great idea. The other patches add h5 support, but this isn't being used yet.

Noltari, 2016-03-06:
I tested those patches and there are no more tiemouts, however with baud rates different to 115200 the device doesn't show up:

921600:
```
root@OpenWrt:/# hciattach /dev/ttyAMA0 bcm43xx-3wire 921600
[  316.863846] uart-pl011 3f201000.uart: no DMA platform data
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
root@OpenWrt:/# hciconfig
root@OpenWrt:/# hciconfig hci0 up
Can't get device info: No such device
```

Default (3000000):
```
root@OpenWrt:/# hciattach /dev/ttyAMA0 bcm43xx-3wire
[   24.839780] uart-pl011 3f201000.uart: no DMA platform data
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 3000000 bit/s
Device setup complete
root@OpenWrt:/# hciconfig
root@OpenWrt:/# hciconfig hci0 up
Can't get device info: No such device
```

clivem, 2016-03-06:
I've patched the bluez 5.36 Fedora 23 package....

$ sudo hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -
bcm43xx_init
Flash firmware /lib/firmware/brcm/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete

$ hciconfig 
hci0:	Type: BR/EDR  Bus: UART
	BD Address: B8:27:EB:4C:9B:56  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:654 acl:0 sco:0 events:33 errors:0
	TX bytes:419 acl:0 sco:0 commands:33 errors:0

clivem, 2016-03-06:
@pelwell BTW, the gist patches.... Thank you, but not to appear ungrateful, the first three don't apply until the spaces are changed to tabs..... ;)

Noltari, 2016-03-06:
@clivem I get the following when I use your hciattach parameters:
```
root@OpenWrt:/# hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -
[   36.624301] uart-pl011 3f201000.uart: no DMA platform data
bcm43xx_init
Flash firmware /lib/firmware/BCM43430A1.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
root@OpenWrt:/# [   42.967027] Bluetooth: hci0 command 0x1003 tx timeout
[   44.967021] Bluetooth: hci0 command 0x1001 tx timeout
[   46.967022] Bluetooth: hci0 command 0x1009 tx timeout

root@OpenWrt:/# hciconfig
hci0:   Type: BR/EDR  Bus: UART
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:0 acl:0 sco:0 events:0 errors:0
        TX bytes:12 acl:0 sco:0 commands:3 errors:0
```
Timeouts, timeouts everywhere xD

pelwell, 2016-03-07:
Gist drag & drop doesn't work for me - why isn't there an upload button? - and the tabs got lost in the cut & paste. It should be fixed now.

pelwell, 2016-03-07:
FYI there is a new overlay - pi3-miniuart-bt - that enables Bluetooth using the mini-UART. There may be throughput limitations, and it is even more important to clamp cpu_freq to 250, so caveat emptor. You will also need to edit the systemd service `/lib/systemd/system/hciuart.service`.

See https://github.com/raspberrypi/linux/commit/866cf9417ee07f6695f28af068a518b71df5916c.

clivem, 2016-03-07:
With thanks to @pellwell, packaged the wifi and BT stuff for Fedora23.

BCM43430 wifi (firmware bin and nvram txt)
SRC: http://www.squeezecommunity.org/repo/fedora/23/SRPMS/brcm43430-firmware-1.0-3.fc23.src.rpm
RPM: http://www.squeezecommunity.org/repo/fedora/23/armhfp/brcm43430-firmware-1.0-3.fc23.noarch.rpm

BRCM4348 bluetooth (systemd service, udev rule to "up" hci, and hcd firmware}
SRC: http://www.squeezecommunity.org/repo/fedora/23/SRPMS/brcm43438-bluetooth-1.0-3.fc23.src.rpm
RPM: http://www.squeezecommunity.org/repo/fedora/23/armhfp/brcm43438-bluetooth-1.0-3.fc23.noarch.rpm

Modified bluez (PhilE patches for hciattach firmware upload)
SRC: http://www.squeezecommunity.org/repo/fedora/23/SRPMS/bluez-5.36-1.fc23.2.src.rpm
RPM: http://www.squeezecommunity.org/repo/fedora/23/armhfp/bluez-5.36-1.fc23.2.armv7hl.rpm

Once the license conditions surrounding those Broadcom wifi/BT firmware binaries are clarified, and if they are re-distributable, I will make a Fedora 23 image available.

Noltari, 2016-03-10:
By the way, I did more tests yesterday and I was able to get WiFi working.
I was getting some exceptions of firmware load that were fixed with these patches:
https://dev.openwrt.org/changeset/48959
(OpenWrt uses a v4.4 custom kernel)
I also sent a couple of patches to openwrt-devel which enable WiFi support on the Raspberry Pi 3:
http://patchwork.ozlabs.org/patch/595197/
http://patchwork.ozlabs.org/patch/595198/

On the other hand, I did some more tests and I couldn't get Bluetooth working at baud rates > 115200 (timeouts), but it was working perfectly with baud rates <= 115200.
It may be related to the kernel config used in OpenWrt:
https://github.com/openwrt-es/openwrt/blob/brcm2708-next/target/linux/brcm2708/bcm2710/config-4.4
I'm still investigating why this is happening.

BTW thanks for sharing those links @clivem.

markab, 2016-03-15:
Can I just ask, is the internal bluetooth implementation in the standard release for rpi3 supposed to work? I seem to have many compatibility issues connecting devices, but the same devices work on the same pi3 with an external bluetooth dongle.

agherzan, 2016-04-13:
@pelwell I see a couple of patches you sent for bluez5. Are these patches intended to go upstream in bluez5 or are they rpi3 specific?

pelwell, 2016-04-13:
They are Pi3 specific, at least in their current form. Any upstream patch would have to be more extensive, adding extra command-line parameters to make the behaviour configurable.

WayneKeenan, 2016-05-26:
@pelwell   Thank you, I applied your patches to BlueZ 5.39, please look here for context:  https://github.com/ukBaz/python-bluezero/issues/30

There are a few other Bluetooth/BlueZ/BLE related issues going on but without going into details here I wondered if you happen have any other Pi3 (or Pi2) specific BlueZ patches?

Thanks
Wayne

Ruffio, 2016-08-17:
@clivem has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-17 by clivem

oliv3r, 2019-03-02:
> With thanks to @pelwell, I packaged the wifi and BT stuff for Fedora 23.
> 
> BCM43430 wifi (Broadcom proprietary firmware bin and nvram txt)
> SRC: http://www.squeezecommunity.org/repo/fedora/23/SRPMS/brcm43430-firmware-1.0-3.fc23.src.rpm
> RPM: http://www.squeezecommunity.org/repo/fedora/23/armhfp/brcm43430-firmware-1.0-3.fc23.noarch.rpm
> Once the license conditions surrounding those Broadcom wifi/BT firmware binaries are clarified, and if they are re-distributable, I will make a Fedora 23 image available.

@clivem have you ever manage to wrap this up? What is the status of the firmware.bin? Since the raspberry pi org is redistributing them, they should be redistributable implicitly already? If so, what is the canonical source for these firmware's? They still haven't shown up in linux-firmware.git ...

islight, 2021-04-26:
> I am using a BT module on Pi2, first I enable the UART flow control by:
> gpio -g mode 16 alt3
> gpio -g mode 17 alt3
> and then using the brcm-patchram-plus tool to load the BT firmware to the BT chip.
> Try google it.
> 
> You also need to remove the console=ttyAMA0 in /boot/cmdline.txt

it works for me