raspberrypi/linux #1315: Configuring I2C pin allocation

6by9, 2016-03-01:
Just a question on how you'd like stuff done.
I'm wanting to add device tree parameters for setting the I2C GPIO pins, mainly so I can get to the camera via i2c0 and GPIOs 28&29. 

Would you prefer they were added as overrides on the base dt, or by following the approach in https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=120938&p=825883 (following @pelwell 's recommendation in https://www.raspberrypi.org/forums/viewtopic.php?p=779990#p779990)?

I have lifted the overlays from that forum post and have a patch sorted (with appropriate credit to the author), but wanted to confirm that was your preferred approach before creating a PR.
Thanks.

pelwell, 2016-03-01:
Create a PR with an overlay (or as many as you need). The risks with overlays are non-existent - if you don't use them they do nothing - and they are slightly less "approved" than a mod to the base DTBs would be.

6by9, 2016-03-02:
Ta. Will push later.

closed 2016-03-05 by 6by9