raspberrypi/linux #1317: Raspberry Pi Compute Module error  mmc0: command never completed

JiapengLi, 2016-03-03:
I am using the Raspbian Jessie 2016-02-26 image, and I got some mmc errors during the system booting. When return back to wheezy 2014-12-2, the issue disappears.

```
Raspbian GNU/Linux 8 raspberrypi ttyAMA0

raspberrypi login: Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.18+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #846 Thu Feb 25 14:11:56 GMT 2016
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Compute Module Rev 1.0
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x14 bcm2708.serial=0xd694b121 smsc95xx.macaddr=B8:27:EB:94:B1:21 bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 436836K/458752K available (5792K kernel code, 488K rwdata, 1788K rodata, 352K init, 720K bss, 13724K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xdc800000 - 0xff000000   ( 552 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc076f1f0   (7581 kB)
[    0.000000]       .init : 0xc0770000 - 0xc07c8000   ( 352 kB)
[    0.000000]       .data : 0xc07c8000 - 0xc0842260   ( 489 kB)
[    0.000000]        .bss : 0xc0842260 - 0xc08f63e0   ( 721 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:522
[    0.000000] clocksource stc: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000015] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000056] Switching to timer-based delay loop, resolution 1000ns
[    0.000345] Console: colour dummy device 80x30
[    0.001271] console [tty1] enabled
[    0.001328] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001412] pid_max: default: 32768 minimum: 301
[    0.001790] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001860] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002911] Initializing cgroup subsys blkio
[    0.003005] Initializing cgroup subsys memory
[    0.003075] Initializing cgroup subsys devices
[    0.003137] Initializing cgroup subsys freezer
[    0.003199] Initializing cgroup subsys net_cls
[    0.003318] CPU: Testing write buffer coherency: ok
[    0.003437] ftrace: allocating 19747 entries in 58 pages
[    0.107910] Setting up static identity map for 0x81c0 - 0x81f8
[    0.110088] devtmpfs: initialized
[    0.120934] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.121388] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.123380] pinctrl core: initialized pinctrl subsystem
[    0.124279] NET: Registered protocol family 16
[    0.129773] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.131274] bcm2708.uart_clock = 3000000
[    0.136147] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.136224] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.136432] Serial: AMBA PL011 UART driver
[    0.136734] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a PL011 rev2
[    0.505638] console [ttyAMA0] enabled
[    0.510035] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.556094] bcm2708-dmaengine 20007000.dma: DMA legacy API manager at f2007000, dmachans=0xf35
[    0.564823] bcm2708-dmaengine 20007000.dma: Initialized 7 DMA channels (+ 1 legacy)
[    0.573393] bcm2708-dmaengine 20007000.dma: Load BCM2835 DMA engine driver
[    0.580322] bcm2708-dmaengine 20007000.dma: dma_debug:0
[    0.586477] SCSI subsystem initialized
[    0.590638] usbcore: registered new interface driver usbfs
[... 155 lines cut ...]
[    2.551513] mmc0:>cmd op 1 arg 0x0 flags 0xe1 - resp 00000000 00000000 00000000 00000000, err 0
[    2.562233] mmc0: =========== REGISTER DUMP ===========
[    2.568589] mmc0: SDCMD  0x00004001
[    2.573212] mmc0: SDARG  0x00000000
[    2.577825] mmc0: SDTOUT 0x0000ee88
[    2.582436] mmc0: SDCDIV 0x000007fd
[    2.587055] mmc0: SDRSP0 0x40ff8080
[    2.591682] mmc0: SDRSP1 0x00003f7f
[    2.596303] mmc0: SDRSP2 0xffffffff
[    2.600929] mmc0: SDRSP3 0xffffffff
[    2.605550] mmc0: SDHSTS 0x00000010
[    2.610173] mmc0: SDVDD  0x00000001
[    2.614791] mmc0: SDEDM  0x00010800
[    2.619422] mmc0: SDHCFG 0x0000040a
[    2.624011] mmc0: SDHBCT 0x00000000
[    2.628537] mmc0: SDHBLC 0x00000000
[    2.633019] mmc0: ===========================================
[    2.813668] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.934061] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[    2.942144] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.953755] smsc95xx v1.0.4
[    3.018134] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-20980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:94:b1:21
[   70.584955] random: nonblocking pool is initialized

```

pelwell, 2016-03-03:
Because sdhost is still a relatively new driver which has had a major revision recently, there is more logging output under error conditions than might otherwise be the case.

My understanding is that an MMC device will return an error to CMD1 (SEND_OP_COND) until it is ready, and that this can take 100s of milliseconds.It is also possible that it is slow to respond and that the timeout in the new driver is too short. However, polling until the command succeeds is valid behaviour.

If these few initial CMD1 timeouts ("command never completed", with SDCMD set to 0x00004001) are the only errors you are seeing then I don't think you need to be concerned. I'll either make sure the messages disappear in a future update.

JiapengLi, 2016-03-04:
@pelwell Thank you for your reply. My issue is when I get these errors Compute Module stops starting, 
the firmware stuck at some place. The log I pasted is the full log I could get during the error starting, after that it stops.

dpslwk, 2016-03-04:
I've also seen the same on Compute Modules with the jessie-lite 26th image
Plug in a USB to ethernet adapter and boot stalls with logs very similar to above, 
also seen it on occasion with a USB Wifi dongle (RT5370)

pelwell, 2016-03-04:
I recommend you add `dtoverlay=mmc` to your config.txt, which will revert to using the old driver and interface, until the problem is better understood.

bggardner, 2016-03-30:
I've had the same error message(s) with multiple Compute Modules.  Some CMs would boot, then error after a number of reboots, and others would error at first boot.  It appears to be worse in the 2016-03-18 image, but not confirmed.  Adding `dtoverlay=mmc` to `/boot/config.txt` as @pelwell suggested seems to have fixed the issue (thanks!).

jaanvi, 2016-04-11:
I have the same problem with external I2C device connected to the raspberry pi compute module with jessie-lite(2016-03-18). Adding dtoverlay=mmc to /boot/config.txt  has fixed the issue.

benpye, 2016-06-16:
I appear to be hitting this same issue when trying to use the sdio overlay and reverting the eMMC interface to mmc. Has there been any progress on this issue? I am currently being blocked by this as I require an SD card for user accessible files.

https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=151655

pelwell, 2016-06-16:
Can you explain why you think you can't use the sdhost driver on the external SD interface? Yes SDIO using sdhost is problematic because it aggressively clock gates the SD clock, and for that reason SDIO support has never been tested on it which means it's almost certainly broken, but a regular SD card should be just fine. If you can't figure out the DT overlay needed, create a new issue including what you've done so far and I'll take a look. Alternatively you could go straight to a Pull Request, with the understanding that there will be some back-and-forth before the finished overlay becomes part of the standard distribution.

Driving the eMMC through SDHOST appears to be failing because an early timeout is marginal, and I haven't yet found a way to increase it. I'm hoping to get a bit of Broadcom support on this.

Ruffio, 2016-08-17:
@JiapengLi has your issue been resolved? If so, please close this issue. Thanks.

JiapengLi, 2016-08-18:
@Ruffio Thank you for notification. It is solved.

closed 2016-08-18 by JiapengLi

mildred, 2017-12-03:
I'm getting this issue with ArchLinux ARM (don't know when it was last updated, this iw a read only filesystem. It appears at bootup and blocks boot. I have a 4G/LTE device connected when this happens. If I disconnect the usb port (the LTE/4G dongle) the boot completes correctly and adding the dongle later in boot corrects the problem.

JamesH65, 2017-12-03:
Does this happen with Raspbian? We don't support Arch, but if it happens on Raspbian then we can take a look. Also need to know the type and vendor ID of the USB device.

mildred, 2017-12-08:
the usb 4G dongle is a Huawei E3372 (12d1:14dc)

JamesH65, 2017-12-08:
Does it still go wrong under Raspbian?

JamesH65, 2017-12-08:
If you hot plug the device (ie plug in after boot), what does lsusb and dmsg say?

mildred, 2018-01-04:
Sorry, I haven't tried. I changed the SD card for a new one and I don't have the problem any more. Perhaps the SD card was too old. The filesystem on it was mounted read-only to avoid damaging the SD card too early, but it didn't work well enough. Previously, I had a number of problems with linux system on flash devices when not mounted read only. The flash device (sd card and usb drive, both) would corrupt really quickly.