raspberrypi/linux #1319: kexec new kernel freeze after show Starting new kernel

huangjiansword, 2016-03-03:
step 1,build new kernel
I build a new kernel from source 3.18.7+ follow https://www.kernel.org/doc/Documentation/kdump/kdump.txt
CONFIG_KEXEC=y
CONFIG_SYSFS=y
CONFIG_DEBUG_INFO=Y
CONFIG_CRASH_DUMP=y
CONFIG_PROC_VMCORE=y
AUTO_ZRELADDR=y

then I copy new zImage from arch/arm/boot to SD Card /boot, and delete old kernel.img, then rename zImage to kernel.img

then I reboot the pi (A+), everything is OK.

and I checked config parameter with zless /proc/config.gz, it's all correct.

step2: 
I install kexec-tools at the new kernel pi.

sudo apt-get install kexec-tools

kexec's version is 2.0.3 ( but I tried to build kexec 2.0.11 and installed, also freeze at the same place)

step 3,run new kernel
 when I run this:
sudo kexec --type=zImage /boot/kernel.img --command-line="$(cat /proc/cmdline)" 

it show that the device is shutdown and  umount local filesystems

but freezed after show "kexec: Starting new kernel"

what's wrong with it?

huangjiansword, 2016-03-03:
pi@raspberrypi ~ $ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait

pi@raspberrypi ~ $ cat /boot/config.txt
# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

huangjiansword, 2016-03-03:
`pi@raspberrypi ~ $ uname -a
Linux raspberrypi 3.18.7+ #33 PREEMPT Thu Mar 3 00:42:32 CST 2016 armv6l GNU/Linux
`

huangjiansword, 2016-03-03:
I tried many method from this reference:
http://www.bootc.net/archives/2012/05/22/another-new-rpi-kernel/
https://github.com/raspberrypi/linux/issues/27
https://github.com/raspberrypi/linux/issues/958
https://www.kernel.org/doc/Documentation/kdump/kdump.txt
http://www.dedoimedo.com/computers/kdump.html
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=24670&sid=bea05ce6eda0f6c50fb2c99ccbb2ab33
https://www.raspberrypi.org/forums/viewtopic.php?f=71&t=16703&p=170239&hilit=kexec#p170239

but the issue remains.

Ruffio, 2016-08-17:
@huangjiansword has your issue been resolved? If so, please close this issue. Thanks.

popcornmix, 2016-08-17:
Duplicate of https://github.com/raspberrypi/linux/issues/27

closed 2016-08-17 by popcornmix