raspberrypi/linux #1322: can't set CONFIG_DRM_VC4=y in 4.5

greyltc, 2016-03-03:
I can't seem to set `CONFIG_DRM_VC4=y` when I use `make menuconfig` in the rpi-4.5.y branch. Although I can in rpi-4.1.y.

I think I should be able to do this based on these words that I read on [this site](https://dri.freedesktop.org/wiki/VC4/):
> The Raspberry Pi Foundation's kernel tree now includes the driver in the current rpi-update 4.1 branch, and also the 4.5 branch.

@anholt or anyone else, is this a bug or am I doing something wrong?

ED6E0F17, 2016-03-03:
Seems to be missing a patch:

"depends on ARCH_BCM2835 || COMPILE_TEST"

in  https://github.com/raspberrypi/linux/blob/rpi-4.5.y/drivers/gpu/drm/vc4/Kconfig

popcornmix, 2016-03-04:
I've added a commit that should allow CONFIG_DRM_VC4 to be enabled for BCM270x.
@greyltc does that help?

ED6E0F17, 2016-03-04:
arch/arm/boot/dts/overlays/vc4-kms-v3d-overlay.dts is missing in 4.5

popcornmix, 2016-03-04:
cherry-picked...

greyltc, 2016-03-04:
@popcornmix I can now set `CONFIG_DRM_VC4=y`, thanks!

closed 2016-03-04 by greyltc