raspberrypi/linux #1324: build failure with CONFIG_DRM_VC4=y on 4.5 branch

greyltc, 2016-03-04:
When I try to build the 4.5 branch with `CONFIG_DRM_VC4=y` I get the following build failure.
@popcornmix, maybe some of [the cherrypicks you recently made related to this](https://github.com/raspberrypi/linux/issues/1322) weren't complete. Looks like we're missing a `linux/basic_mmio_gpio.h`:
```
drivers/gpio/gpio-bcm-virt.c:16:35: fatal error: linux/basic_mmio_gpio.h: No such file or directory
compilation terminated.
scripts/Makefile.build:258: recipe for target 'drivers/gpio/gpio-bcm-virt.o' failed
make[2]: *** [drivers/gpio/gpio-bcm-virt.o] Error 1
scripts/Makefile.build:407: recipe for target 'drivers/gpio' failed
```

popcornmix, 2016-03-05:
I've pushed a fix. Can you try now?

greyltc, 2016-03-05:
Thanks very much. It builds now.

closed 2016-03-05 by greyltc