raspberrypi/linux #1326: rpi3: TTYS0 does run too slow

msperl, 2016-03-06:
The new TTYS0 as the default interface does not run at 115.2k but instead at 71.8k using firmware version: 6e8b794818e06f50724774df3b1d4c6be0b5708c.

msperl, 2016-03-06:
So looking at the DT I see:
* we use a multiplier clock clk_uart1 for the auxiliar clock 
  * we multiply the core clock by 2
* core clock is 400MHz (cat /sys/kernel/debug/clk/core/clk_rate)
* so the multiplier clock (to make the uart driver happy, as the HW only uses a internal divider of 8 not 16 as typical other divices) is 800MHz. (cat /sys/kernel/debug/clk/clock/clk_rate)

But this seems to be incorrect - the actual clock should (still) be 500MHz.
115200/800*500 = 72000kHz
So this still means we run the VPU clock at 250MHz while we assume it runs at 400MHz.

So maybe this is a firmware issue not a kernel issue itself.

pelwell, 2016-03-06:
1) Remember that the downstream RPi DT isn't changing any clock settings, it is telling Linux what they are. By declaring clk_uart1 to be twice the actual frequency we are getting the uart driver to generate a divisor that is twice as large. This factor of two accounts for the discrepancy in the oversampling rate:

    baudrate = core_clk/(16 * div) = core_clk/(8 * div'), where div' = div*2

2) The actual difference in baudrate stems form the fact that, by default, the core clock varies with turbo mode. The default turbo core clock is 400MHz, while the minimum, non-turbo clock is 250MHz. If you wish to use the mini uart, and not everybody does, then you need to stop the core clock from changing. If you have an adequate power supply to avoid under-voltage, and adequate cooling for the work load, then you can set `force_turbo=1` to keep the core clock at 400MHz. If instead either of those two pre-requisites is in doubt then your only option is to limit the core frequency to 250MHz (`core_freq=250`).

msperl, 2016-03-06:
The device tree shows:
* uart1_clk is a multiple of 2 of core
* core is a fixed clock with a value of 400MHz - i.e. this is hardcoded in the device-tree!

This results in a uart1_clk of 800MHz

Looking at the distinct registers for vpu, PLLC_CORE0 and PLLC.

It seems as if we really run:
* PLLC at 1GHz (19.2MHz * 52.08333206176757812500)
* PLLC_core0 at 500MHz (divider of 2)
* vpu at 125MHz (divider of 4)

P.s: here the registers values useing peekpoke with the ARM mapped io registers for the RASP2):
*PLLC:
 * registers:
  * A2W_PLLC_CTRL: Value at address 0x3F102120 (0x76f64120): 0x21034
  * A2W_PLLC_FRAC: Value at address 0x3F102220 (0x76f69220): 0x15554
  * A2W_PLLC_ANA1: Value at address 0x3F102134 (0x76f70134): 0x613277
 * That results in:
  * ndiv = 0x34
  * fdiv = 0x15554/(1<<20)
  * pdiv = 0x01
 * so the effective multiplier is 52.08333206176757812500
  * this means 1GHz (19.2MHz*52.0833)
* PLLC_CORE:
 * registers:
  * A2W_PLLC_CORE0: Value at address 0x3F102620 (0x76f1e620): 0x2
 * That results in:
  * divider = 2
 * so this means PLLC_CORE0 runs at 500MHz

Note there is one clock register called VPU - the register values are:
* CM_VPU_CTL: Value at address 0x3F101008 (0x76fa3008): 0x245
* CM_VPU_DIV: Value at address 0x3F10100C (0x76efb00c): 0x4000
This means the plld_core0 parent (=5) is used and divided by 4, so it runs at 125MHz.
But that does not fit anywhere into the picture...

pelwell, 2016-03-06:
See my previous response.

msperl, 2016-03-06:
@pelwell I see what you mean - but the main problem is that we have a mismatch between what the DT says (400MHz) and what the values that the firmware sets the clock registers to (250MHz) right during boot.

If I plug in the RPI3 and directly after boot I connect to the serial, then I would expect it to work as before with the "correct" baud rate.

Note that changing boot/config.txt to include "core-freq=250" changes:
root@rasp3a:~# cat /sys/kernel/debug/clk/clock/clk_rate
500000000
while without it contains:
root@rasp3a:~# cat /sys/kernel/debug/clk/clock/clk_rate
800000000

So my point is:
* the firmware sets the PLLC_core to 500 in any case (=non turbo mode) during boot (so core=250).
* but this does not match with the device-tree, which is still set to 400MHz.

But when setting "core-freq=250" it changes the fixed value inside the device-tree to 250.

When setting "force_turbo=1" the UART still runs at 72kBaud after boot and the pllc registers are still set identical.

I guess the only real solution would be moving to the clock framework to use the actually configured values (still with those "variations" due to turbo) instead. But then we still have the register conflict for i2s right now. Maybe when the new framework includes the pcm clock then we can really merge it.

For now I will have to use "dtoverlay=pi3-disable-bt" or "core_freq=250".

Maybe you should add some more specific comments about the UART in /boot/config.txt.
Actually I guess you should disable it by default, because it does not behave as expected with a default install using the "wrong" baud-rates...

dl9pf, 2016-03-06:
The board should come up in the default config with the right frequency. So its a bug that should be fixed in the next release of the firmware or raspbian. I tried 3 different usb/serial converters before finding this issue as it is not what you expect.

Ruffio, 2016-08-17:
@msperl has your issue been resolved? If so, please close this issue. Thanks.

msperl, 2016-08-30:
I have to admit I have not checked with a new pristine install on a PI3, but as the UART is by default disabled now (and that there is an overlay to expose ttyAMA instead of ttyS0) that probably changes everything...

closed 2016-08-30 by msperl