raspberrypi/linux #1330: mcp3422.ko Kernel module

mypiandrew, 2016-03-08:
Hi 

Would it be possible to include a mcp3422.ko file as part of the standard build (in a similar fashion to the can bus mcp251x.ko please?

pelwell, 2016-03-08:
CONFIG_MCP3422=m is the required setting. The module is only 10KB.

@popcornmix?

popcornmix, 2016-03-08:
Sounds okay. Will add.

popcornmix, 2016-03-09:
This is in latest rpi-update firmware. 
@mypiandrew please test and close if happy.

mypiandrew, 2016-03-09:
Hi,  Wow that was quick!

I'm probably doing something silly here, but have done rpi-update which ran sucessfully (still running on Wheezy at the moment) 

can't get the modules to install : 

root@raspberrypi:/# uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux

root@raspberrypi:/# insmod /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko
Error: could not insert module /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko: Unknown symbol in module

root@raspberrypi:/# insmod /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp320x.ko
Error: could not insert module /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp320x.ko: Unknown symbol in module

Not sure why there are two different .ko files in that directory, was only expecting one - but then kernel modules is not my forte to be honest

Just for fun I tried the other module area :

root@raspberrypi:/# insmod /lib/modules/4.1.19-v7+/kernel/drivers/iio/adc/mcp3422.ko
Error: could not insert module /lib/modules/4.1.19-v7+/kernel/drivers/iio/adc/mcp3422.ko: Invalid module format

That's expected I guess as I'm not running that kernel

This might be helpful :
[  569.320659] mcp3422: disagrees about version of symbol module_layout
[  606.009641] mcp3422: Unknown symbol devm_iio_device_alloc (err 0)
[  606.009697] mcp3422: Unknown symbol devm_iio_device_register (err 0)
[  778.460881] mcp320x: Unknown symbol devm_iio_device_alloc (err 0)
[  778.460949] mcp320x: Unknown symbol iio_device_unregister (err 0)
[  778.460981] mcp320x: Unknown symbol iio_device_register (err 0)
**[  937.631274] mcp3422: Unknown symbol devm_iio_device_alloc (err 0)
[  937.631327] mcp3422: Unknown symbol devm_iio_device_register (err 0)**

The last 2 were generated by running :

insmod /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko

Any ideas as to what's not happening here ?

popcornmix, 2016-03-09:
Does this work for you:
```
pi@domnfs:~ $ sudo modprobe mcp3422
pi@domnfs:~ $ lsmod
Module                  Size  Used by
mcp3422                 3979  0 
industrialio           34435  1 mcp3422
rpcsec_gss_krb5        21204  0 
joydev                  9024  0 
evdev                  10278  6 
brcmfmac              177635  0 
brcmutil                5501  1 brcmfmac
cfg80211              407580  1 brcmfmac
rfkill                 16036  2 cfg80211
bcm2835_gpiomem         2860  0 
bcm2835_rng             1763  0 
uio_pdrv_genirq         2944  0 
uio                     7753  1 uio_pdrv_genirq
i2c_dev                 5671  0 
fuse                   80694  1 
ipv6                  338660  44 
```

popcornmix, 2016-03-09:
I assume you rebooted after running rpi-update?
What does ``uname -a`` report?

mypiandrew, 2016-03-09:
Yes I rebooted 

root@raspberrypi:/# modprobe mcp3422
ERROR: could not insert 'mcp3422': Unknown symbol in module, or unknown parameter (see dmesg)

root@raspberrypi:/# uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux

root@raspberrypi:/# dmesg | tail -5
[  937.631274] mcp3422: Unknown symbol devm_iio_device_alloc (err 0)
[  937.631327] mcp3422: Unknown symbol devm_iio_device_register (err 0)
[ 1222.438448] i2c i2c-1: Failed to register i2c client mcp3424 at 0x6e (-16)
[ 1321.081637] mcp3422: Unknown symbol devm_iio_device_alloc (err 0)
[ 1321.081690] mcp3422: Unknown symbol devm_iio_device_register (err 0)

mypiandrew, 2016-03-09:
I'm still running Wheezy if that has any bearing on things?

popcornmix, 2016-03-09:
Lets check with Pi1
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux
pi@domnfs:~ $ sudo modprobe mcp3422
pi@domnfs:~ $ lsmod
Module                  Size  Used by
mcp3422                 4965  0 
industrialio           45124  1 mcp3422
cfg80211              479279  0 
rfkill                 21508  1 cfg80211
joydev                  9320  0 
evdev                  10534  2 
bcm2835_gpiomem         3703  0 
bcm2835_rng             2207  0 
uio_pdrv_genirq         3526  0 
uio                    10078  1 uio_pdrv_genirq
i2c_dev                 6386  0 
fuse                   86348  1 
ipv6                  358811  28 
```
still seems okay. What does this report:
```
pi@domnfs:~ $ md5sum /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko
cd7a7db6b0608ed8658f0bfdd1e975ff  /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko
```

mypiandrew, 2016-03-09:
I'm sorry about this, I didn't mean to cause trouble :(

mypiandrew, 2016-03-09:
root@raspberrypi:/# md5sum /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko
cd7a7db6b0608ed8658f0bfdd1e975ff  /lib/modules/4.1.19+/kernel/drivers/iio/adc/mcp3422.ko

popcornmix, 2016-03-09:
Wheezy is no longer supported by us and I am on Jessie, but I can't think why that could cause a difference.

Can you try running ``sudo depmod`` and rebooting just to be sure.

mypiandrew, 2016-03-09:
I already have a RTC present on the I2c 
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- 6e --
70: -- -- -- -- -- -- -- --

root@raspberrypi:/# lsmod
Module                  Size  Used by
rtc_ds1307             10459  0
i2c_dev                 6386  0
sg                     20575  0
sierra                  8968  0
ftdi_sio               32602  0
usbserial              29593  2 sierra,ftdi_sio
i2c_bcm2708             5988  0
spi_bcm2835             7868  0
bcm2835_gpiomem         3703  0
bcm2835_rng             2207  0
uio_pdrv_genirq         3526  0
uio                    10078  1 uio_pdrv_genirq

mypiandrew, 2016-03-09:
oh!

we have a winner

root@raspberrypi:~# modprobe mcp3422
root@raspberrypi:~# lsmod
Module                  Size  Used by
mcp3422                 4965  0
industrialio           45124  1 mcp3422
rtc_ds1307             10459  0
i2c_dev                 6386  0
sg                     20575  0
ftdi_sio               32602  0
sierra                  8968  0
usbserial              29593  2 sierra,ftdi_sio
i2c_bcm2708             5988  0
spi_bcm2835             7868  0
bcm2835_gpiomem         3703  0
bcm2835_rng             2207  0
uio_pdrv_genirq         3526  0
uio                    10078  1 uio_pdrv_genirq

root@raspberrypi:~# echo mcp3424 0x6e > /sys/bus/i2c/devices/i2c-1/new_device
root@raspberrypi:~# cd /sys/bus/i2c/devices/1-006e/iio\:device0/
root@raspberrypi:/sys/bus/i2c/devices/1-006e/iio:device0# ls -l
total 0
-r--r--r-- 1 root root 4096 Mar  9 19:27 dev
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage0_raw
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage0_scale
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage1_raw
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage1_scale
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage2_raw
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage2_scale
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage3_raw
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage3_scale
-rw-r--r-- 1 root root 4096 Mar  9 19:27 in_voltage_sampling_frequency
-r--r--r-- 1 root root 4096 Mar  9 19:27 in_voltage_scale_available
-r--r--r-- 1 root root 4096 Mar  9 19:27 name
-r--r--r-- 1 root root 4096 Mar  9 19:27 sampling_frequency_available
lrwxrwxrwx 1 root root    0 Mar  9 19:27 subsystem -> ../../../../../../../bus/iio
-rw-r--r-- 1 root root 4096 Mar  9 19:27 uevent

mypiandrew, 2016-03-09:
Did :

depmod
reboot

And then the above

mypiandrew, 2016-03-09:
As fast as my little fingers can cut wires I'll hook up Chan0 to the 1.8v rail  for a test of my hardware

Thank-you very very much, I really appreciate the time and support

popcornmix, 2016-03-09:
rpi-update does run depmod, but perhaps it didn't complete. Looks like it's okay now (and worth remembering that depmod updates module dependencies if you have problems again).

mypiandrew, 2016-03-09:
As Dracula would say...."Fangs" :-)

closed 2016-03-09 by mypiandrew