raspberrypi/linux #1334: SPI CE0 and CE1 in reverse order?

dienet, 2016-03-09:
Hi

I have noticed that when running kernel v4.4.y (and perhaps on 4.2, 4.3, 4.5) the CE0 pin and CE1 pin are in incorrect order.
So when I use in my driver's dts:
`reg = <0>;`
the CE1 pin will be used as CS.
and when:
`reg = <1>;`
the CE0 pin will be used as CS.

I think the problem lies in those lines from bcm2709-rpi-2-b.dts file:

>	spi0_cs_pins: spi0_cs_pins {
>		brcm,pins = <8 7>;
>		brcm,function = <1>; /* output */
>	};

I think that the order of brcm,pins is inncorect and should be:
`brcm,pins = <7 8>;`

Is it a bug or this behavior is intentional?

pelwell, 2016-03-09:
No, that is correct. GPIO7 is CE1 and GPIO8 is CE0, although using GPIO CS you can use any GPIOs you like.

dienet, 2016-03-10:
Ok thank you for clarifying this.

closed 2016-03-10 by dienet