raspberrypi/linux #1335: RPi3 Frequency Scaling

robingroppe, 2016-03-10:
Just a question...
Is the limitation that the new cpu can only switch between 600 and 1200MHz hardware or software related?

popcornmix, 2016-03-10:
The hardware and software supports arbitrary frequencies.
The ondemand governor we use by default just uses 600MHz and 1200MHz (on Pi3).
In general we believe it does not make sense to choose intermediate frequencies. When busy you may as well run as fast as you can, allowing the drop to the lower frequency sooner (and behaving more responsively for the user).

robingroppe, 2016-03-10:
Okay. Can you give me a hint where to modify that?
I really want to check that.

Ferroin, 2016-03-10:
Stupid question, but why is this apparently a hard coded policy in the driver?  While I can understand the reasoning for this policy in general (the bit about responsiveness that is, I don't have enough information about the power consumption profiles of the CPU to render an opinion on the power savings from doing things this way), there are perfectly valid reasons for stepping the frequency up and down instead of just jumping between high and low, and while I can understand defaulting to a particular policy, it's kind of ridiculous that there isn't any way (at least from the documentation I've read) to change it from user space.

robingroppe, 2016-03-10:
My hope is that this could also improve the temperature of the soc under modereate load.
Is there a way to modify this?

popcornmix, 2016-03-10:
You can set, say ``arm_freq=850`` into config.txt and reboot to see the effect of running at a lower max frequency is.
You can add additional frequencies to ``bcm2835_freq_table`` here:
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/cpufreq/bcm2835-cpufreq.c
Or you can set the frequency explicity using the mailbox driver. E.g. source the following:

```
function vcmailbox()
{
  /opt/vc/bin/vcmailbox $*
}

function set_freq()
{
  local clock=$1
  local freq=$2   
  if [ $freq -le 1000000 ]; then
   let "freq *= 1000000"
  fi
  echo "set_freq($clock, $freq)"
  A=$(vcmailbox 0x00038002 12 8 $clock $freq 1)
}
```
then
```
pi@domnfs:~ $ echo performance | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
pi@domnfs:~ $ set_freq 3 900
set_freq(3, 900000000)
pi@domnfs:~ $ vcgencmd measure_clock arm
frequency(45)=900000000
```

robingroppe, 2016-03-10:
Okay. I have modified the freq_table and the kernel is building. Fingers crossed i got the table right.

robingroppe, 2016-03-10:
Nope, got it wrong.
How is this supposed to look?

static struct cpufreq_frequency_table bcm2835_freq_table[] = {
	{0, 0, 0},
	{0, 0, 0},
	{0, 0, CPUFREQ_TABLE_END},
};

robingroppe, 2016-03-10:
> Lot of info here:

Yes i read that already. But still not smarter. ;)

clivem, 2016-03-11:
@popcornmix I know this isn't pretty, it's a quick 30 second hack patch, (was just intended to give @robingroppe a pointer), adding an intermediate 900MHz op point. But it doesn't actually work..... It seems that after the call to firmware to set 900MHz, that and every subsequent attempt to set freq via firmware call, returns 1.2GHz actual. Any ideas?

[rpi-kernel-cpufreq-hack-patch.txt](https://github.com/raspberrypi/linux/files/168276/rpi-kernel-cpufreq-hack-patch.txt)

bcm2835_cpufreq_module_init:127: IN
bcm2835-cpufreq:bcm2835_cpufreq_get_clock:115: Min frequency = 600000
bcm2835-cpufreq:bcm2835_cpufreq_get_clock:115: Max frequency = 1200000
bcm2835-cpufreq: min=600000 mid=900000 max=1200000
bcm2835-cpufreq:bcm2835_cpufreq_get_clock:115: Current frequency = 600000
bcm2835-cpufreq:bcm2835_cpufreq_driver_get:200: cpu0: freq=600000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 600000 -> 1200000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 2: freq 600000->1200000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 1200000 -> 900000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 1: freq 1200000->1200000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 900000 -> 600000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 0: freq 900000->1200000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 600000 -> 900000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 1: freq 600000->1200000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 900000 -> 600000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 0: freq 900000->1200000
bcm2835-cpufreq:bcm2835_cpufreq_set_clock:94: Setting new frequency = 600000 -> 900000 (actual 1200000)
bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:187: ondemand: 1: freq 600000->1200000

Ruffio, 2016-08-17:
@robingroppe has your issue been resolved? If so, please close this issue. Thanks.

buxit, 2016-11-03:
i agree that the kernel should expose more frequencies. the `cpufreq_userspace` governor is useful in some situations.

@popcornmix your `vcmailbox` snippet is great and works as expected. where can i find a list of all possible (settable) `$clock`s?

popcornmix, 2016-11-03:
@buxit see https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface

robingroppe, 2016-11-04:
The race to idle strategy may be valid for tasks that can be finished but if you have an endless task it would be better not to ramp up to full freq.
Totally forgot this issue. I would still love to see this working out.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Closing this issue as questions answered/resolved.

closed 2017-05-18 by JamesH65

Kungergely, 2019-03-03:
I'm experiencing an issue with a RPi3 where the system seems to insist on running at 600 MHz instead of the 1.2 GHz base frequency every time a USB device is plugged in. The workaround of writing 1200000 to /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq seems to help somewhat, but when trying to run vcmailbox it fails with the following error message:
` ./vcmailbox 0x00038002 12 8 3 900`
`ioctl_set_msg failed:-1`
`0x00000020 0x00000000 0x00038002 0x0000000c 0x00000008 0x00000003 0x000004b0 0x00000000`

Any hints on why is this happening?

pelwell, 2019-03-03:
What does `vcgencmd get_throttled` report?

saikat007, 2021-01-15:
@popcornmix @clivem Is it possible to scale the CPU core voltage as well after the booting process, let's say using some shell commands similar to how we can switch frequency? For example, I want to lower the CPU voltage from 1.2V(default) to 1.1V, Is there a way to do it using some tool or from the terminal by changing some file in /sys? I am using a Raspberry Pi 3B+ device.