raspberrypi/linux #1341: SSH password auth disabled on latest build

eaigner, 2016-03-11:
Can't SSH into my Pi 3 with the new build. Why has it been disabled? This effectively semi-bricks Pi's for people using them without a desktop environment.

popcornmix, 2016-03-11:
I'm still happily ssh-ing into my Pi3 with latest software. I don't believe anything has changed.
Can you explain exactly what you did between it working and not working (e.g. rpi-update or apt-get upgrade or something else?)

eaigner, 2016-03-11:
I just flashed the latest raspbian build onto the SD, and now password auth over SSH is disabled. I went the same route as before (assuming the SSH default password wasn't changed).

    ssh pi@<pi-ip>

popcornmix, 2016-03-11:
Ah, there is an option in ``sudo raspi-config`` Advanced options->SSH where this can be enabled/disabled. Does enabling that make it work?

eaigner, 2016-03-11:
The problem is, how am I supposed to run this command if I'm unable to SSH into the machine (as I said, I have no peripherals).

popcornmix, 2016-03-11:
Well, my understanding is that ssh is enabled by default.
If it's not working we'll have to work out why, and that may involve doing some experiments with a keyboard/screen attached.

eaigner, 2016-03-11:
I get the "Password:" prompt for the `keyboard-interactive` auth, but the default `raspberry` password is rejected.

eaigner, 2016-03-11:
Here is the verbose SHS log

```
$ ssh -v pi@169.254.181.244

OpenSSH_6.9p1, LibreSSL 2.1.8
debug1: Reading configuration data /Users/eaigner/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug1: Connecting to 169.254.181.244 [169.254.181.244] port 22.
debug1: Connection established.
debug1: identity file /Users/eaigner/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.9
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.9
debug1: match: OpenSSH_6.9 pat OpenSSH* compat 0x04000000
debug1: Authenticating to 169.254.181.244:22 as 'pi'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: server->client chacha20-poly1305@openssh.com <implicit> none
debug1: kex: client->server chacha20-poly1305@openssh.com <implicit> none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ecdsa-sha2-nistp256 SHA256:8+jHT5OJHVXfHOKcIG76zzuq+AnPCk1gsKgf0seD9cM
debug1: Host '169.254.181.244' is known and matches the ECDSA host key.
debug1: Found key in /Users/eaigner/.ssh/known_hosts:1
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: Roaming not allowed by server
debug1: SSH2_MSG_SERVICE_REQUEST sent
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey,keyboard-interactive
debug1: Next authentication method: publickey
debug1: Offering RSA public key: /Users/eaigner/.ssh/id_rsa
debug1: Authentications that can continue: publickey,keyboard-interactive
debug1: Trying private key: /Users/eaigner/.ssh/id_dsa
debug1: Trying private key: /Users/eaigner/.ssh/id_ecdsa
debug1: Trying private key: /Users/eaigner/.ssh/id_ed25519
debug1: Next authentication method: keyboard-interactive
Password:
debug1: Authentications that can continue: publickey,keyboard-interactive
Password:
debug1: Authentications that can continue: publickey,keyboard-interactive
Password:
debug1: Authentications that can continue: publickey,keyboard-interactive
debug1: No more authentication methods to try.
Permission denied (publickey,keyboard-interactive).
```

popcornmix, 2016-03-11:
Are you 100% sure you are using the right IP address for the Pi?
You have
> debug1: Remote protocol version 2.0, remote software version OpenSSH_6.9

I have:
> debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Raspbian-5+deb8u1

which suggests you may not be connecting to raspbian.

eaigner, 2016-03-11:
My Pi is attached via an ethernet cable to `en0`, so yes. It couldn't be another IP.

PS: I use Raspbian Lite (is there a difference?)

eaigner, 2016-03-11:
Moved my Pi to the router so it can get an IP via DHCP. Now it works again. Why does a self assigned IP not work?

popcornmix, 2016-03-11:
Can you post the ```ssh -vvv``` log with it working?

eaigner, 2016-03-11:
```
$ ssh -vvvv pi@10.0.1.17

OpenSSH_6.9p1, LibreSSL 2.1.8
debug1: Reading configuration data /Users/eaigner/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: ssh_connect: needpriv 0
debug1: Connecting to 10.0.1.17 [10.0.1.17] port 22.
debug1: Connection established.
debug1: identity file /Users/eaigner/.ssh/id_rsa type 1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ecdsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ed25519 type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/eaigner/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.9
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Raspbian-5+deb8u1
debug1: match: OpenSSH_6.7p1 Raspbian-5+deb8u1 pat OpenSSH* compat 0x04000000
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 10.0.1.17:22 as 'pi'
debug3: hostkeys_foreach: reading file "/Users/eaigner/.ssh/known_hosts"
debug3: record_hostkey: found key type ECDSA in file /Users/eaigner/.ssh/known_hosts:9
debug3: load_hostkeys: loaded 1 keys from 10.0.1.17
debug3: order_hostkeyalgs: prefer hostkeyalgs: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug2: kex_parse_kexinit: curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group-exchange-sha1,diffie-hellman-group14-sha1,diffie-hellman-group1-sha1
debug2: kex_parse_kexinit: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,ssh-ed25519-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ssh-dss-cert-v01@openssh.com,ssh-rsa-cert-v00@openssh.com,ssh-dss-cert-v00@openssh.com,ssh-ed25519,ssh-rsa,ssh-dss
debug2: kex_parse_kexinit: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,arcfour256,arcfour128,aes128-cbc,3des-cbc,blowfish-cbc,cast128-cbc,aes192-cbc,aes256-cbc,arcfour,rijndael-cbc@lysator.liu.se
debug2: kex_parse_kexinit: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,arcfour256,arcfour128,aes128-cbc,3des-cbc,blowfish-cbc,cast128-cbc,aes192-cbc,aes256-cbc,arcfour,rijndael-cbc@lysator.liu.se
debug2: kex_parse_kexinit: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1,hmac-md5-etm@openssh.com,hmac-ripemd160-etm@openssh.com,hmac-sha1-96-etm@openssh.com,hmac-md5-96-etm@openssh.com,hmac-md5,hmac-ripemd160,hmac-ripemd160@openssh.com,hmac-sha1-96,hmac-md5-96
debug2: kex_parse_kexinit: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1,hmac-md5-etm@openssh.com,hmac-ripemd160-etm@openssh.com,hmac-sha1-96-etm@openssh.com,hmac-md5-96-etm@openssh.com,hmac-md5,hmac-ripemd160,hmac-ripemd160@openssh.com,hmac-sha1-96,hmac-md5-96
debug2: kex_parse_kexinit: none,zlib@openssh.com,zlib
debug2: kex_parse_kexinit: none,zlib@openssh.com,zlib
debug2: kex_parse_kexinit:
debug2: kex_parse_kexinit:
debug2: kex_parse_kexinit: first_kex_follows 0
debug2: kex_parse_kexinit: reserved 0
debug2: kex_parse_kexinit: curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group14-sha1
debug2: kex_parse_kexinit: ssh-rsa,ssh-dss,ecdsa-sha2-nistp256,ssh-ed25519
debug2: kex_parse_kexinit: aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,chacha20-poly1305@openssh.com
debug2: kex_parse_kexinit: aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,chacha20-poly1305@openssh.com
debug2: kex_parse_kexinit: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: kex_parse_kexinit: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: kex_parse_kexinit: none,zlib@openssh.com
debug2: kex_parse_kexinit: none,zlib@openssh.com
debug2: kex_parse_kexinit:
debug2: kex_parse_kexinit:
[... cut ...]
debug1: Trying private key: /Users/eaigner/.ssh/id_ecdsa
debug3: no such identity: /Users/eaigner/.ssh/id_ecdsa: No such file or directory
debug1: Trying private key: /Users/eaigner/.ssh/id_ed25519
debug3: no such identity: /Users/eaigner/.ssh/id_ed25519: No such file or directory
debug2: we did not send a packet, disable method
debug3: authmethod_lookup password
debug3: remaining preferred: ,password
debug3: authmethod_is_enabled password
debug1: Next authentication method: password
pi@10.0.1.17's password:
```

popcornmix, 2016-03-11:
> debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Raspbian-5+deb8u1

The working connection reports itself as raspbian, and the non-working is not raspbian. Pretty sure you were failing to connect to a device other than the pi.

eaigner, 2016-03-11:
Since theres obviously something else going wrong, I'll close this for now.

closed 2016-03-11 by eaigner