raspberrypi/linux #1344: Identified commit that breaks vc4

grbd, 2016-03-12:
Hi
I've recently been experimenting trying to get the vc4 drivers to work with 4.5
I've not got vc4 working fully yet (I still need to have a mess with some of the config settings which I think is the next problem)
But I've discovered that one of the commits in this tree prevents the system from booting if dtoverlay=vc4-kms-v3d is enabled in config.txt

The offending commit appears to be
```
commit 8924ecf62e9ee8ec991d6361152e17268cc7ed6f
Author: Eric Anholt <eric@anholt.net>
Date:   Mon Oct 12 11:23:34 2015 -0700

    clk: bcm2835: Also build the driver for downstream kernels.

    Signed-off-by: Eric Anholt <eric@anholt.net>
```

If I use the commit just before it then the system will boot (I still get a black screen but at least I can ssh onto the box
and look at dmesg to see what's going on with vc4)

```
# Working Version
git clone https://github.com/raspberrypi/linux.git test1
git checkout -b rpi-4.5.y origin/rpi-4.5.y
git checkout b47868e0b58e112a1bebee16e35144673e09643b
```

I'm using the default config make bcm2709_defconfig
and the dtb files I'm using are actually from the very latest version (since that one has the files for the rpi3 added in)

popcornmix, 2016-03-12:
Perhaps @anholt could let us know if any additional commits are needed for vc4-kms-v3d use on 4.5 kernel.

grbd, 2016-03-14:
* For info I did also try the config from https://people.freedesktop.org/~anholt/dotconfig-bcm2835-20160309.txt
 * Also tried two different monitors
 * Also I re-tried it after the repo was upped to 4.5-rc7 but with the same result
 * Firmware is the latest pull from March the 11th master https://github.com/raspberrypi/firmware/commits/master

The build process I'm using includes
```
make -j4 zImage modules dtbs
make modules_install
scripts/mkknlimg arch/arm/boot/zImage /boot/kernel7-gentoo10-vc4.img
```

cmdline.txt
```
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/sda3 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait cma=256M@512M
```

config.txt (uncommented bits)
```
kernel=kernel7-gentoo10-vc4.img
avoid_warnings=2
dtoverlay=vc4-kms-v3d

# default entries
dtparam=audio=on
hdmi_force_hotplug=1
```

The only thing left I can think of is to hookup something to the gpio to try and capture something from the gpio uart, since I can't see anything on screen, just that the capslock key stops responding on a usb attached keyboard

grbd, 2016-03-14:
Hmm the new version drm-vc4-4.5 from anholt's repo doesn't boot on my rpi2 or rpi3 yet, but the caps lock key still keeps working. So I think it's a little closer

closed 2017-05-18 by JamesH65