raspberrypi/linux #1345: Broken userspace crypto in linux-4.1.18

g-andrade, 2016-03-12:
Hi,

I've recently tried to mount a luks device under a Raspbian jessie-lite installation on a RPi 3, which is running kernel 4.1.18; it kept refusing to map it, giving out the following error:
`# Userspace crypto wrapper cannot use aes-xts-plain (-22).`
`Failed to setup dm-crypt key mapping for device /dev/sda2.`
`Check that kernel supports aes-xts-plain cipher (check syslog for more info).`

..and not any info to be found in syslog, contrary to the suggestion.

This kept happening even though I tried to load all sorts of modules I could think of that might had been a dependency; eventually I found the reference to [this particular discussion](https://www.mail-archive.com/linux-crypto@vger.kernel.org/msg17905.html) in the linux-crypto mailing list, and it seems to fit.

Info from luksDump:
`Version:        1`
`Cipher name:    aes`
`Cipher mode:    xts-plain`
`Hash spec:      sha512`

\# uname -a
`Linux raspberry 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux`

\# cryptsetup benchmark
`# Tests are approximate using memory only (no storage IO).`
`PBKDF2-sha1       152409 iterations per second`
`PBKDF2-sha256     100824 iterations per second`
`PBKDF2-sha512      21557 iterations per second`
`PBKDF2-ripemd160  137970 iterations per second`
`PBKDF2-whirlpool   10570 iterations per second`
`#  Algorithm | Key |  Encryption |  Decryption`
`     aes-cbc   128b           N/A           N/A`
` serpent-cbc   128b           N/A           N/A`
` twofish-cbc   128b           N/A           N/A`
`     aes-cbc   256b           N/A           N/A`
` serpent-cbc   256b           N/A           N/A`
` twofish-cbc   256b           N/A           N/A`
`     aes-xts   256b           N/A           N/A`
` serpent-xts   256b           N/A           N/A`
` twofish-xts   256b           N/A           N/A`
`     aes-xts   512b           N/A           N/A`
` serpent-xts   512b           N/A           N/A`
` twofish-xts   512b           N/A           N/A`

[Contents of /proc/crypto](https://github.com/raspberrypi/linux/files/170612/crypto_info.txt)
[Output of lsmod](https://github.com/raspberrypi/linux/files/170613/lsmod_info.txt)

g-andrade, 2016-03-12:
Ok, I've in the mean time realized that the fixes have been backported on 762330b, 1ce5c14, 7327b23 and 64c4131. We'll have to wait for a 4.1.19 build to hit the repo, I suppose?

popcornmix, 2016-03-12:
rpi-update contains the fix. There is a bump to the apt-get firmware due imminently (next few days) if you'd prefer to wait for the stable version.

popcornmix, 2016-03-18:
apt-get kernel includes this fix. Okay to close?

g-andrade, 2016-03-18:
Sure; I had thought about closing it before but I preferred to wait for the fix to hit the repo. Thanks!

Sent from my phone. Please excuse my brevity.

closed 2016-03-18 by popcornmix