raspberrypi/linux #1346: Raspberry Pi 2 B Crash after couple of hours

ksy5662, 2016-03-13:
Hi.

I have experienced every crash with my Raspberry Pi 2 B model.
3.8A maximum power Samsung 16GB micro SD class 10 memory...

Raspbian Jessi image with apt-get update and apt-get upgrade

Here are two screenshot after crash

anybody give me any suggestion to cope with

[image]
[image]

popcornmix, 2016-03-13:
Can you run in a higher resolution mode where we might see more of the kernel backtrace?
Do you have any peripherals attached? Do you still get the panic with them removed?
What does ``uname -a`` and ``vcgencmd version`` report?
Are you overclocking? If you are try removing that.
Do you see any flickering of the PWR LED?

JosephGarrone, 2016-03-28:
We too are having this issue:

[image]

ksy5662, 2016-03-28:
I got my money back from distributor. They said it is problem^^

popcornmix, 2016-03-29:
Closing as it sounds like a hardware issue.

closed 2016-03-29 by popcornmix

JosephGarrone, 2016-03-29:
We managed to resolve this by turning off device tree in the raspi config. Not sure if that makes it hardware or kernel issue.